genodelabs/genode #5254: Uncaught Double_insertion with nvme and platform_drv in Sculpt 24.04.1

cvparker, 2024-06-20:
This happens with Sculpt 24.04.1 (4b9d02fe31e80ed831553be49436554169dcf68b) on Apple MacBook with the main drive installed in nvme.

It reads the partition table, but following any access to a partition, the platform_drv component fails with the message:

Error: Uncaught exception of type 'Genode::Final_table<Intel::Level_1_descriptor<12u> >::Double_insertion'

So I am able to trace this a little, the exception is thrown within Driver::Session_component::alloc_dma_buffer(unsigned long, Genode::Cache), specifically within Genode::Registry<Driver::Io_mmu_domain>::for_each<...>(...), so I assume this is related to IO_mmu in some way. If it helps, the address it attempts to allocate is 0x800000 with size 0x400000.

Interestingly an exception is thrown twice when attempting to allocate that region, the first time on startup. In that case, based on the log this is probably happening when the nvme partition table is read. However, this exception is not fatal, because it's of type Genode::Out_of_ram. The second time is when trying to use a partition and the exception is Double_insertion. One theory is that, prior to the first exception, the page table in IO_mmu gets partially filled, but the allocator ignores that this happened due to the exception. So, when another request is made, the table is partially full, although the allocator thinks it's a free region.

I considered giving more RAM to various components (platform_drv and nvme), but that didn't help, and no component appears to be actually close to its limit. The next step in my debugging would be to try to figure out where/why the Out_of_ram is thrown. I suppose that this is in a higher level page allocator during allocation of space for a lower level page table? But, why isn't this fixed by increasing the quota? I'm curious if anyone here can offer insights.

cvparker, 2024-06-20:
In response to @jschlatow:

There seem to be two problems here:
1. The Double_insertion seems to be triggered by a preceding Out_of_ram
    during DMA buffer allocation. This is definitely a bug.
2. The initial Out_of_ram exception might also be an indicator that
    something strange is happening. What component is logging the
    exception? Normally, the DMA buffer allocation is retried by the
    Client with additional RAM. The allocation should therefore succeed
    during the second try and not log any exception.

1) Yes, I think so too.
2) To clarify, the component logging the exception is the platform_drv, but this logging only occurs as a result of my intervention in order to trace the problem. On a stock build it will only log the fatal exception. You can see my debugging strategy here: https://github.com/cvparker/genode/commit/45f4d6a52c5727c2544c14817eaf4159d32f27e7

jschlatow, 2024-06-21:
> 2\. To clarify, the component logging the exception is the platform_drv, but this logging only occurs as a result of my intervention in order to trace the problem. On a stock build it will only log the fatal exception. You can see my debugging strategy here: [cvparker@45f4d6a](https://github.com/cvparker/genode/commit/45f4d6a52c5727c2544c14817eaf4159d32f27e7)

Thanks for the clarification. I have a few follow up questions: After the Out_of_ram exception, do you see the same "insert_translation vaddr..." message? As mentioned before, the client should retry the DMA buffer allocation with more RAM, I would like to know whether the second try uses the same buffer addresses. Secondly, can you please log the virtual and physical addresses and size when the Double_insertion occurs?

cvparker, 2024-06-21:
To answer the first question, you can see that in addition to exceptions I've logged any access for the specific range where the Double_insertion occurs. There are only two such accesses, and both of them lead to exceptions (first Out_of_ram, and second Double_insertion). So if the client retries the allocation I suppose it will be given different buffer addresses, or else my logging strategy is incomplete and I miss it somehow.

Here's what I have easily on hand. Note that my only photo showing the stack before the throw was before the address and size were printed. It occurs to me that I might be able to pull logs off directly with a USB stick, but for now I have cell photos of the screen. I attached stack traces here too.

[image]
[image]
[image]
[stack_trace.txt](https://github.com/user-attachments/files/15928106/stack_trace.txt)

jschlatow, 2024-06-24:
@cvparker I was able to synthetically reproduce your problem. Furthermore, I suspect that your platform does not seem to support super pages, which would explain why an Out_of_ram during a mapping of a 4MB buffer that does not cross a 1GB boundary leaves partial mappings.

Can you give a834f27 a try?

cvparker, 2024-06-25:
Your revision a834f27 worked. I was able to mount the nmve drive partition that I wanted, run fsck, and browse files. Here's the log, if that's helpful, and the output of /report/drivers/iommu. However, I did not yet set Intel::Io_mmu::_verbose to true.

[log.after_success.txt](https://github.com/user-attachments/files/15963722/log.after_success.txt)

[iommu.txt](https://github.com/user-attachments/files/15963724/iommu.txt)

I don't know how to check for super pages support, is there a /proc/cpuinfo or /proc/meminfo from running linux on the same machine that would reveal it?

Although the first theory of this bug seems correct, I also looked at the try block in repos/os/src/drivers/platform/session_component.cc:382. For example if Dma_allocator::Out_of_virtual_memory could also be thrown from add_range (I don't think so), or if alloc_buffer also needs to be undone in the event of Out_of_ram or Out_of_caps concurring in add_range.

jschlatow, 2024-06-25:
> Your revision [a834f27](https://github.com/genodelabs/genode/commit/a834f275618702af964b9b4922891847e083d41a) worked. I was able to mount the nmve drive partition that I wanted, run fsck, and browse [...]

Perfect!

> [log.after_success.txt](https://github.com/user-attachments/files/15963722/log.after_success.txt)
>
> [iommu.txt](https://github.com/user-attachments/files/15963724/iommu.txt)
>
> I don't know how to [...]

That information is encoded in bits 34 and 35 of the capability register (dumped in /report/drivers/iommu). In your case, these bits are zeroed.

> Although the first theory of this bug seems correct, I also looked at the try block in repos/os/src/drivers/platform/session_component.cc:382. For example if Dma_allocator::Out_of_virtual_memory [...]

Thanks for looking into that. `Dma_allocator::Out_of_virtual_memory` would be thrown before `add_range` is called. The reversal of `alloc_buffer()` is taken care of by the `Guard` object.

jschlatow, 2024-06-25:
@chelmuth Please merge a834f27 to staging.

commit c44f9889 (2024-07-08): platform/pc: remove translations after Out_of_ram

Since page tables might need to be allocated during
insert_translation(), Out_of_ram or Out_of_caps exceptions might occur.
Entries that have already been added by insert_translation() must thus be
removed once one of those exceptions occurred.

Fixes #5254
commit 09c40688 (2024-08-27): platform/pc: remove translations after Out_of_ram

Since page tables might need to be allocated during
insert_translation(), Out_of_ram or Out_of_caps exceptions might occur.
Entries that have already been added by insert_translation() must thus be
removed once one of those exceptions occurred.

Fixes #5254
closed 2024-08-29 by chelmuth