genodelabs/genode #5259: usb: set control transfer timeout in libusb

skalk, 2024-06-26:
Re-enable the setting of a timeout in the libusb backend implementation, otherwise certain devices (e.g. smartcard-reader) won't do any progress.

nfeske, 2024-06-26:
Merged to staging.

chelmuth, 2024-07-08:
I'm wondering if this is an intermediate step as you planned to apply a default timeout in the host driver and remove the timeout from the API.

skalk, 2024-07-30:
@chelmuth that is true, but after looking at the different timeout values more closely again, I decided against the default timeout, as they can be quite different. Therefore, having the option within the API seemed necessary. Alternatively, we would need the ability to cancel a single request to track timeouts client-wise.

chelmuth, 2024-07-30:
Thanks for the clarification - hence, timeouts in the API make much sense :-)

closed 2024-08-29 by chelmuth