genodelabs/genode #5260: use /data/depot as depot directory in debug info

jschlatow, 2024-06-27:
Currently, debug archives published by genodelabs refer to source files in /data/depot. When building debug archives on another system, however, the debug info files will refer to a different depot directory. By using GCC's `--debug-prefix-map` argument, we can make sure that debug archives always refer to source files at /data/depot.

jschlatow, 2024-06-27:
Implemented by 41ae56f

nfeske, 2024-06-27:
The /data prefix is just circumstantial, just because the build machine happens to have this particular directory. I wonder, should this better be dropped from the patch?

jschlatow, 2024-06-28:
> The /data prefix is just circumstantial, just because the build machine happens to have this particular directory. I wonder, should this better be dropped from the patch?

That'd actually be better. I was under the assumption that our gdb_x86 expects the depot at /data/depot, well, it does not. Dropping the /data prefix will thus fix the "No such file or directory" messages for on-target debugging with gdb. I'll prepare a fixup.

jschlatow, 2024-06-28:
Fixup commit 1780f75 drops the /data prefix.

nfeske, 2024-06-28:
Very good. Merged as https://github.com/genodelabs/genode/commit/a4211209760511e677bab59f6d0bc15510cb0b30 to staging.

nfeske, 2024-07-01:
Funnily, the use of `#` as separating character for `sed` trips off `make` version 4.2:

```
repos/base/mk/global.mk:128: *** unterminated call to function 'shell': missing ')'.  Stop.
```

Fixup https://github.com/genodelabs/genode/commit/5251c91ede0c98a98444a74ca567db9d88dbe67b addresses this problem by using `|` instead.

commit a4211209 (2024-06-28): use /depot as depot directory in debug info

By using GCC's --debug-prefix-map argument, we can make sure that debug
archives always refer to source files at /depot. With this change, GDB
can be pointed to the correct source-file location by using the `set
substitute-path /depot /path/to/local/depot`.

Fixes #5260
commit b5c91074 (2024-07-02): use /depot as depot directory in debug info

By using GCC's --debug-prefix-map argument, we can make sure that debug
archives always refer to source files at /depot. With this change, GDB
can be pointed to the correct source-file location by using the `set
substitute-path /depot /path/to/local/depot`.

Fixes #5260