genodelabs/genode #5261: Make policy on AI generated or assisted code submissions

axiomsss, 2024-07-02:
I believe it is important to make an explicit policy about AI generated code submissions and or AI assisted code submissions. 

This is due to many reasons. However I believe the most prescient reason is legal, as AI generated content is in a grey area of legality and copyright law, especially in the EU.

For instance, the Gentoo project has publically made a hardline stance on banning all AI generated code until further notice. I believe this is the correct move.

I understand that it may not be possible to validate if code has been AI generated, however a policy should at least be in place to protect the project and let future contributors be aware of such guidelines.

nfeske, 2024-07-02:
The guiding information about contributing to Genode are given at the [Contributions](https://genode.org/community/contributions) page.

Contributions are accepted only once the author gives the written consent to the [Genode Contributor's Agreement](https://genode.org/community/gca.pdf). This agreement contains the following clause:

> You certify and warrant that your Contributions to Genode Labs’s Products do not violate the intellectual
property rights of third parties [...]

Assuming that AI-generated content violates the intellectual property rights of the authors of the underlying training material, as a contributor, I would not claim such a legal position for any code not written by myself in person.

axiomsss, 2024-07-02:
> Assuming that AI-generated content violates the intellectual property rights of the authors of the underlying training material, as a contributor, I would not claim such a legal position for any [...]

Yes, I mostly agree with this. However I believe there is ambguity and a user could bend this interpretation.

A user may believe that content produced by an AI is actually their work because they prompted it, or made several prompts to have it produce the code they want, when in reality, the majority of the code is still AI generated, even though it has been prompted by a user originally.

The ratio of characters input to prompt text to AI output text would be in favour of it being more AI content than user content. Say, for every word of their prompt, the AI will output 10 words. This would be a 10:1 ratio of it being more AI than user content, even though the user did have their own original content to create the output.

So I think it may be beneficial to explicitly make policies on this to avoid user submission ambiguity, as some users may believe their prompts or other inputs constitute original work, when it is debatable whether it is truly their own work.

While I personally do not use any AI tools and I have no intention to for the forseeable future, it is increasingly becoming apparent that AI code and content will become more prevasive and isn't going to disappear, so it is helpful to try and put policies and guidelines in place early rather than too late.

nfeske, 2024-07-03:
Thanks for having taken the time to express your concerns. You made your point very clear. Please be assured that I personally share your opinion on the AI matter.

That said, wearing the project-maintainer hat, the establishing of a policy is an entirely different matter than the formulation of an opinion. Policy, if unenforced, ultimately creates room for uncertainty. As maintainer, it would be my job to do the enforcement. Since I'm not in the position to dictate the tools used by contributors, enforcement is unrealistic. Hence, uncertainty is all that remains. For this reason, this issue won't be addressed in the way you prefer.

I believe that collaboration between humans is best founded on mutual trust and a strong joint sense of responsibility. Trust does not follow from policy. It grows with positive social interactions, fostered reputation, and consistent behavior over time. In the context of Genode's source repository, the group of contributors is rather tight-knit. AI pollution won't become a thing in our endemic code base unless this group of reputable persons let it happen, which I doubt.

Hoping that I was able to clarify, I'd appreciate you closing this issue.

axiomsss, 2024-07-03:
Yes you have made a good point. I also don't think it is possible to ensure how a user has created their code, I also don't think it would really be a legal issue on the project because it is simply not possible for maintainers to know how code was created besides the submitter themselves, so we must trust them telling the truth.

I apologise for making you use your time to comment on this issue. I thought it's just an interesting grey area at present which has never been an issue in FOSS until recent so I was curious about an AI policy, however the current policy quote that you linked does encompass this, albeit not directly stating AI tools.

closed 2024-08-01 by nfeske