genodelabs/genode #5264: lx_emul: update default linux port to version 6.6.35

ssumpf, 2024-07-04:


ssumpf, 2024-07-04:
These commits mark the first three steps. This is a starting point and more testing on all platforms is still required.

ssumpf, 2024-07-25:
The commits above (issue5264_dde_linux-6.6 branch) + [c989248](https://github.com/ssumpf/genode-zynq/commit/c989248f617514ce7a08a4c7748235944ea94f6d) have been tested on hardware (everything but arm_v7) and could go to staging.

chelmuth, 2024-07-25:
Could the branch easily be rebased on staging incl. #5236 ?

ssumpf, 2024-07-25:
> Could the branch easily be rebased on staging incl. #5236 ?

@chelmuth: I will try.

ssumpf, 2024-07-25:
@chelmuth: Okay, all virt_linux using _generated_dummies.c_ have to be re-created. On a first test with lxip (x86_64), I see segfaults (looks like missing function pointers)  and things like:

`Error: Lx_kit::Scheduler::_current is zero!`

The update to staging with #5236 requires to redo all virt_linux tests on all platforms. This might take a day or two depending what issues in what driver/protocol stack and/or platform show up.

chelmuth, 2024-07-26:
@ssumpf please consider fixup 1f826998c2e3b7ba6204cda3d6255958b0522bdc on staging

ssumpf, 2024-08-01:
My [dde_linux-6.6-pc_linux](https://github.com/ssumpf/genode/tree/dde_linux-6.6-pc_linux) branch can be used as a basis to update `pc_linux` based x86 drivers.

chelmuth, 2024-08-01:
Great, I'm going to update the pc_nic driver today.

BTW, are you planning to fast-forward 6.6.35 to 6.6.current (currently .43)?

chelmuth, 2024-08-02:
I have a question regarding `extern unsigned long loops_per_jiffy` which is used in *kernel/workqueue.c* `wq_cpu_intensive_thresh_init()` for calculation. As the symbol ends up in *generated_dummies.c* in all current adaptions the calculation results in `0` in all cases. May this become an issue?

ssumpf, 2024-08-02:
> I have a question regarding `extern unsigned long loops_per_jiffy` which is used in _kernel/workqueue.c_ `wq_cpu_intensive_thresh_init()` for calculation. As the symbol ends up in _generated_dummies.c_ in all current adaptions the calculation results in `0` in all cases. May this become an issue?

As far as I can see `wq_cpu_intensive_thresh_us` will be set to `USEC_PER_SEC` because `bogo < 4000` in _kernel/workqueue.c_. Since `wq_cpu_intensive_tresh_us` determines if something is marked as "CPU  intensive and excluded from concurrency", everything running longer than a second will be marked "intensive".

chelmuth, 2024-08-02:
@ssumpf please consider the following commits.

8226beb0845 pc_nic: update to 6.6.x
c469dd87156 fixup "pc: add common dummies + generic dep.list entries" (trace/events/maple_tree.h)
7201cb2977f fixup 'dde_linux: Linux 6.6.35 port' (remove devio patch)

x86_32 is still missing but x86_64 on U7411 netperf_lwip results look good.

```
! PERF: TCP_STREAM              938.09 Mbit/s ok
! PERF: TCP_MAERTS              932.15 Mbit/s ok
```

ssumpf, 2024-08-02:
@chelmuth: I will rebase to your state and push usb_host next, Josef in turn will use my branch as a basis to add Wifi. Is c469dd8 missing the common dummies part?

chelmuth, 2024-08-02:
> Is [c469dd8](https://github.com/genodelabs/genode/commit/c469dd871566bca91796ca38ee9890bdf79b55a2) missing the common dummies part?

No, it's just the missing maple_tree header. Please also push the virt_linux maple_tree stuff, so I can also test netperf_lxip ;-)

chelmuth, 2024-08-02:
Commit 96ee94b8de2d5f997e7b8fdf1a8ed96673dd3e67 adds x86_32.

chelmuth, 2024-08-02:
@ssumpf I got the following error from lxip in netperf_lxip on run 3.

```
[init -> netserver_genode] Error: spinlock contention!
[init -> netserver_genode] Error: Function arch_spin_lock not implemented yet!
```

cnuke, 2024-08-02:
* Commit [d25bca6](https://github.com/cnuke/genode/commit/d25bca6dd14f99bab036d28a3830424500499f49) is a prerequisite for the `pc_wifi` driver update

* Commit [3233721](https://github.com/cnuke/genode/commit/32337210868fc5b2d4e60f064c3cc696d90347f2) is still *WIP*.
  It was successfully tested on Fuji4 (8265) and Fuji6 (AX211) running *x86_64* but exhibits incomplete scan results on a X201 (6300) and thus connecting to the testing AP was not possible. The same machine was also used for testing *x86_32* where two distinct page-faults happen early on during mac80211/driver bringup with the backtrace pointing to the SLUB.

cnuke, 2024-08-08:
* Commit [b6a2b3b](https://github.com/cnuke/genode/commit/b6a2b3b474b2100250ea47a4b84aea34d29597a8) addresses the double-free bug uncovered by using the SLUB.
* Commit [a87e4b2](https://github.com/cnuke/genode/commit/a87e4b21a64c7cc19b7123a617c688cc10b910ad) updates the `pc_wifi` driver and was successfully tested on the X201 (32 and 64bit), ultra (32 and 64bit), macho¹ (64bit), fuji4 (64bit) and fuji6. It contains the necessary checks to allow for the ARM drivers still residing on *6.1.20* to compile.

¹) I noticed during testing that macho (AX210) cannot connect to 2.4GHz APs (tried both available networks) and results in
`iwlwifi 02:00.0: Not associated and the session protection is over already...` - 5GHz works fine though. I have yet to investigate this further but judging by the *nic_router_uplinks.run* results (the second network uses 2.4GHz) this issue already existed before the update.

chelmuth, 2024-08-09:
Merged 72317569e08 wifi: fix double free during FW loading to staging already.

Tests of the branch with `run/netperf_lwip` on U7411 (AX201) succeeded for 2.4 (10 MBit/s) and 5 GHz (130 MBit/s), but the throughput drop with 2.4 looks suspicious.

chelmuth, 2024-08-14:
I gathered all fixups and @alex-ab's intel_fb work on https://github.com/chelmuth/genode/commits/dde_linux-6.6-pc_linux. This state works with run/sculpt_test on fuji5 (internal display only).

cnuke, 2024-08-16:
Commit [96ca2fe](https://github.com/cnuke/genode/commit/96ca2fec862bdae8551d2835c5fa5358b5bcbb82) makes the `pc_intel_fb` driver work on my Shuttle DS57U (Celeron 3205U).

cnuke, 2024-08-16:
Commit [e9fbbaf](https://github.com/cnuke/genode/commit/e9fbbaf8bb93448693760121505697b30917e990) adjusts `wifi.inc` to reference the *linux* port.

ssumpf, 2024-08-16:
Commit 1db4f71 propagates thread names to the `Lk_kit::Task`s. d2242e8 increases the priority of ksoftirqd, so software IRQs are serviced directly by ksoftirqd when woken up and not through `__local_bh_enable_ip` of some other task.

chelmuth, 2024-08-19:
> [d2242e8](https://github.com/genodelabs/genode/commit/d2242e8a5fa7f5f7046f880628bd1741669e1e11) increases the priority of ksoftirqd, so software IRQs are serviced directly by ksoftirqd when woken up and not through `__local_bh_enable_ip` of some other task.

It is quire unfortunate that you have to patch linux in this way. Thus, it seems worthwhile to think about alternative approaches.

Do I understand correctly, that in case of the lockup the softirq is already pending when the lxip user task starts its execution in [tcp_v4_rcv](https://elixir.bootlin.com/linux/v6.6.35/C/ident/tcp_v4_rcv)? If yes, may you try to insert an additional `Lx_kit::env().scheduler.execute();` right before unblocking the lxip user task and calling the scheduler in *socket.cc*?

ssumpf, 2024-08-19:
> It is quire unfortunate that you have to patch linux in this way. Thus, it seems worthwhile to think about alternative approaches.
>
> Do I understand correctly, that in case of the lockup the softirq [...]

@chelmuth: Since it happens while receiving I can try to do this before unblocking `lx_nic_client_rx_task` (init.cc). I will let you know of the outcome.

ssumpf, 2024-08-20:
@chelmuth: 1b80acd implements the `Lx_kit::env().scheduler.execute();` for the rx -task. Looks like it's working :smiley:

chelmuth, 2024-08-20:
Again, I updated my branch with the commits mentioned here except the *wifi.inc* adaption which is currently revised by @cnuke. Before merging the commits to staging I'm going to squash several fixups and minor commits.

What do you think about forwarding the Linux version to 6.6.47?

Also, @nfeske noted that we may reconsider the interplay of the I/O signal handler and VFS wakeup in lxip, because the signal currently calls the scheduler directly (and therefore is not free of side effects).

cnuke, 2024-08-20:
Commit [c593685](https://github.com/cnuke/genode/commit/c59368574ebd9f6072164b475bc8d50d7f9b0936) contains the revised change.

ssumpf, 2024-08-20:
> What do you think about forwarding the Linux version to 6.6.47?

I can give it a try.

ssumpf, 2024-08-20:
@chelmuth: The commits above forward to 6.6.47. I test compiled as much as I could.

chelmuth, 2024-08-21:
I'm going to give run/sculpt_test and a modular sculpt some hours to prove itself.

chelmuth, 2024-08-21:
@alex-ab thanks again for the `used` mode attribute in the display connectors! I love it because it takes a blink of an eye to identify the current mode now in long mode lists.

DPI info for the U7411 internal display is correct but my Wimaxit reports 81 DPI on FullHD which sums up to a 60 cm wide screen...

chelmuth, 2024-08-21:
> @chelmuth: The commits above forward to 6.6.47. I test compiled as much as I could.

Do I have to update/adapt additional genode repos (zynq)?

chelmuth, 2024-08-21:
I pushed a clean branch incl. 6.6.47 just now.

ssumpf, 2024-08-26:
> The results of the last days look promising! Though, we should look into `run/intel_fb` on x86_64 (most likely a run-script detection issue) and `run/usb_hid_raw` on x86_32/not_nova (Error: Function wake_q_add_safe not implemented yet!).

@chelmuth: d3c4421 addresses the wake_q_add_safe function for virt/pc_linux.

chelmuth, 2024-08-26:
Merci beaucoup!

ssumpf, 2024-09-09:
@skalk, @cnuke: My `linux-imx-6.6-update` branches (genode and genode-imx) contain the USB host driver for imx8mq (tested), imx (untested) which may be used as a basis for further work.

ssumpf, 2024-09-10:
@cnuke, @skalk: My `genode-imx/linux-imx-6.6-update` branch now contains USB host controller support for the MNT Pocket Reform.

@a-dmg: I have added a stripped down version of your [usb_host/imx8mq: add support for imx8mp_iot_gate](https://github.com/ssumpf/genode-imx/commit/36b8d3807cbad8590581ce506aa21c29dcf5339a) commit. Could you give it a try on the iot_gate and determine if _drivers/usb/dwc3/dwc3-of-simple.c_ and _drivers/usb/dwc3/dwc3-haps.c_ are really required on the platform with kernel 6.6? If so, I will return both files to the _source.list_. For testing you would need the following branches from my profile: `ssumpf/genode/linux-imx-6.6-update` and `ssumpf/genode-imx/linux-imx-6.6-update`. Thx.

a-dmg, 2024-09-11:
@ssumpf I have tested with your branch, and it runs without any change! Indeed _drivers/usb/dwc3/dwc3-of-simple.c_ and _drivers/usb/dwc3/dwc3-haps.c_ can be omitted.

ssumpf, 2024-09-11:
> @ssumpf I have tested with your branch, and it runs without any change! Indeed _drivers/usb/dwc3/dwc3-of-simple.c_ and _drivers/usb/dwc3/dwc3-haps.c_ can be omitted.

@a-dmg: Super and thanks for testing.

ssumpf, 2024-09-11:
@cnuke, @skalk: [7a7f760](https://github.com/ssumpf/genode-imx/commit/7a7f760e98c165b3aec5024b5c67c03be993fb5b) services the watchdog timer if running.

ssumpf, 2024-09-12:
@skalk:
[9360459](https://github.com/ssumpf/genode-imx/commit/9360459415814fce6575bd1296c15203ffaa49cd) solves a driver loading dependency issue for  USB host on imx8mp. 
[6728dc2](https://github.com/ssumpf/genode-imx/commit/6728dc28e4676585742bbf6f9696994cef11071d) renames usb_host/imx8mq to imx8m because we support imx8mq/imx8mp now.

ssumpf, 2024-09-13:
@skalk: [6854b58](https://github.com/ssumpf/genode-imx/commit/6854b581a222b101fbe29cf2c79faaae7473c691) update the framebuffer driver for imx8mq. Next, I will update to your `linux-imx-6.6-update` branch and look into the imx8mp display driver.

cnuke, 2024-09-20:
@ssumpf @skalk commit [4f000a1](https://github.com/cnuke/genode-imx/commit/4f000a14307c1210ca5734e09c27a7e9ba36189b) allows for using the internal display on the *MNT Reform*. External HDMI already worked fine before that.

(We could collect and consolidate all the different dummy drivers and put them into `lx_emul/imx` similar to how it is handled in the `allwinner` repository.)

skalk, 2024-09-24:
@chelmuth: if you're fine with commits 7b08e5426c0b42490a306459ee0475735e7c978f 071237b1dc4f9ca9c277793c6660fbd07bed7f56 and 7bfcae563345470c096c41aeefd2f25289ab6bac please consider it for staging. As soon as it enters staging, I would update the genode-imx repository to reflect the recent update work.

chelmuth, 2024-09-24:
I merged the series as is.

cnuke, 2024-10-01:
The following commits add support for the QCA9377 wireless device found in the *MNT Pocket Reform* via an adapted version of the `qcacld-2.0` driver:

* [9dfd7e2](https://github.com/cnuke/genode/commit/9dfd7e251b3649bc33e5be9b9eed611f51b5e7d5) fixes the firmware pointer check uncovered by the `qcacld-2.0` driver
* [529a453](https://github.com/cnuke/genode/commit/529a453ffa97df6e3058b5e953125647e6435eec) adds the necessary firmware images to the `linux-firmware` port
* [fd39ff1](https://github.com/cnuke/genode-imx/commit/fd39ff11db94cf710d50cfa372a9cb2c5563f577) adds the driver to the `linux-imx` port
* [ecc3716](https://github.com/cnuke/genode-imx/commit/ecc3716551102bf6f8fd9f7eed3541d7065b8e78) adds support in the platform driver for `imx8mp`
* [80c60cb](https://github.com/cnuke/genode-imx/commit/80c60cb510e1c191bc833928a1c61685b32143f6) adds the driver component itself

@ssumpf I would be nice if you could give the `imx8mp_qcacld2_wifi.run` run-script a spin as I would be interested in the *MAC-address* your Pocket uses.

ssumpf, 2024-10-01:
@cnuke: Cool! Will do.

ssumpf, 2024-10-01:
@cnuke: I get an IP-address here :+1:  The pocket uses `38:7C:76:22:64:4E` for DHCP requests.

cnuke, 2024-10-02:
@ssumpf alright, thanks for testing!

ssumpf, 2024-10-04:
@skalk: My `linux-imx-6.6-update` branch is based on `cnuke/linux-imx-6.6-update-2024-10-01` (wifi) branch and adds internal-display support for the MNT Pocket Reform. Since there are some consolidations (iomux, gpc) necessary it would make sense to me to put all updates on a fresh `linux-im-6.6-update` branch or on staging and squash any fixups.

@cnuke: When extracting _initcall_order.h_ `__initcall_hdd_module_init6` and `__initcall_ath10k_sdio_init6` are not extracted. While the later is not necessary, I guess, the former is. Is there something missing?

cnuke, 2024-10-07:
@ssumpf `__initcall_hdd_module_init6` is the proper one indeed, I'll look into that problem.

cnuke, 2024-10-11:
Commit [4d83729](https://github.com/cnuke/genode-imx/commit/4d8372977c096fa9b956d6d5bd16eb63149ac7bd) adds support for the *MNT Pocket* to the `imx8mq_sd_card` driver.

@skalk the branch also contains the changes I mentioned and the *pinctrls* currently set clock the *USDHC* devices low (I noticed you used 200mhz variant).

cnuke, 2024-12-05:
The following commits update `a64_linux` to *orange-pi-6.6-20240105-1827* that correlates to Linux *6.6.9*, the various fix-up commits are intended to ease reviewing the changes.

* [1766a8b](https://github.com/cnuke/genode-allwinner/commit/1766a8b28f0b7ae5ec4dc49c40d35c397930ad96) updates the contrib sources. I imported the corresponding branch (see commit message) and applied the GPU lima debug commit, which allows for setting the VA from the out side, on top as it is no longer available in the *orange-pi* branch. It also introduces a *R-PIO* and a new *PIO*-dummy driver (the existing *PIO*-stub driver requires `pin.cc`) to resolve the dependencies in the DT. The update also enables usage of the SLUB by default.
* [99ce0b9](https://github.com/cnuke/genode-allwinner/commit/99ce0b9effe424b03f32be325af6e0c03f483f29) updates `a64_wifi`.
* [190e6d4](https://github.com/cnuke/genode-allwinner/commit/190e6d4a1f58ebc2d0d48c14bd97b6bf0c428bff) updates `de_fb`.
* [984b8c5](https://github.com/cnuke/genode-allwinner/commit/984b8c5bf07569416114bc8edb917fe617b90415) `goodix_touch`.
* [e74b564](https://github.com/cnuke/genode-allwinner/commit/e74b564901faf05757c6c3e746527bf70deaba26) updates `a64_usb_host`.
* [ec734dd](https://github.com/cnuke/genode-allwinner/commit/ec734dd7765ac338cb80aa05ae5454bffa8a6e01) updates `a64_sd_card`.
* [33c4904](https://github.com/cnuke/genode-allwinner/commit/33c4904edce8dca7df55ac040ba4f93ea11d1875) updates `pinephone_camera`.
* [6f40da6](https://github.com/cnuke/genode-allwinner/commit/6f40da62f8c2b93e112304c3a3330159dad89420) updates `lima_gpu`.
* [d233762](https://github.com/cnuke/genode-allwinner/commit/d23376258c1dc3d15c0e871433f5ec891af5125e) updates `emac_nic`.

All drivers were briefly tested with Sculpt and/or the existing run-scripts.

nfeske, 2024-12-10:
@cnuke thanks a ton! This looks excellent. I will merge it right after we have published Sculpt 24.10.3.

nfeske, 2024-12-11:
Merged to staging.