genodelabs/genode #5282: wifi: won't work on old T-series (msi ? firmware ?)

ttcoder, 2024-07-10:
Thought it would be nice to have not only wired internet but also wireless, so I included [depot/genodelabs/pkg/pc_wifi/2024-10-05](https://depot.genode.org/genodelabs/pkg/pc_wifi/2024-05-01.tar.xz) in my H/o/G distro, but it fails on all 3 laptops I tried:

- T410
- T430
- T530

The output is always the same, with just a minor change (the version number/year of the min/max firmware varies from laptop to laptop):

[image]

I suppose two lines are of note:
- "pci_enable_msi failed"
- "no suitable firmware found"

It could make sense for me to try out wifi in SculptOS to make sure my h/o/g setup is not to blame, though the error messages do not hint at it being the culprit it seems ?
I configured `platform_drv` to set info=yes and msix=no I think (gotta check).

Anyway it just might be that my hardware is too old to have wireless support in Genode and we should close as "won't fix",,. But thought I'd give it a try since someone on the Genode team owns a T430 as I recall.

chelmuth, 2024-07-10:
The driver requests *iwlwifi-2000-5.ucode* or *iwlwifi-2000-6.ucode*. I had a look into contrib/linux-firmware-6e3c66d35626a700781b70cd45f3a1095d34f212/firmware/ but either file is missing from the port (source is https://github.com/cnuke/dde_linux_firmware).

You may copy the ucode from any linux installation by, for example, `cp /lib/firmware/iwlwifi-2000-6.ucode contrib/linux-firmware-6e3c66d35626a700781b70cd45f3a1095d34f212/firmware/`, rebuild the raw/pc_wifi_firmware depot archive, and test again.

chelmuth, 2024-07-10:
Alternatively, the ucode can be downloaded from https://wireless.wiki.kernel.org/_media/en/users/drivers/iwlwifi-2000-ucode-18.168.6.1.tgz (https://www.intel.com/content/www/us/en/support/articles/000005511/wireless.html).

ttcoder, 2024-07-10:
Good news: both the T430 and T530 work with a 'patched' depot archive. Been browsing the web with Falkon this afternoon on those laptops. Adding them up to the wired networking ones, I now have 4 computers in this house setup with Genode and Falkon, and trying to "sell" that to my family -- my world take-over plans are shaping up ^^

As to the T410, turns out it works even without the patch, actually : the failure of my initial test was actually due to a misconfiguration at my end.

Thanks much @chelmuth 

The .ucode file weighs a little more than 300KB, should it be included in a next iteration of the depot package? Or I can write some sort of script to 'patch' the pkg at build time at my end, if that's preferred.

cnuke, 2024-07-30:
@chelmuth I don't mind adding the respective firmware-images - the question is if we should do that proactively, i.e. add all that could potentially be used as not all T/X hardware dating back to X201 might share the same device - or on a case by case basis like we have done in the past and I would do now.

chelmuth, 2024-07-30:
From my perspective, we should add firmware that we ourselves or a community member tested on concrete devices (that should be mentioned in the commit message). Proactively adding kind of *any* firmware is not desirable.

cnuke, 2024-07-30:
Commit [b0b0d7e](https://github.com/cnuke/genode/commit/b0b0d7e79dff7bdc2537654483df05323a7d8c06) adds the firmware image and while there also for other devices.

commit b0b0d7e7 (2024-07-30): linux-firmware: add AX200, 9560 and T430/T530

This commit adds the firmware image for the AX200 device as found
in the Tuxedo Pulse 15 Gen1, the 9560 as found in the Starlite and
the for devices found in the T430/T530.

Fixes #5282.