genodelabs/genode #5301: libc: add support for kqueue(2)

atopia, 2024-07-18:
Add support for FreeBSD's kqueue / kevent interface.

chelmuth, 2024-09-02:
I'd like to merge the feature with commits 8fd0781aedfb8440c50d76b9d9828a5d2efd61e7 and 0777b7eeadc21db8b330d39d64128079f3fd4894. The latter is a fixup proposal addressing the following topics.

- Replacement of proactive changes regarding the (deprecated) `Plugin` interface because the compatibility with sockets is not yet in place
- Coding style in `class Kqueue`

@atopia do you acknowledge the merge?

chelmuth, 2024-09-02:
Please also consider fixup 87d371eaf4c6f6c27fc607387bd242b5f626d4a1.

chelmuth, 2024-09-03:
I cleanup the my branches https://github.com/chelmuth/genode/commits/kqueue and https://github.com/chelmuth/genode/commits/aio.

atopia, 2024-09-03:
Apart from the open question in your latest commit on the kqueue branch, the cleanup looks good to me, thanks for the cleanup work!

chelmuth, 2024-09-09:
a89b523361927b2defe944c9f553d208ddaf0112 seems to break `make run/depot_autopilot TEST_PKGS=test-libc_execve` on all platforms. @atopia please investigate.

nfeske, 2024-09-09:
Just to bring @atopia on the (hopefully) right track, `execve` is locally implemented by the libc and ldso by removing everything above the libc and loading the new program into the local PD. For this mechanism to work, the libc must not hold any pointer to an application-level allocation because those allocations are wiped by execve. From the commit, I see that the kqueue implementation dynamically allocates from the malloc heap, which is an application-level allocator, hence allocations not preserved across execve.

To solve this issue, one first need to clarify the expected semantics of the interplay between kqueue and execve. Are the allocated objects expected to survive execve or not?

Should the answer be yes - the objects must be kept - the allocations must be changed into libc-internal allocations by replacing the use of the malloc-allocator by the Kernel::_heap.
Should the answer be no - the object state should be discarded as side effect of execve - the allocations must be cleanly (and completely) reverted as part of the execve procedure before `reset_malloc_heap` is called.

nfeske, 2024-09-09:
Commit 467b8bcd952602806e6a8406990742b4d4e6022e changes the allocator used by the kqueue implementation. It fixes the execve test. Fork still works.

@atopia may you check that the preservation of the kqueue state across execve is intended behavior?

atopia, 2024-09-09:
Thanks for the clarification and the quick fix! kqueue(2) states:

> The kqueuex() system call also creates a	new kernel  event  queue,  and additionally  takes the flags argument, which is	a bitwise-inclusive OR of the following flags:
>
>       KQUEUE_CLOEXEC  The returned file descriptor is automatically closed on execve(2)
>  The `fd = kqueue()` call	is equivalent to `fd = kqueuex(0)`.

From the last bit I would argue that preserving kqueue state across execve is intended behavior for `kqueue()`.

commit 6ad0fa27 (2024-09-06): libc: implement kqueue(2)

Fixes #5301
commit a89b5233 (2024-09-06): libc: implement kqueue(2)

Fixes #5301
commit cdc45e15 (2024-10-07): libc: implement kqueue(2)

Fixes #5301
closed 2024-10-08 by chelmuth