genodelabs/genode #5302: libc: add support for AIO

atopia, 2024-07-18:
Add support for the AIO(4) asynchronous IO facility.

chelmuth, 2024-09-02:
I replaced the commit for async jobs by my initial proposal commit a6aced7c316331df33bb419b50dec6f313cacdf8 that keeps the semantic as tight as possible: onl one new API function `monitor_async()` in `Monitor::Pool`. This requires a rebase of other topic commits after addressing the upcoming review.

chelmuth, 2024-09-02:
I finished the first review - please address my inline comments. Note, I made remarks about coding style issues but only once. Please lookout for further similar errors by yourself.

chelmuth, 2024-09-03:
I cleanup the my branches https://github.com/chelmuth/genode/commits/kqueue and https://github.com/chelmuth/genode/commits/aio.

atopia, 2024-09-05:
I added fixup commits for the issues brought up during review to https://github.com/atopia/genode/tree/kqueue-aio and pushed a new branch with the commits squashed on top of your aio branch to https://github.com/atopia/genode/tree/aio-rebased

atopia, 2024-09-05:
I missed your comment regarding `Vfs_plugin::Sync` because it went to my private email address, I will add fixup commits for that in the afternoon.

atopia, 2024-09-11:
@chelmuth please give my [atopia/genode/tree/aio-rebased](aio-rebased) branch a spin

chelmuth, 2024-09-16:
Please see my new *aio* branch that includes all your changes plus some minor adjustments (libc port-hash update, indentation fixes).

Commit 4ed972c9c7797b1b1b08c447baf96ab6bba65a62 is special as it proposes to change the signatures of `async_read` and `async_write` from non-const reference parameters to result type. I suggest to move the seek-position handling into `Vfs_plugin::write()` as aio has no notion of *seek pos*, which removes the need for `first_run`.

I also noticed *aio.cc* suffers from indentation levels above 4, which could be addressed in a fixup in my opinion.

atopia, 2024-09-16:
Thanks for the little fixes!

chelmuth, 2024-09-17:
> As for the second idea: Why do you think that AIO does not have a seek pos? Quoting _aio_write(2)_:
> 
> > If O_APPEND is not  set for the file descriptor, the write operation will occur at the absolute position from the beginning of the file plus iocb->aio_offset.

The last sentence of the quoted paragraph from https://pubs.opengroup.org/onlinepubs/9799919799/functions/aio_write.html reads like follows.

> After a successful call to enqueue an asynchronous I/O operation, the value of the file offset for the file is unspecified.

Blending this with `O_APPEND` does not re-enable a seek position, but an append mode. Do you already ensure the order of operations as stated in the spec?

Unfortunately, I entirely missed the fact that the VFS API still depends on the seek position stored in the VFS handle for implementing a `pread/pwrite` like semantic. Let's discuss offline tomorrow after I had some time to ponder on this.

atopia, 2024-09-17:
Yes, I remember that the file offset is unspecified, but honestly I've just relied on the seek mechanism internally (basically thinking that just before it is unspecified after queueing an async op does not mean that I can't use it internally from within the subsystem). However - you are raising a valid point w.r.t. the order of operations. Although my understanding is that the monitor will run the Jobs in the order they were queued, for example async writes could still be interleaved if there is a partial write, then the job returns as incomplete, then the next async job on the same file runs with another partial write etc. I'll also think about what the requirements and guarantees really should be and then we can discuss offline!