genodelabs/genode #5310: hw: introduce cpu local memory area

skalk, 2024-07-30:
Nowadays, we have a fixed size array of per CPU kernel-stacks and CPU objects (aggregating e.g., scheduler, timer, idle thread,...) within the base-hw kernel. This static array approach is problematic in case of platforms with CPU counts not known at compilation time. Either we loose a lot of memory to keep space for all potential CPUs, or we don't support all of the available ones. Also the CPU numbering doesn't need to be continous necessarily, which led to a translation step in between CPU number and array access in the past on x86. Furthermore, the continous per CPU kernel-stack array had no room for a stack-boundary protection page.

Having a dedicated virtual memory area per CPU with a fixed layout including stack, cpu object data, and potential for more cpu local data is a natural step to solve above described limitations, and to open room for future optimizations if cpu-local in-kernel operation gets extended without polluting global data (and locks).

As a first step, it is planned to move kernel stacks and cpu objects  inside a dedicated virtual memory area across all architectures. Thereby, the area gets backed with memory in bootstrap after knowing the concrete CPU count.

This issue was part of former issue #1252

chelmuth, 2024-07-31:
All platforms fail to build...

```
.../repos/base-hw/src/core/kernel/cpu.cc:25:10: fatal error: hw/memory_consts.h: No such file or directory
   25 | #include <hw/memory_consts.h>
```

chelmuth, 2024-07-31:
Commit 69507865692b768a2a077991fcc21388e0fe468a fixes builds inside build directory by explicitly stating the dependency to hw/memory_consts.h. But, `./tool/depot/create UPDATE_VERSIONS=1 FORCE=1 genodelabs/bin/x86_64/base-hw-pc` still fails. Note, using `make -j1` in a clean build directory triggers the issue reliably.

skalk, 2024-07-31:
@chelmuth sorry for the hassle! Fixup commit 1835000 is still needed to build the base-hw-* packages.

chelmuth, 2024-08-01:
We're almost there. The last missing piece is *memory_consts.s* on riscv/migv, which ommit https://github.com/genodelabs/genode-riscv/commit/c5ffe94ce6a9917dbed70ae5208366db2ef67ac7 addresses. @kalk please check

skalk, 2024-08-13:
@chelmuth: yes almost, I've found another minor issue of the commit corrected by fixup commit 8b8aa43. You riscv fixup commit seems correct to me.

cproc, 2025-03-19:
The hw kernel does not start on one of my x86 test machines when hyper-threading is disabled in the BIOS setup and git bisect found commit 9258004. It works when hyper-threading is enabled.

The log output is:

```
Bender Version 0.9-beta7-51-g28ba2ad-dirty
Patching BDA with I/O port 0xf0a0.
Bender: Hello World.
Need 002e2000 bytes to relocate modules.
Relocating to 7fd1e000: 
Copying 3019640 bytes...

```

The `kernel initialized` message is missing.

This is the NOVA log output with CPU info:

```
Bender Version 0.9-beta7-51-g28ba2ad-dirty                                                                                                                   
Patching BDA with I/O port 0xf0a0.                                                                                                                           
Bender: Hello World.                                                                                                                                         
hwp config: eeo=na irq=0 hwp=1 epp=0x00 epb=0x0                                                                                                              
Need 0030b000 bytes to relocate modules.                                                                                                                     
Relocating to 7fcf5000:                                                                                                                                      
Copying 3007288 bytes...                                                                                                                                     
Copying 178192 bytes...                                                                                                                                      
                                                                                                                                                             
                                                                                                                                                             
NOVA Microhypervisor v9-a19cd2a (x86_64): [gcc 12.3.0] [MBI]                                                                                                 
                                                                                                                                                             
[ 0] TSC:2112000 kHz BUS:0 kHz DL                                                                                                                            
[ 0] CORE:00:00:0 6:8e:a:7 [f4] Intel(R) Core(TM) i7-8650U CPU @ 1.90GHz MWAIT+E+I                                                                           
[ 1] CORE:00:01:0 6:8e:a:7 [f4] Intel(R) Core(TM) i7-8650U CPU @ 1.90GHz MWAIT+E+I                                                                           
[ 2] CORE:00:02:0 6:8e:a:7 [f4] Intel(R) Core(TM) i7-8650U CPU @ 1.90GHz MWAIT+E+I                                                                           
[ 3] CORE:00:03:0 6:8e:a:7 [f4] Intel(R) Core(TM) i7-8650U CPU @ 1.90GHz MWAIT+E+I                                                                           
                                                                                                                                                             
Hypervisor NOVA (API v9)                                                                                                                                     
Hypervisor features IOMMU                                                                                                                                    
Hypervisor features VMX                                                                                                                                      
Hypervisor reports 4x1 CPUs                                                                                                                                  
mapping: affinity space -> kernel cpu id - package:core:thread                                                                                               
 remap (0x0) -> 0 - 0:0:0 boot cpu                                                                                                                           
 remap (1x0) -> 1 - 0:1:0                                                                                                                                    
 remap (2x0) -> 2 - 0:2:0                                                                                                                                    
 remap (3x0) -> 3 - 0:3:0
```

chelmuth, 2025-03-19:
This could be related to https://github.com/genodelabs/genode/issues/5215#issuecomment-2611916033 ff. May you have a look into */proc/cpuinfo* or run NOVA patched like follows?

```diff
+++ b/src/acpi_madt.cpp
@@ -54,6 +54,7 @@ void Acpi_table_madt::parse_lapic (Acpi_apic const *ptr)
     Acpi_lapic const *p = static_cast<Acpi_lapic const *>(ptr);
 
     if (p->flags & 1 && Cpu::online < NUM_CPU) {
+        trace(0, "parse_lapic: cpu=%u acpi_id=%u apic_id=%u", Cpu::online, p->acpi_id, p->apic_id);
         Lapic::id   [Cpu::online  ] = p->apic_id;
         Cpu::apic_x2[Cpu::online  ] = false;
         Cpu::acpi_id[Cpu::online++] = p->acpi_id;
@@ -65,6 +66,7 @@ void Acpi_table_madt::parse_x2apic (Acpi_apic const *ptr)
     Acpi_x2apic const *p = static_cast<Acpi_x2apic const *>(ptr);
 
     if (p->flags & 1 && Cpu::online < NUM_CPU) {
+        trace(0, "parse_x2apic: cpu=%u acpi_id=%u apic_x2=%u", Cpu::online, p->uid, p->id);
         Lapic::id   [Cpu::online  ] = p->id;
         Cpu::apic_x2[Cpu::online  ] = true;
         Cpu::acpi_id[Cpu::online++] = p->uid;
```

chelmuth, 2025-03-19:
https://github.com/genodelabs/genode/issues/5468 is the sparse APIC ID issue.