genodelabs/genode #5311: tresor: add readme file

m-stein, 2024-07-30:
The Tresor library is a non-trivial implementation of a likewise non-trivial system. Nonetheless the documentation around it consists of a conglomerate of partly out-dated articles and release notes. Therefore, I'd add a README file that bundles the most essential information and that is kept up-to-date with the progression of the source code.

m-stein, 2024-07-30:
Fixed by 092c850c97160d2e7d39b64dd14e4e05bd087cbf.

nfeske, 2024-07-30:
Wow, the volume of information alone is super impressive. Thank you so much for conserving all these valuable insights in such a palatable way.

m-stein, 2024-07-31:
Thanks for having reviewed and merged it so quickly! :)
I hope it will facilitate the future work with the project.

commit 3d3fd918 (2024-07-30): tresor: add readme file

Fix #5311