genodelabs/genode #5313: intel_fb may fail to initialize external displays with Multi Stream Transport (MST) support

atopia, 2024-08-01:
When using a display with support for Multi Stream Transport (MST) such as a Dell U2723QE, intel_fb may fail to initialize the display with an ENOSPC error:
```
[runtime -> intel_fb]       DP-5:  enable name='3840x2160' id=1  mode=3840x2160@60  fb=3840x2160 - failed, error=-28
```

In the working case, the driver reports something like the following:
```
[runtime -> intel_fb]       DP-3:  enable name='3840x2160' id=1  mode=3840x2160@60  fb=3840x2160
```

In case of the Dell U2723QE, disabling MST in the display fixes the error.

Anecdotally, disabling the GPU driver for the graphics card or bumping the quota of `intel_fb_drv` and `intel_gpu_drv` does not resolve the issue.

alex-ab, 2024-08-28:
I can reproduce it with intel_fb.run on our 6.1.20 dde_linux based port on U4711 notebook (Sculpt 24.04). The -ENOSPC originates from the [6.1.20 linux](https://elixir.bootlin.com/linux/v6.1.20/source/drivers/gpu/drm/display/drm_dp_mst_topology.c#L5145) contrib sources.

With the upcoming dde_linux 6.6.47 based port I can't reproduce it by now. I will check a Sculpt based setup tomorrow.

alex-ab, 2024-08-29:
With 24.08 based master Sculpt works in principle (same quota warnings on intel_fb and intel_gpu). The same -ENOSPC error code from the same contrib source origin happens also on 6.6.47, but it is handled differently in the contrib sources. In intel_dp_mst.c the function intel_dp_mst_find_vcpi_slots_for_bpp (renamed compared to 6.1.20 from intel_dp_mst_compute_link_config) retries on error and reduces some bpp value (24->18), which the 6.1.20 does not (stays at 24) and causing the same error again again in a kind of endless loop triggered by our port.

I would vote for a won't fix, since the 6.1.20 dde_linux driver is obsolete with today, beside there is really demand by a customer. It would probably require to find/port some of the fixes on [intel_dp_msi.c](https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/log/drivers/gpu/drm/i915/display/intel_dp_mst.c?h=v6.6.47) since 6.1.20.

chelmuth, 2024-08-30:
> In intel_dp_mst.c the function intel_dp_mst_find_vcpi_slots_for_bpp (renamed compared to 6.1.20 from intel_dp_mst_compute_link_config) retries on error and reduces some bpp value (24->18) [...]

I'm curious: Does this mean some MST link gets less than 24 (resp. only 18) color bits per pixel?

> I would vote for a won't fix, since the 6.1.20 dde_linux driver is obsolete with today [...]

I agree.

alex-ab, 2024-08-30:
I added the diff of the file. The two places with the printk are added by me. In the slots>=0 check in the 6.6.47 driver is an additional drm_dp_mst_atomic_check where the return code -ENOSPC is "eaten" and then the bpp is decreased in the loop. Besides that I have no further information.

atopia, 2024-10-23:
I can't reproduce the issue on 6.6.47, closing.

atopia, 2024-10-23:
Testing discrete display support with the Dell U2723QE with MST enabled (but not in use) and the Lenovo X1 Nano Gen2 has revealed more issues with MST:
```
[core]  ROM: [0000000067bb1000,0000000067ea1bf8) pc_intel_fb
[runtime] child "intel_fb"
[runtime]   ELF binary: pc_intel_fb
[runtime -> intel_fb] --- Intel framebuffer driver started ---
[runtime -> intel_fb] SLUB: HWalign=64, Order=0-1, MinObjects=0, CPUs=1, Nodes=1
[runtime -> intel_fb] NR_IRQS: 4352, nr_irqs: 4352, preallocated irqs: 256
[runtime -> intel_fb] clocksource: dde_counter: mask: 0xffffffffffffff max_cycles: 0x1d854df40, max_idle_ns: 3526361616960 ns
[runtime -> intel_fb] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns
[runtime -> intel_fb] clocksource: Switched to clocksource dde_counter
[runtime -> intel_fb] i915 00:02.0: [drm] GT0: Incompatible option enable_guc=3 - GuC submission is N/A
[runtime -> intel_fb] disabling PPGTT to avoid GPU code paths
[runtime -> intel_fb] i915 00:02.0: [drm] VT-d active for gfx access
[runtime -> intel_fb] Error: could not read I/O port resource 0x3cc
[runtime -> intel_fb] Error: could not write I/O port resource 0x3c2
[runtime -> intel_fb] i915 00:02.0: [drm] Failed to load DMC firmware i915/adlp_dmc.bin. Disabling runtime power management.
[runtime -> intel_fb] i915 00:02.0: [drm] DMC firmware homepage: https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/i915i915 00:02.0: [drm] [ENCODER:244:DDI B/PHY B] failed to retrieve link info, disabling eD
[runtime -> intel_fb] __acpi_video_get_backlight_type -> native
[runtime -> intel_fb] [drm] Initialized i915 1.6.0 20201103 for 00:02.0 on minor 0
[runtime -> intel_fb] sched_clock: Marking stable (1070404000, 8267000)->(1080292000, -1621000)
[runtime -> intel_fb]      eDP-1:  enable name='2160x1350' id=1  mirror   mode=2160x1350@60  fb=3840x2160
[runtime -> intel_fb]       DP-3:  enable name='3840x2160' id=1  mirror   mode=3840x2160@60  fb=3840x2160
[runtime] Warning: intel_fb: RAM upgrade of Capture failed
[runtime] child "intel_fb" requests resources: ram_quota=32400K
[runtime] Warning: intel_fb: RAM upgrade of Capture failed
[runtime] child "intel_fb" requests resources: ram_quota=32400K
[runtime -> intel_fb] setup - framebuffer  - connector id=0, virtual=3840x2160, physical=3840x2160
[runtime -> intel_fb]      eDP-1:  enable name='2160x1350' id=1  mirror   mode=2160x1350@60  fb=3840x2160
[runtime -> intel_fb]       DP-3:  enable name='3840x2160' id=1  mirror   mode=3840x2160@60  fb=3840x2160
[runtime -> intel_fb]      eDP-1:  enable name='2160x1350' id=1  mirror   mode=2160x1350@60  fb=2160x1350
[runtime -> intel_fb] setup - framebuffer  - connector id=0, virtual=2160x1350, physical=3840x1350
[runtime -> intel_fb]      eDP-1:  enable name='2160x1350' id=1  mirror   mode=2160x1350@60  fb=3840x2160
[runtime -> intel_fb]       DP-5:  enable name='3840x2160' id=1  mirror   mode=3840x2160@60  fb=3840x2160
[runtime -> intel_fb] setup - framebuffer  - connector id=0, virtual=3840x2160, physical=3840x2160
[runtime -> intel_fb]      eDP-1:  enable name='2160x1350' id=1  mirror   mode=2160x1350@60  fb=3840x2160
[runtime -> intel_fb]       DP-5:  enable name='3840x2160' id=1  mirror   mode=3840x2160@60  fb=3840x2160
[runtime -> intel_fb]      eDP-1:  enable name='2160x1350' id=1  mirror   mode=2160x1350@60  fb=3840x2160
[runtime -> intel_fb]       DP-5:  enable name='3840x2160' id=1  mirror   mode=3840x2160@60  fb=3840x2160
[runtime -> intel_fb]      eDP-1:  enable name='2160x1350' id=1  mirror   mode=2160x1350@60  fb=3840x2160
[runtime -> intel_fb]       DP-5:  enable name='3840x2160' id=1  mirror   mode=3840x2160@60  fb=3840x2160
[runtime -> intel_fb]      eDP-1:  enable name='2160x1350' id=1  mirror   mode=2160x1350@60  fb=3840x2160
[runtime -> intel_fb]       DP-5:  enable name='3840x2160' id=1  mirror   mode=3840x2160@60  fb=3840x2160
[runtime -> intel_fb]      eDP-1:  enable name='2160x1350' id=1  discrete mode=2160x1350@60  fb=2160x1350
[runtime -> intel_fb] setup - framebuffer  - connector id=0, virtual=2160x1350, physical=2160x1350
[runtime -> intel_fb]       DP-5:  enable name='3840x2160' id=1  discrete mode=3840x2160@60  fb=2160x1350
[runtime -> intel_fb] setup - framebuffer  - connector id=5, virtual=3840x2160, physical=3840x2160
[runtime -> intel_fb]      eDP-1:  enable name='2160x1350' id=1  discrete mode=2160x1350@60  fb=2160x1350
[runtime -> intel_fb]       DP-5:  enable name='3840x2160' id=1  discrete mode=3840x2160@60  fb=2160x1350
[runtime -> intel_fb]      eDP-1:  enable name='2160x1350' id=1  discrete mode=2160x1350@60  fb=2160x1350
[runtime -> intel_fb]       DP-5:  enable name='3840x2160' id=1  discrete mode=3840x2160@60  fb=2160x1350
[kernel] [ 0] IOMMU:0xffffffff817ed060 FRR:0 FR:0x6 BDF:0:2:0 FI:0xa0800000 (0)
[kernel] [ 0] IOMMU:0xffffffff817ed060 FRR:0 FR:0x6 BDF:0:2:0 FI:0xa0800000 (0)
[kernel] [ 0] IOMMU:0xffffffff817ed060 FRR:0 FR:0x6 BDF:0:2:0 FI:0xa0800000 (1)
[kernel] [ 0] IOMMU:0xffffffff817ed060 FRR:0 FR:0x6 BDF:0:2:0 FI:0xa0800000 (0)
[kernel] [ 0] IOMMU:0xffffffff817ed060 FRR:0 FR:0x6 BDF:0:2:0 FI:0xa0800000 (0)
[kernel] [ 0] IOMMU:0xffffffff817ed060 FRR:0 FR:0x6 BDF:0:2:0 FI:0xa0800000 (0)
[kernel] [ 0] IOMMU:0xffffffff817ed060 FRR:0 FR:0x6 BDF:0:2:0 FI:0xa0800000 (0)
[kernel] [ 0] IOMMU:0xffffffff817ed060 FRR:0 FR:0x6 BDF:0:2:0 FI:0xa0800000 (1)
[kernel] [ 0] IOMMU:0xffffffff817ed060 FRR:0 FR:0x6 BDF:0:2:0 FI:0xa0800000 (0)
[kernel] [ 0] IOMMU:0xffffffff817ed060 - disabling fault reporting
```

atopia, 2024-11-04:
In the end, the problems with discrete display support most likely triggered coincidentally with the enabled MST setting. In any event, I just tested a number of transitions between resolutions and merged vs. discrete connectors and in my testing the driver worked perfectly fine with the Dell display with MST enabled (but again, no actual daisy-chained display connected via MST). I'll close the issue as the problem scope seems to be resolved with Sculpt 24.10.

closed 2024-11-04 by atopia