genodelabs/genode #5318: hw: generalize IPC-helping / help pager thread

skalk, 2024-08-13:
Currently, on-demand paging in base-hw can conflict with the scheduling parameters one has chosen for a concrete scenario. Due to the fact that page-faults are propagated asynchronously via signals in between a faulting thread and the pager thread, and the fact that the pager thread itself runs without any priority nor CPU quantum. That means a high priority thread with CPU quantum in case of a page-fault gets deactivated, and potentially needs to wait for a long time until its fault gets resolved.
In contrast to ordinary IPC relations, there is no equivalent helping mechanism (donation of CPU time to callee) in the paging protocol. Any optimizations of the current  scheduling algorithm like envisioned by issue #5117 will be still affected by this limitation. Thereby, first we should generalize the helping mechanism and apply it to page-fault resolution.

closed 2025-01-30 by chelmuth

jschlatow, 2025-07-08:
@skalk During my evaluation of the new hw scheduler, I stumbled across the effect that the first execution of a test component takes much longer when there is a CPU burner with a slightly higher warp value. Digging into this, I noticed that the helping mechanism for the pager was ineffective such that the test component had to wait again for the CPU burner for every page fault. As it turned out, the core-specific pager threads were still all running on the boot CPU (which disables pager helping on every CPU other than CPU 0).

Commit a4a4f92 fixes this, however, it triggers the following kernel message:

```
Kernel: Error: do not start core threads on CPU cores different than boot cpu
```

I'm sure that you thought this through when deciding to create a pager thread for each CPU, yet, I would still like to double check before adapting the condition for printing this message. Can you confirm that it is safe to suppress this message when creating the pager threads?

skalk, 2025-07-21:
@jschlatow It's good to ask in this context. Actually, it won't harm in case of the pager threads, but the warning was meaningful to prevent that: at some point in time silently an entrypoint per cpu would be spawned within core. It is not an issue at all, if any thread of a PD is deleted before the PD itself gets destroyed.
Nowadays, since issue #5256 threads are tied to their pds and no longer exist without their pd. Therefore, the whole warning is no issue anymore.