genodelabs/genode #5320: lighttpd: prevent invalid fdarray access

cnuke, 2024-08-13:
The recent update to *1.4.73* contains a new implementation for fd event handling.  On POSIX platforms the returned fd is used to directly address an array slot and it seems to expect that the fds are contiguous and capped. Our current libc implementation violates this assumption (#3568 and #3679 come to mind) and that results in invalid array access, which leads to memory corruption.

(FWIW, the WIN32 implementation addresses a similar short-coming where the array slot for a given fd is queried).

cnuke, 2024-08-13:
Commit [e4c3ba8](https://github.com/cnuke/genode/commit/e4c3ba8140c0765bad96967e7cbf045d19e84b6a) address the issue by over allocating the array and limiting the fds.

I have not checked the POSIX spec w.r.t. to the expected behaviour but would should keep this issue in mind when renovating the libc's socket API implementation.

chelmuth, 2024-08-14:
As cited in #3679, POSIX requires *all functions that open one or more file descriptors  unless specified otherwise, [to] atomically allocate the lowest numbered available (that is, not already open in the calling process) file descriptor at the time of each allocation*. Thus, lighttpd is to correct in its assumptions to some extent, but could also fail if used libraries open files and, thereby, allocate file descriptors.

commit c5de111e (2024-08-13): lighttpd: prevent invalid fdarray access

The fd event handling uses the fd to directly access the array slot and
expects the fds to be contiguous and capped.

Since the returned fds from our libc were much larger than expected,
because the libc itself consumes multiple fds when managing sockets,
using the fd in this manner leads to memory corruption.

This commit limits the maxfds to 63 and always allocates 1024 slots
in the fd-array.

Fixes #5320.
closed 2024-08-29 by chelmuth