genodelabs/genode #5325: libports: add Qt6

cproc, 2024-08-20:


cproc, 2024-08-20:
Commit d035e8c adds support for the Qt 6.6.2 `qtbase`, `qtdeclarative`, `qtshadertools` and `qtsvg` modules with adapted tests and examples.

The `tool/tool_chain_qt6` script builds the Qt6 host tools and installs them in `/usr/local/genode/tool/23.05/qt6` by default to avoid overwriting of existing Qt5 tools with the same names.

Qt6 uses CMake to build its modules and the minimum required CMake version seems to be 3.19. Ubuntu 20.04 provides version 3.16.3, which is too old, and Ubuntu 22.04 provides version 3.22.1, which works for me.

nfeske, 2024-08-21:
I ran into a problem of building the qt6 tools.

On my system, ~/src/genode is a symlink to something like /data/genode.git. When trying to build the tools, I get the following error:

```
~/src/genode$ ./tool/tool_chain_qt6 build
...
...
...
-- Configuring submodule 'qtbase'
CMake Error at qtbase/CMakeLists.txt:56 (message):
  The path "/home/no/src/genode/build/tool/qt6/23.05" contains symlinks.
  This is not supported.  Possible solutions:

      - Map directories using a transparent mechanism such as mount --bind
      - Pass the real path of the build directory to CMake, e.g. using
        cd $(realpath <path>) before invoking cmake <source_dir>.
               
Call Stack (most recent call first):
  qtbase/CMakeLists.txt:63 (qt_internal_check_if_path_has_symlinks)

-- Configuring incomplete, errors occurred!
```

cproc, 2024-08-21:
> I ran into a problem of building the qt6 tools.
> 
> On my system, ~/src/genode is a symlink to something like /data/genode.git. When trying to build the tools, I get the following error:

Commit 36e23c1 should fix the problem.

chelmuth, 2024-08-22:
@cproc could you please double-check that `MAKE += -j10` in *etc/build.conf* is effective for qt6? We had exhaustive load on the build server last night.

nfeske, 2024-08-23:
I just successfully ran the qt6.run script. :-)

The building takes a long time. It would be really nice to drop the implicit build of the qt libraries when building a qt application, taking the libraries from the depot instead. This would improve the developer experience by a fair margin.

nfeske, 2024-08-23:
To substantiate my claim above, a rebuild of app/qt6/qt_launchpad with no changes takes about 30 seconds on my machine. I think this needs addressing as soon as possible. The current situation not only drags the development-test cycle to an uncomfortable level, it also prolongs the duration of our automated build tests where the costs for the qt-wide dependency check are now repeatedly paid for each individual target, which adds up quickly.

@cproc I would very much appreciate you investigating a way to take the qt-wide dependency checks out of the loop when building a qt-using target.

nfeske, 2024-08-23:
My current interpretation of the situation is that the Genode build system assumes that the qt libs are static libraries because it cannot see any symbol file. For static libraries, it naturally invokes the library build. For the qt libraries, the symbol files are located outside the Genode repositories. As an experiment, I manually added the following two lines to the etc/build.conf

```
REPOSITORIES += $(GENODE_DIR)/depot/nfeske_test/api/qt6_base/2024-08-23
REPOSITORIES += $(GENODE_DIR)/depot/nfeske_test/api/qt6_component/2024-08-23
```
This makes the build system aware of the symbol files. With this change, the re-build time goes down to 0.5 seconds.

Of course, this experiment is not a solid solution. When driving the build of qt apps from Genode's build system, we should not incorporate API archives but import the ingredients from the Genode repository or from ports. I'll try to investigate how the Genode build system could cover the case of hosting symbol files in ports.

nfeske, 2024-08-23:
Commit https://github.com/nfeske/genode/commit/1d6d3d25aefe6f3bb5c7a832929bd2478dd35fcd decouples the build of qmake targets from the build of the qt libraries. The sketched solution is a combination of two puzzle pieces.

First, the .lib.so files for the qt libs are now explicitly generated from the corresponding abis as found in the qt port. The former version created symlinks to .lib.so files located in bin, assuming that those files exists (which is not the case when a run script imports from the depot only). Hence, those files had to be built as a precondition for linking the binary. By creating the .lib.so files directly from their respective symbol files, this link dependency disappears.

Second, in the qt app's target.mk, the LIBS no longer refers to any qt library. This prevents the build system from considering those libs are static libraries.

I applied the second change to qt_launchpad as illustration.

@cproc would you like to take the remaining consolidation steps?

1. Apply the same concept to import-qt6_cmake.mk, possibly factoring out some duplication as both files look quite similar.
2. Removing the qt libs from each target's LIBS variable and testing the result.
3. Check that targets can still be built as depot archives. I have not tested this scenario and assume that the `abi_symbol_path` function must have a second version using the `select_from_repositories` in the depot case.

@chelmuth suggested that the manual include of import-qt6_qmake in the target.mk files could potentially be replaced by specifying qt6_qmake in LIBS (the same for cmake). I haven't tried this but it sounds like a nice idea.

cproc, 2024-08-26:
Commit 55857b6 enhances https://github.com/nfeske/genode/commit/1d6d3d25aefe6f3bb5c7a832929bd2478dd35fcd according to the suggestions above except for specifying qt6_qmake in LIBS, which made the build system complain about a missing qt6_qmake.mk.

nfeske, 2024-08-26:
Thank you @cproc for the immediate application of the sketched solution. Your changes look good. It is especially nice to see so many repetitive lines vanish. I merged your fixup to staging just now. 

Regarding your remark above, have you considered supplying `qt6_qmake` and `qt6_cmake` as an empty static libs? Don't get me wrong. I'm not insisting, just inquiring the potential for further streamlining the Qt-app-developer's experience.

cproc, 2024-08-26:
Commit a69ce98 uses `qt6_cmake` and `qt6_qmake` in LIBS.

nfeske, 2024-08-26:
Very good! Merged to staging.

cproc, 2024-08-27:
Commit aa626f0 applies the Qt build improvements to qt5 and commit a21f038 removes qt6_base from LIBS in qt6_declarative.mk and applies the qt5 build changes related to Falkon and other yet to be updated apps to qt6.

cproc, 2024-08-27:
> Shouldn't `qt6_base` also be removed from `LIBS` in _lib/mk/qt6_svg.mk_ and _lib/mk/qt6_shadertools.mk_ ?

Yes, commit c0357f6 removes it from theses files as well.

chelmuth, 2024-08-27:
> Yes, commit [c0357f6](https://github.com/genodelabs/genode/commit/c0357f67cef0951ea1e04f0a5cc7dcc22ae8e105) removes it from theses files as well.

Oh great ;) Now, I have a follow-up question as no target does and never should declare `LIBS += qt6_base`: How about moving qt6_base.mk to *src/qt6_base/target.mk* to prevent future misuse?

nfeske, 2024-08-27:
> How about moving qt6_base.mk to src/qt6_base/target.mk to prevent future misuse?

I agree that it would be good to remove this potential stumbling block, unless there exists a case where a target deliberately wants to have this dependency. But I cannot see such a case.

For covering the other qt6 libs in the same fashion, we could use the convention src/qt6/base/target.mk, src/qt6/declarative/target.mk, src/qt6/svg/target.mk, ... This way, all qt6 libs could be built via `make qt6` whereas an individual part can be build via `make qt6/svg`.

cproc, 2024-08-28:
Commits c66cddd and 6db851d move the library mk files to `src/qt6` and `src/qt5`.

cproc, 2024-08-29:
Commits 611fba8 and d0b7f27 filter out the `base-$(KERNEL)` include path which gets automatically added to the `INCLUDES` variable when building with `KERNEL=...` and which caused time-consuming Qt6 rebuilds during the nightly tests.

chelmuth, 2024-08-29:
@cproc I'm not going to merge these commits today, as tomorrow is release day. Please provide both changes as stand-alone commits on top of current staging.

Could you also provide some background, why its crucial to filter `-I` include paths to prevent rebuilds? I can't imagine that any Qt object depends on headers in these paths.

cproc, 2024-08-29:
Commits 2cf0b92 and 22ac725 are stand-alone versions on top of current staging.

> Could you also provide some background, why its crucial to filter `-I` include paths to prevent rebuilds? I can't imagine that any Qt object depends on headers in these paths.

My guess is that CMake detects the change in compiler flags and that this triggers the rebuild. When I compare the content of the build directories after switching kernels, I see differences in `CMakeCache.txt`, `flags.make` and `link.txt` files. With the filter commits these differences vanish and I don't get a full rebuild.

There are still some differences between the first and second `make` execution (without switching kernels), though, for example:

```
first run:

-DCMAKE_C_FLAGS_RELEASE=-O3 -DNDEBUG
-DCMAKE_C_FLAGS_RELWITHDEBINFO=-O2 -g -DNDEBUG
-DCMAKE_CXX_FLAGS_RELEASE=-O3 -DNDEBUG
-DCMAKE_CXX_FLAGS_RELWITHDEBINFO=-O2 -g -DNDEBUG
-DCMAKE_CXX_STANDARD=17

second run:

-DCMAKE_C_FLAGS_RELEASE=-DNDEBUG -O2
-DCMAKE_C_FLAGS_RELWITHDEBINFO=-g -DNDEBUG -O2
-DCMAKE_CXX_FLAGS_RELEASE=-DNDEBUG -O2
-DCMAKE_CXX_FLAGS_RELWITHDEBINFO=-g -DNDEBUG -O2
-DCMAKE_CXX_STANDARD=20
```

so it seems like CMake also uses some information from the previous build, and this seems to cause at least relinking of the libraries, but not the extensive recompiling seen when switching kernels. The build directory content comparison above was made after the second `make` run without kernel to keep it separate from this other build issue.

closed 2024-08-29 by chelmuth

chelmuth, 2024-08-29:
> ```
> -DCMAKE_CXX_FLAGS_RELEASE=-O3 -DNDEBUG
> -DCMAKE_CXX_STANDARD=17
> ...
> -DCMAKE_C_FLAGS_RELEASE=-DNDEBUG -O2
> -DCMAKE_CXX_STANDARD=20
> ```

So, the O level decreases while the c++ standard increases... no pun intended.

> so it seems like CMake also uses some information from the previous build, and this seems to cause at least relinking of the libraries, but not the extensive recompiling seen when switching kernels. [...]

Sigh, at least there is some light at the end of the tunnel.

chelmuth, 2024-10-14:
@cproc `./tool/ports/check_port_source qt6-host` fails due to `ERROR 404: Not Found.` please investigate.

chelmuth, 2024-10-14:
We may use the archive URL https://download.qt.io/archive/qt/6.6/6.6.2/single/qt-everywhere-src-6.6.2.tar.xz

cproc, 2024-10-14:
> We may use the archive URL https://download.qt.io/archive/qt/6.6/6.6.2/single/qt-everywhere-src-6.6.2.tar.xz

Commit 7cf3918 updates the URL.

commit d0c35b9d (2024-08-20): libports: add Qt6

Fixes #5325