genodelabs/genode #5326: Behavior of 'app/extract' as deployed by Genodians.org

cnuke, 2024-08-20:
The way the `extract` component is implemented and operated in the *genodians.org* appliance can lead to undesired behavior. A few days ago updating the website stopped as one of the extract operations failed:

```
[…]
[init -> genodians -> import -> fetchurl] fetch https://github.com/nfeske/genodian/archive/master.zip to /download/nfeske.zip
[init -> genodians -> import -> fetchurl] fetch https://github.com/mickenx/genodian/archive/main.zip to /download/mickenx.zip
[init -> genodians -> import -> fetchurl] fetch https://github.com/m-stein/genodian/archive/master.zip to /download/m-stein.zip
[…]
[init -> genodians -> import -> extract] extracted '/download/m-stein.zip' to '/content/m-stein/'
[init -> genodians -> import -> extract] Warning: reading from archive /download/mickenx.zip failed
[init -> genodians -> import -> extract] Error: Uncaught exception of type 'Genode::Exception'
[init -> genodians -> import -> extract] Warning: abort called - thread: ep
[…]
```

Since the `extract` component aborted the managing `sequence` component dwelt in the extract step as it only reacts upon a child exiting.

nfeske, 2024-08-20:
Thanks @cnuke addressing this problem. Adding those new options looks sensible to me.

Implementation-wise, I think the bad smell comes from the use of exceptions. It would be very nice to follow up this work with the replacement of exceptions by the `Attempt` pattern. What do you think?

cnuke, 2024-08-23:
@nfeske thanks for your suggestions, the following commits were adapted accordingly

* [f19ced8](https://github.com/cnuke/genode/commit/f19ced8b58504dd0581cc5b875007760260f07e6) adds the README
* [129d3ba](https://github.com/cnuke/genode/commit/129d3bad1711b4a2894d17064239acd31ddf7857) adds both options to influence the batch-processing
* [cd39e3c](https://github.com/cnuke/genode/commit/cd39e3cfb8c08c04c417f0095638b2cf8491a433) picks up your remake about the exceptions and replaces them with the `Attempt` pattern.

nfeske, 2024-08-23:
Beautiful work! Thank you very much. The README is the icing on the cake. :) Swiftly merged to staging.

nfeske, 2024-08-29:
Merged in master.

closed 2024-08-29 by nfeske