genodelabs/genode #5336: libc: pthread_once memory leak

ssumpf, 2024-08-30:
Currently `pthread_once` dynamically allocates a `pthread_mutex` per `pthread_once_t`.  Because this mutex opaque to the `pthread_once` caller it is not freed upon release of `pthread_once_t`.

ssumpf, 2024-08-30:
Related to https://github.com/genodelabs/genode/issues/5224

ssumpf, 2024-08-30:
6f57f7f is a possible improvement.

chelmuth, 2024-09-03:
@ssumpf @cproc commit c881828a088e757caadcbb090603112482a7665c breaks *run/gdb*. Please investigate.

ssumpf, 2024-09-03:
> @ssumpf @cproc commit [c881828](https://github.com/genodelabs/genode/commit/c881828a088e757caadcbb090603112482a7665c) breaks _run/gdb_. Please investigate.

Yes

ssumpf, 2024-09-03:
@chelmuth, @cproc:
I see:
```
info mem
Using memory regions provided by the target.
Num Enb Low Addr           High Addr          Attrs 
0   y  	0x0000000000001000 0x0000000000002000 rw nocache 
1   y  	0x0000000000002000 0x0000000000004000 rw nocache 
2   y  	0x0000000000004000 0x0000000000006000 rw nocache 
3   y  	0x0000000000008000 0x0000000000010000 rw nocache 
4   y  	0x0000000000010000 0x0000000000012000 rw nocache 
5   y  	0x0000000000030000 0x0000000000100000 rw nocache 
6   y  	0x0000000000100000 0x000000000014c000 rw nocache 
7   y  	0x0000000001000000 0x0000000001002000 rw nocache 
8   y  	0x0000000001002000 0x0000000001003000 rw nocache 
9   y  	0x00000000400fb000 0x00000000400ff000 rw nocache 
10  y  	0x00000000401ef000 0x00000000401ff000 rw nocache 
11  y  	0x00000000402f7000 0x00000000402ff000 rw nocache 
12  y  	0x00000000403fd000 0x00000000403ff000 rw nocache
 (gdb)*** Error: memory map is not as expected
```
Is this what is observed nightly? Or something else? It occurs with and without c881828 here. Please provide more detail.

ssumpf, 2024-09-03:
@chelmuth, @cpoc: Nevermind, I think I know what you mean and can reproduce it on hardware.

ssumpf, 2024-09-03:
@chelmuth: 487d4e5 restores the old semantic and frees the allocated mutex (using refcounters). It can be squashed into c881828 leaving its commit message intact. Tested with gdb and Mesa gears (no leaks there).

chelmuth, 2024-09-04:
Could you please elaborate why you stepped back from your former idea of using a single recursive mutex? Does GDB violate your assumptions?

ssumpf, 2024-09-04:
@chelmuth: With GDB the case described in the comment applies. Pthread.2 does not return from `init_call` and Pthread.3 never gets the global lock. Hence, I went back to one lock per pthread_once-context.

chelmuth, 2024-09-05:
The dynamic allocation still bothered me. Therefore, I implemented a stress test in commit 4d0e7d9c4db33ae67ec11d2f7605238a67ae9356 and a pthread_once alternative in 1ac6f313fcb61ccdc1cc2827af22a1f770672b40 that is inspired by our CXX guard. @ssumpf may you give mesa a spin?

ssumpf, 2024-09-05:
> The dynamic allocation still bothered me. Therefore, I implemented a stress test in commit [4d0e7d9](https://github.com/genodelabs/genode/commit/4d0e7d9c4db33ae67ec11d2f7605238a67ae9356) and a pthread_once alternative in [1ac6f31](https://github.com/genodelabs/genode/commit/1ac6f313fcb61ccdc1cc2827af22a1f770672b40) that is inspired by our CXX guard. @ssumpf may you give mesa a spin?

Looks awesome and works with Mesa :+1:

chelmuth, 2024-09-06:
Yay, the adaption reproducibly breaks libc resolv... any volunteers who like to look into the resolv-code abyss?

chelmuth, 2024-09-06:
Commit 644cd5bc81f098873d4d6a24365fc2433096a061 ensures that C symbol `_pthread_once` is aliased to `pthread_once`. Otherwise, the weak implementation from *libc/gen/_pthread_stubs.c* is used, which returns `ENOSYS` and results in `EAI_FAIL` in `getaddrinfo()`.

commit 96b6662f (2024-09-05): libc: remove dynamic alloc from pthread_once

Fixes #5336
commit 06e43017 (2024-09-05): libc: remove dynamic alloc from pthread_once

Fixes #5336
commit ace79ac6 (2024-09-06): libc: remove dynamic alloc from pthread_once

Fixes #5336
commit 899893cd (2024-10-07): libc: remove dynamic alloc from pthread_once

Fixes #5336
closed 2024-10-08 by chelmuth