genodelabs/genode #5337: libc: basic support for SIGALRM

nfeske, 2024-08-30:
Some POSIX applications like netperf use the POSIX `alarm` mechanism, which schedules a SIGALRM signal after a specified amount of seconds. To allow using netperf w/o patches, Genode's libc should emulate that.

nfeske, 2024-08-30:
Commit https://github.com/nfeske/genode/commit/05269d5fa0d5b783d80e26c4695595303b6c4224 implements the mechanism accompanied with a simple test.

@alex-ab would you like to try applying it to netperf? This could have a nice effect on the nightly tests.

alex-ab, 2024-09-02:
@nfeske: I removed all the patches and the setitimer emulation from netperf and gave it a short test on the nightly x86_64 test machine. Looks good so far.

nfeske, 2024-10-08:
Fixed in master.

closed 2024-10-08 by nfeske