genodelabs/genode #5338: Using packet stream in multi-core scenarios may get stuck on x86

chelmuth, 2024-09-02:
Even the most simple packet-stream-based client-server scenario may get stuck in corner cases if two cores observe conflicting values of `Packet_descriptor_queue::_head` and `_tail`. In this case a communication partner may skip to send a wakeup signal, because, for example, `Packet_descriptor_queue::single_element()`  returns wrongly return false. Inserting an `mfence` instruction before reading `_head/_tail` renders the test functional, but I'm unsure if this is the correct solution though.

Note, the test machine comes with an Intel Tigerlake CPU.

cproc, 2024-09-09:
> May we need something like [Linux `wmb()`](https://elixir.bootlin.com/linux/v6.6.48/source/arch/x86/include/asm/barrier.h#L14) when changing the meta data and before sending wakeups? Any ideas are much appreciated.

Or maybe something like in commit 7db7c86?

Reference: https://gcc.gnu.org/onlinedocs/gcc/_005f_005fatomic-Builtins.html

chelmuth, 2024-09-09:
> Or maybe something like in commit [7db7c86](https://github.com/genodelabs/genode/commit/7db7c86342103ef7ca7d2858640b5998efc3b945)?
> 
> Reference: https://gcc.gnu.org/onlinedocs/gcc/_005f_005fatomic-Builtins.html

This looks promising. I tested a minimal compilation unit and it boils down to the use of `xchg` for storing the value and nothing special in the read case. I'd be interested in @udosteinberg's stance on this.

udosteinberg, 2024-09-09:
In a classic producer/consumer scenario, the producer needs to use a release-store and the consumer needs to use an acquire-load for the head/tail pointers. Current NOVA has an `Atomic` class which encapsulates all this nicely.

chelmuth, 2024-09-10:
@udosteinberg thanks for guidance, but now I'm even more confused. I tested your suggestions like follows.

*atomic.cc*
```c++
static unsigned _var;

extern "C" void store_relaxed(unsigned v) { __atomic_store (&_var, &v, __ATOMIC_RELAXED); }
extern "C" void store_seq_cst(unsigned v) { __atomic_store (&_var, &v, __ATOMIC_SEQ_CST); }
extern "C" void store_release(unsigned v) { __atomic_store (&_var, &v, __ATOMIC_RELEASE); }

extern "C" unsigned load_relaxed() { unsigned v; __atomic_load (&_var, &v, __ATOMIC_RELAXED); return v; }
extern "C" unsigned load_seq_cst() { unsigned v; __atomic_load (&_var, &v, __ATOMIC_SEQ_CST); return v; }
extern "C" unsigned load_acquire() { unsigned v; __atomic_load (&_var, &v, __ATOMIC_ACQUIRE); return v; }
```

*Build*
```
> genode-x86-gcc -v |& tail -1
gcc version 12.3.0 (GCC) 

> genode-x86-gcc -m64 -O2 -o atomic.o -c atomic.cc  && genode-x86-objdump -ldSC atomic.o

atomic.o:     file format elf64-x86-64

Disassembly of section .text:

0000000000000000 <store_relaxed>:
store_relaxed():
   0:	89 3d 00 00 00 00    	mov    %edi,0x0(%rip)        # 6 <store_relaxed+0x6>
   6:	c3                   	ret
   7:	66 0f 1f 84 00 00 00 	nopw   0x0(%rax,%rax,1)
   e:	00 00 

0000000000000010 <store_seq_cst>:
store_seq_cst():
  10:	87 3d 00 00 00 00    	xchg   %edi,0x0(%rip)        # 16 <store_seq_cst+0x6>
  16:	c3                   	ret
  17:	66 0f 1f 84 00 00 00 	nopw   0x0(%rax,%rax,1)
  1e:	00 00 

0000000000000020 <store_release>:
store_release():
  20:	89 3d 00 00 00 00    	mov    %edi,0x0(%rip)        # 26 <store_release+0x6>
  26:	c3                   	ret
  27:	66 0f 1f 84 00 00 00 	nopw   0x0(%rax,%rax,1)
  2e:	00 00 

0000000000000030 <load_relaxed>:
load_relaxed():
  30:	8b 05 00 00 00 00    	mov    0x0(%rip),%eax        # 36 <load_relaxed+0x6>
  36:	c3                   	ret
  37:	66 0f 1f 84 00 00 00 	nopw   0x0(%rax,%rax,1)
  3e:	00 00 

0000000000000040 <load_seq_cst>:
load_seq_cst():
  40:	8b 05 00 00 00 00    	mov    0x0(%rip),%eax        # 46 <load_seq_cst+0x6>
  46:	c3                   	ret
  47:	66 0f 1f 84 00 00 00 	nopw   0x0(%rax,%rax,1)
  4e:	00 00 

0000000000000050 <load_acquire>:
load_acquire():
  50:	8b 05 00 00 00 00    	mov    0x0(%rip),%eax        # 56 <load_acquire+0x6>
  56:	c3                   	ret
```

From all these cases only `store_seq_cst()` seems to generate *special* code, while the suggested `store_release()` and `load_acquire()` are just `mov` instructions. So, am I wrong assuming that the generated assembly in our cases would not differ between the current plain C++ and a variant which uses an `Atomic` class?

udosteinberg, 2024-09-10:
There are many different things to keep in mind here. An x86 CPU guarantees the following (Intel SDM Vol3. 9.2.2):
* Reads are not reordered with other reads.
* Writes are not reordered with older reads.
* Writes to memory are not reordered with other writes, with the following exceptions [...]
* Reads may be reordered with older writes to different locations but not with older writes to the same location.

So it's not surprising that the compiler emits plain `MOV` instructions on x86. The story would be different on Arm.

But at an entirely different level, the C++ compiler can apply the "as-if" rule (https://en.cppreference.com/w/cpp/language/as_if), which could result in any of the following:
* load tearing (e.g., instead of loading a word atomically, load the upper and lower half separately or byte-by-byte)
* store tearing (e.g., instead of storing a word atomically, store the upper and lower half separately or byte-by-byte)
* load reordering
* store reordering

All of these would maintain proper semantics on a uniprocessor, but can break producer-consumer code across codes. Any of these would be visible in the compiler-generated output (e.g. via `objdump`) but just because the compiler emits code without reordering today doesn't mean it won't do that tomorrow.

So not only do you need to convince the CPU not to reorder or tear loads/stores, you also need to convince the compiler to emit load/store instructions in the order that producer/consumer code needs them.