genodelabs/genode #5339: intel/display: support panel self refresh

alex-ab, 2024-09-09:
Panel Self Refresh is feature for lowering device power by letting the display panel refresh independently, when the screen contents stay static.
[Linux kernel description](https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/drivers/gpu/drm/i915/display/intel_psr.c?h=v6.6.47#n42)
[Phoronix feature stories](https://www.phoronix.com/search/Panel+Self+Refresh)

alex-ab, 2024-09-09:
The first commit enables the support, but is not so effective since it ever marks the whole framebuffer dirty. With the second commit, solely if according to the capture session something changed, the framebuffer is marked as dirty, which then may trigger the wakeup of the panel. In principal the commits can also be squashed, I personally would prefer them to stay separated.

nfeske, 2024-09-10:
Very nice to see the direction the code is taking, as you just mentioned to me in person. Merged to staging.

commit b9c61f1c (2024-09-10): intel/display: enable panel self refresh support

Fixes #5339
commit 0414e492 (2024-10-07): intel/display: enable panel self refresh support

Fixes #5339
closed 2024-10-08 by chelmuth