genodelabs/genode #5340: wm: account caps/ram use to clients

nfeske, 2024-09-09:
Since introduced in https://genode.org/documentation/release-notes/14.08#New_GUI_architecture, the window manager (wm) acts as a transparent wrapper of the interface of the Nitpicker GUI server. It thereby transparently adds the notion of "windows" without tainting the Nitpicker GUI server with additional complexity. However, as the wm's original implementation was primarily focused on functionality and validating the concept of splitting the decorator and layouter into distinct components, it did not attempt to manage resources as rigidly as the nitpicker GUI server.

By now, the wm has become omni-present for Sculpt users. As resource multiplexer, it deserves the same rigidity regarding resource accounting as the nitpicker GUI server. Right now, the wm pays the overhead of wrapping nitpicker's GUI session interface out of its own pocket. Hence, we usually equip it with "sufficient" resources up front. However large those resource assignment are, however, the presence of unexpectedly many GUI clients can still result in the depletion of these resources, prompting the wm to issue a resource request.

The goal of this issue is making the wm resilient against resource exhaustion by accounting all used caps and RAM allocations to the respective client.

cproc, 2024-09-27:
Commit 4c50dd5 adapts the Qt run scripts.

cproc, 2024-10-07:
Commit c1d3072 increases the caps in the Qt textedit run scripts. The resource request occured when opening multiple menus.

cproc, 2024-10-07:
Commit 6780960 increases the caps for Qt textedit in the `tiled_wm` run scripts.

nfeske, 2024-10-08:
Fixed in master.

closed 2024-10-08 by nfeske