genodelabs/genode #5341: driver/wifi: rework CTRL interface manager

cnuke, 2024-09-17:
The following commits streamline the interaction between our `Wifi::Manager` and the `wpa_supplicant` using the CTRL interface.

cnuke, 2024-09-17:
* [1f43930](https://github.com/cnuke/genode/commit/1f43930a0859db674d107b0b2b9c8edd9fdf035e) converts the errno values within the netlink message so that the supplicant take proper action (instead of being somewhat confused because Linux doas not match FreeBSD)
* [8413b8d](https://github.com/cnuke/genode/commit/8413b8dc0ac6201bc76c0aefcd4b9bd520ff44e8) update the `wpa_supplicant` to the latest release
* [a05a277](https://github.com/cnuke/genode/commit/a05a277a3d04552828e1aba43f2723cfbbfa4477) contains the major rework
* [120f78a](https://github.com/cnuke/genode/commit/120f78a72f14c97dc781c89e4f355adeffd91733) adapts the `sculpt_manager` to the changed defaults.

The rework more or less behaves similar to the current version where not all anticipated betterment is yet in place. Those, as well as the updated *README* will come afterwards. Moreover the current interplay of `Action` and `Fifo` is capable of development and could also go hand in hand with making the helper functions for parsing the message part of the `Msg_buffer`. However, as those should also not required structural changes, I would schedule them for a later date.

chelmuth, 2024-09-19:
@cnuke please check nic_router_uplinks, netperf_lwip_wifi, and netperf_lxip_wifi.

cnuke, 2024-09-19:
* [bb0c50e](https://github.com/cnuke/genode/commit/bb0c50ef631c73c51011766bc2a0173bdf7ff057) deals with the special fds
* [94b7ede](https://github.com/cnuke/genode/commit/94b7ede449ac1c2d3021c7775c6569cbdf9c2833) update the *README* (the best-practice section is still MIA)
* [8359f07](https://github.com/cnuke/genode/commit/8359f075a3ca5bbbd9e6853f3784c517080ef727) adapts the network tests

cnuke, 2024-09-20:
* [783c656](https://github.com/cnuke/genode/commit/783c6567a49e07fc831eabc563ace832c236b24e) adapts the *README*
* [77e9546](https://github.com/cnuke/genode/commit/77e9546127bdc9e656184c755d8c3bf0fdf6d351) renames the file manager file
* [eae53ba](https://github.com/cnuke/genode/commit/eae53ba45d2902fe4b89f5776378b5f87778c885) removes `connect_scan`
* [163177d](https://github.com/cnuke/genode/commit/163177d0bcce828ee830a2d5d5534ab7d982856f) adapts the network tests

chelmuth, 2024-09-20:
Thanks, merged all to staging. @cnuke please check if dba7b7dc30d159f4595b32ab809e7d72cb424b8c is okay.

closed 2024-10-08 by chelmuth