genodelabs/genode #5342: nova.run failing on UPX-EDGEI7-A10-1664-F01 and PcEngines

trimpim, 2024-09-18:
While updating to Genode 24.08 I encountered a failure rate of about 40% for `nova.run` when executed on the platforms mentioned in the title.

When executed on QEMU the success rate is 100%

The log on the UPX is ad follows:
```
NOVA Microhypervisor v10-bca1aa3 (x86_64): [gcc 12.3.0] [MBI2]

[ 0] TSC:1992000 kHz BUS:0 kHz DL
[ 0] CORE:00:00:0 6:8e:c:7 [c6] Intel(R) Core(TM) i7-8665UE CPU @ 1.70GHz MWAIT+E+I
[ 4] CORE:00:00:1 6:8e:c:7 [c6] Intel(R) Core(TM) i7-8665UE CPU @ 1.70GHz MWAIT+E+I
[ 2] CORE:00:02:0 6:8e:c:7 [c6] Intel(R) Core(TM) i7-8665UE CPU @ 1.70GHz MWAIT+E+I
[ 6] CORE:00:02:1 6:8e:c:7 [c6] Intel(R) Core(TM) i7-8665UE CPU @ 1.70GHz MWAIT+E+I
[ 1] CORE:00:01:0 6:8e:c:7 [c6] Intel(R) Core(TM) i7-8665UE CPU @ 1.70GHz MWAIT+E+I
[ 5] CORE:00:01:1 6:8e:c:7 [c6] Intel(R) Core(TM) i7-8665UE CPU @ 1.70GHz MWAIT+E+I
[ 7] CORE:00:03:1 6:8e:c:7 [c6] Intel(R) Core(TM) i7-8665UE CPU @ 1.70GHz MWAIT+E+I
[ 3] CORE:00:03:0 6:8e:c:7 [c6] Intel(R) Core(TM) i7-8665UE CPU @ 1.70GHz MWAIT+E+I

Hypervisor NOVA (API v10)
Hypervisor features IOMMU
Hypervisor features VMX
Hypervisor reports 4x2 CPUs
mapping: affinity space -> kernel cpu id - package:core:thread
 remap (0x0) -> 0 - 0:0:0 boot cpu
 remap (0x1) -> 4 - 0:0:1
 remap (1x0) -> 1 - 0:1:0
 remap (1x1) -> 5 - 0:1:1
 remap (2x0) -> 2 - 0:2:0
 remap (2x1) -> 6 - 0:2:1
 remap (3x0) -> 3 - 0:3:0
 remap (3x1) -> 7 - 0:3:1
ROM modules:
 ROM: [000000007fdf3000,000000007fdf3201) catalog.xml
 ROM: [000000007fdf2000,000000007fdf2189) config
 ROM: [0000000000003000,0000000000007000) core_log
 ROM: [000000007fea4000,000000007ff145e0) init
 ROM: [000000007ff15000,000000007fffe230) ld.lib.so
 ROM: [0000000000102000,0000000000105000) platform_info
 ROM: [000000007fdf4000,000000007fea3a98) test-nova

174108K kernel memory

Genode 24.08-86-g2cc3bdcfa04 <local changes>
16080 MiB RAM and 63245 caps assigned to init
[init -> test-nova] testing base-nova platform
[ 0] sys_create_pt: Non-PD CAP (0x8eb50)
[init -> test-nova] res=4 create_pt
...
[init -> test-nova] res=4 pt_ctrl 30
[ 0] sys_pt_ctrl: Bad PT CAP (0x85df)
[init -> test-nova] res=4 pt_ctrl 31
[init -> test-nova] Error: map=0x112932 remap=0xe30c0 --> diff=0x2f872 freq_tsc=1992000 97 us
[init -> test-nova] delegation session_cap->copy1->copy2 cap<584716>->cap<584712>->cap<584720>
[init -> test-nova] detected 4x2 CPUs.
[init -> test-nova] Warning: pager: created
[init -> test-nova] main 0 0
[init -> test-nova] mapper: hello
[init -> test-nova] main 1000 1
[ 1] overmap attempt MEM - node - PD:0xffffffff81060b80->0xffffffff81060b80 SB:0x00000010 RB:0x00000011 O:0x00 A:0x7 SUB:0
[ 1] overmap attempt MEM - node - PD:0xffffffff81060b80->0xffffffff81060b80 SB:0x00000010 RB:0x00000011 O:0x00 A:0x7 SUB:0
[ 1] overmap attempt MEM - node - PD:0xffffffff81060b80->0xffffffff81060b80 SB:0x00000010 RB:0x00000011 O:0x00 A:0x7 SUB:0
[ 1] overmap attempt MEM - node - PD:0xffffffff81060b80->0xffffffff81060b80 SB:0x00000010 RB:0x00000011 O:0x00 A:0x7 SUB:0
...
[init -> test-nova] 0x4c000000
[init -> test-nova] 0x4e000000
[init -> test-nova] still alive - done
```

On PcEngines the log is much longer, but the error looks similar:
```
[2024-09-11 06:45:09] [init -> test-nova] res=4 pt_ctrl 30
[2024-09-11 06:45:09] [ 0] sys_pt_ctrl: Bad PT CAP (0x85df)
[2024-09-11 06:45:09] [init -> test-nova] res=4 pt_ctrl 31
[2024-09-11 06:45:09] [init -> test-nova] Error: map=0x6c3ac7 remap=0x6d5466 --> diff=0x1199f freq_tsc=998134 72 us
[2024-09-11 06:45:09] [init -> test-nova] delegation session_cap->copy1->copy2 cap<584716>->cap<584712>->cap<584720>
[2024-09-11 06:45:09] [init -> test-nova] detected 4x1 CPUs.
[2024-09-11 06:45:09] [init -> test-nova] Warning: pager: created
[2024-09-11 06:45:09] [init -> test-nova] main 0 0
[2024-09-11 06:45:09] [init -> test-nova] mapper: hello
[2024-09-11 06:45:09] [init -> test-nova] main 1000 1
[2024-09-11 06:45:09] [ 1] overmap attempt MEM - node - PD:0xffffffff81063b80->0xffffffff81063b80 SB:0x00000010 RB:0x00000011 O:0x00 A:0x7 SUB:0
[2024-09-11 06:45:09] [ 1] overmap attempt MEM - node - PD:0xffffffff81063b80->0xffffffff81063b80 SB:0x00000010 RB:0x00000011 O:0x00 A:0x7 SUB:0
[2024-09-11 06:45:09] [ 1] overmap attempt MEM - node - PD:0xffffffff81063b80->0xffffffff81063b80 SB:0x00000010 RB:0x00000011 O:0x00 A:0x7 SUB:0
...
[2024-09-11 06:45:36] [init -> test-nova] 0x4a000000
[2024-09-11 06:45:36] [init -> test-nova] 0x4c000000
[2024-09-11 06:45:37] [init -> test-nova] 0x4e000000
[2024-09-11 06:45:38] [init -> test-nova] still alive - done
```

After this no further log output comes over the serial line.

chelmuth, 2024-09-18:
What's the exact model of *PcEngines*? Does it bring AMD CPUs?

@alex-ab the UPX sits on my desk and can be tested with minor preparations.

trimpim, 2024-09-18:
coreboot reports thew following:

```
AMD GX-412TC SOC
```

alex-ab, 2024-09-18:
@trimpim: Thanks for reporting.

This specific failing part of the test tries to guess, based on time measurements, whether the write combing feature of Genode dataspaces can be misused (by evil user/attacker) on the NOVA kernel. On Qemu the test is disabled, therefore it never fails. Note: the whole test component tries to behave badly, in order to see that the Genode/NOVA is robust, therefore you see a lot of complain messages during the test.

After looking a bit on the code, the loop with touching the whole memory as warm up is not correct, which may potentially explain the timely jitter during lazy paging during the actual test measurement. I added a adjusted test commit, which fix this warm up issue. Additionally, I added a reference measurement  for the non write-combined memory case to the test and make the output a bit more human readable. 

A bit more background: The test is a bad case test, to trick the kernel into mapping the write-combined dataspace as non-write-combined, which is not supported by hardware and may lead to undefined behavior, according to the CPU vendors. By measuring the time to fill the both memory regions (write-combined and the potential not write-combined), the test tries to "guess" whether the "attack" was successful. Currently a variance of 10µs is considered as a failed test.

Can you please test the commit, whether it improves the situation ? If not, either we have to increase the variance we accept or turn this whole test just in a warning instead of failure. We will see.

trimpim, 2024-09-18:
@alex-ab Thanks for the explanation.

I did test your patch on the UPX, which I have on my desk, for 30 runs, all were successful. I will test on the PcEngines in the evening, when the pipeline isn't used by anyone else.

trimpim, 2024-09-19:
@alex-ab I did 5 runs on the pcengines board, 4 of them failed.

```
[init -> test-nova] res=4 pt_ctrl 31
[init -> test-nova] memory non writecombined          8558 us
[init -> test-nova] memory     writecombined          7084 us
[init -> test-nova] memory     writecombined remapped 7147 us
[init -> test-nova] variance   writecombined tests    62 us
[init -> test-nova] Error: PAT test considered failed - time difference above 10us
[init -> test-nova] delegation session_cap->copy1->copy2 cap<584716>->cap<584712>->cap<584720>
[init -> test-nova] detected 4x1 CPUs.
[init -> test-nova] Warning: pager: created
```

chelmuth, 2024-09-19:
Two results out of 5 runs on i7-1185G7 (fuji5)

```
[init -> test-nova] memory non writecombined          1398 us
[init -> test-nova] memory     writecombined          748 us
[init -> test-nova] memory     writecombined remapped 782 us
[init -> test-nova] variance   writecombined tests    33 us
[init -> test-nova] Error: PAT test considered failed - time difference above 10us

[init -> test-nova] memory non writecombined          1397 us
[init -> test-nova] memory     writecombined          752 us
[init -> test-nova] memory     writecombined remapped 749 us
[init -> test-nova] variance   writecombined tests    3 us
```

Maybe we could increase the error limit to 100us?

alex-ab, 2024-09-19:
Thanks for your testing. I added a commit, increasing the timeout to 100us.

chelmuth, 2024-09-19:
Great, merged to staging.

commit af00f158 (2024-09-19): nova: make write-combined unit test more robust

- touch all memory before time measurements, to reduce lazy paging jitter
- diff of 100us between 2 memset runs are now considered a failure (before 10us)
- add refrence measurements of same region size which is not write-combined
- make the output easier parse able of write-combined test

Fixes #5342
commit 4554ec42 (2024-10-08): nova: make write-combined unit test more robust

- touch all memory before time measurements, to reduce lazy paging jitter
- diff of 100us between 2 memset runs are now considered a failure (before 10us)
- add refrence measurements of same region size which is not write-combined
- make the output easier parse able of write-combined test

Fixes #5342
closed 2024-10-08 by chelmuth