genodelabs/genode #5344: capture session: stop/wakeup protocol, propagate dpi info

nfeske, 2024-09-18:
To make the capture session fit to accommodate multi-monitor scenarios, the following enhancements are desired:

- When using multiple monitors, our current approach of a strictly periodic mode of operation of both the display driver (aka capture client) and the nitpicker GUI server is no longer practical. With a growing number of monitors, we'd need to increase the activity. Also, each display has an independent refresh rate after all. So there cannot be one period to fit them all. So it is time to switch from a periodic mode of operation to an event-driven approach. The GUI stack should become active only for the subset of displays where something interesting happens. As long as a static picture is displayed, there shouldn't be any interaction between the display driver and the GUI server. Hence, we need to extend the capture session with a way stop and resume the capturing depending.

- Each of the multiple monitors can have a different physical resolution (DPI). To allow GUI applications to become DPI-aware, we need to propagate this information from the display driver through the GUI server to the GUI client.

nfeske, 2024-09-18:
The commit series above implements the changes described above. Note that users of `Capture::Connection::Screen` must be (slightly) adapted to supply the `mm` attribute.

You can find the new protocol illustrated by the fb_sdl driver. I had to refactor fb_sdl such that the SDL event loop is operated by a dedicated thread while the main entrypoint is receptive to capture-wakeup signals. The new version no longer consumes CPU time when idle.

@alex-ab it would be great to enhance intel_fb by following the logic of fb_sdl.

alex-ab, 2024-09-24:
I added two commits for intel/display to support the stop/wakeup protocol. It depends on the commits of #5345.

nfeske, 2024-10-08:
Fixed in master.

closed 2024-10-08 by nfeske