genodelabs/genode #5345: intel/display: support multiple capture sessions

alex-ab, 2024-09-20:
Currently, our port of the Intel display driver supports solely to mirror one Capture session to all physical attached displays.

For advanced usage patterns, e.g. multi-head/multi-screen/multi-display, support is required, to establish for each connected display a separate Genode Capture session. Our ported driver thereby requires support to manage per discrete (not-mirrored) connector a specific framebuffer, instead of one framebuffer used across al mirrored connectors. The consequence of this will be, that the memory usage will increase significantly, depending on the attached displays and the chosen mode, because of the discrete usage.

alex-ab, 2024-09-23:
I adjusted the labeling according to the connector names and updated the README. The commit is ready for usage.

nfeske, 2024-09-23:
Super cool! Thank you @alex-ab for realizing our ideas piece by piece in such nice way.

As a minor detail, the word "discret" should be "discrete".

chelmuth, 2024-09-24:
Fantastic indeed ;-)

@alex-ab how does `force_width/height` affect the driver now? I could imagine to move the documentation into section *Non-discrete usage of connectors* and, thus, limit the scope of forced virtual resolution to the *mirrored* connectors. In the future, it could be moved into the `<merge>` node.

alex-ab, 2024-09-24:
Thanks for your input. I added two fixups, which should address all your comments. Please check.

alex-ab, 2024-09-24:
I added the support for labeling the merge node and squashed all together.

chelmuth, 2024-09-24:
With the `<merge>` node available, I vote for moving `force_width/height` from global config scope into `<merge width="..." height="...">`.

alex-ab, 2024-09-25:
@chelmuth: I added several fixups to the discussed topics.

commit 8062d4b1 (2024-09-26): intel/display: use one capture session per display

Fixes #5345
closed 2024-10-08 by chelmuth