genodelabs/genode #5351: framebuffer session: safe access to pixel/alpha/input portions of the GUI buffer

nfeske, 2024-09-25:
The memory layout of the virtual framebuffer as created a GUI server is defined as the pixel buffer, followed by the alpha channel, followed by the input mask. The latter two parts exist only when using an alpha channel. Currently, the access to those portions involves finicky pointer calculations scattered through the code base. To ease the life of GUI clients and simplify the implementation of the GUI server, we should better provide a central place with safe accessor methods.

Since the GUI-buffer layout depends on the mode and the use of an alpha channel, it would be natural to host those accessors in the `Framebuffer::Mode` type. Before we can do that, however, we have to make the `use_alpha` attribute part of the `Mode`. Currently, this attribute is propagated as distinct argument.

Accessing the pixels, alpha values, and input mask should best become possible via `Byte_range_ptr` and via the `Surface` API.

nfeske, 2024-09-25:
The commit series above implements this change.

Note that it introduces an API incompatibility because the `Framebuffer::Mode` got the new `alpha` attribute.

nfeske, 2024-09-26:
Merged to staging.

@cproc would you take care of adapting qt5 and qt6 to these changes?

cproc, 2024-09-27:
Commit 29ea16d adapts qt5 and qt6.

nfeske, 2024-09-30:
Thank you @cproc.

nfeske, 2024-10-08:
Fixed in master.

closed 2024-10-08 by nfeske