genodelabs/genode #5352: nitpicker: support panorama spanning mulitple monitors

nfeske, 2024-10-01:
So far, the nitpicker GUI server handed out the same picture to each capture client. So each client obtained a mirror the same picture. By subjecting each client to a policy defining a window within a larger panorama, a driver creating one capture session per monitor becomes able to display the larger panorama spanning the connected displays.

nfeske, 2024-10-01:
Implemented by https://github.com/genodelabs/genode/commit/21815ef62725e022047d50fb16723619211973b3

closed 2024-10-08 by chelmuth

chelmuth, 2024-10-17:
The following commits address cases of dynamic capture screen sizes in intel_fb.

664ebd0760b intel/display: clear fb on screen size change
e9502af5969 WIP nitpicker: clip affected rects on capture to screen size
87d563f38b4 WIP intel/display: clip affected rects on capture to screen size

The WIP commits are alternatives to address clipping to the actual screen size to prevent stale pixel data on screen.

commit 21815ef6 (2024-10-01): nitpicker: configurable capture policies

Capture clients used to always capture the view stack at the origin of
the coordinate system. So each capture client obtained a mirror of the
same picture. This patch allows for the placement of capture clients on
larger panorama using Genode's usual label-based policy-selection
approach. Thereby, each monitor in multi-monitor scenario can display a
different portion of the panorama.

The patch takes special care to always keep the pointer in a visible
position. The pointer cannot be moved to any area that is not captured.
Should the only capture clien [... cut]