genodelabs/genode #5353: gui session: provide mode info as XML

nfeske, 2024-10-02:
With the emerging multi-monitor support, the existing notion of a `Gui::Session::mode` does no longer suffice. A nitpicker client (e.g., the window manager or a multi-monitor-aware application) needs to know the overall panorama dimensions as well as the panorama parts captured by the driver(s). The structured and dynamic nature of this information suggests the use of XML, similar to how the platform driver hands out device information to its clients.

With this change, we can ultimately rectify an old wart of the existing interface when using the window manager: The initial request for the mode returns the screen size. Once the client's window is defined, the call returns the window size (or the requested window size). At this point, the screen dimensions can no longer be obtained. This is limitation when considering the dynamic change of screen resolutions.

The XML information returned by a new `info` dataspace will look as follows:
```
<info ...>
  <capture .../>
  ...
</info>
```

Nitpicker hands out the panorama size as `<info>` attributes and lists each capture client as `<capture>` node:
```
<info width="800" height="600">
  <capture label="intel_fb -> eDP" width="640" height="480" width_mm="400" height_mm="300"/>
  <capture label="vnc_server" xpos="160" ypos="120" width="640" height="480"/>
</info>
```
As illustrated by the example, DPI formation is provided as `capture` attributes where available.

The window manager hands out a virtual panorama that corresponds to the display the client is shown. There is one `<capture>` node that corresponds to the client's window size as requested by the window layouter. Here, the DPI information in the `<info>` node refer to the virtual panorama corresponding to used physical display. Note that for a new client that has no window yet, the `<capture>` node may be empty.
```
<info width="640" height="480" width_mm="400" height_mm=300>
  <capture width="320" height="240"/>
</info>
```

The `Connection::mode` method will always return the panorama size.
A new `Connection::captured_rect` method will return one captured rectangle that is suitable to use as the size of the client's top-level view if the client is resizeable.

nfeske, 2024-10-02:
Commit https://github.com/genodelabs/genode/commit/d3c94b65856ec3183e1d9e18e1d88303754f9e0b implements this change and adjusts all components in Genode's main repository.
Commit https://github.com/genodelabs/genode-world/commit/a7dc2db8a9410f562f005d0304e0e8e2247d4833 adjusts genode-world.

@cproc Could you again take care about adapting Qt5 and Qt6 to this change? Even though I have adapted qt_avplay, I was unable to test it. Could you give it a try? BTW, this step is the final disruptive API change before the Sculpt 24.10 release.

@cnuke I've adjusted sdl2 to the best of my knowledge. I'd appreciate you having a look at the changes are sensible. However, when I attempt to resize the window in the run/sdl2 scenario, bad things happen:
```
[init -> test-sdl2] 192 SDL error: Window surface is invalid, please call SDL_GetWindowSurface() to get a new surface
[init -> test-sdl2] Error: ID space not empty at destruction time
```
The symptom does not seem to be new. The current master branch also has this problem.

cnuke, 2024-10-04:
@nfeske alright, I am going to take a look.

nfeske, 2024-10-04:
Thank you for attending this issue so quickly.

cproc, 2024-10-07:
Commit 9bcf91a adapts Qt5 and Qt6.

chelmuth, 2024-10-07:
@cproc it seems your Qt adaptions introduce a regression with new *windows*. For testing, you may use tiled_wm and open the Panel menu.

cproc, 2024-10-07:
> @cproc it seems your Qt adaptions introduce a regression with new _windows_. For testing, you may use tiled_wm and open the Panel menu.

Commit 3ae90cd should fix the problem.

nfeske, 2024-10-08:
Fixed in master.