genodelabs/genode #5354: tool/run/image/uboot: provide dtb for FIT images

trimpim, 2024-10-02:
We are updating to a new U-BOOT version for our imx8mp_iot_gate devices, as we need the
improved support for the TPM from it.

This nes version complains about the size of the device tree being zero bytes. This is because it
no longer provides the kernel boot arguments via ATAGS, but passes them via
by injecting them in the flattened device tree..

chelmuth, 2024-11-20:
@skalk what's your stance according this issue?

skalk, 2024-11-21:
Well, to be honest I haven't tested it yet. I wonder whether it is possible to circumvent the need for DTB, like we did it with non-FIT images using the `bootm` command. As an example, you can have a look at the image run-plugin for the MNT pocket reform within the genode-imx repository. There we use the following commands:
```
bootm start $image_load_addr
bootm loados
bootm go
```
But I haven't tested yet whether this works with FIT images and with your latest u-boot as well? If yes, I would hesitate to take the additional lines into the u-boot image run-tool.

trimpim, 2024-11-22:
@chelmuth and @skalk We will have to check, how your suggested changes could be applied together with measured/secure boot.

trimpim, 2024-11-27:
@chelmuth  and @skalk unfortunately `bootm go` doesn't work for measured boot.

We are fine with keeping this change in our own repository. So you may close this issue.

skalk, 2024-12-02:
@trimpim It was not my intention to counteract your request. If this command work-around does not meet the requirements for measured boot, I'm fine with the limited changes of your patch. I'll do a test in our infrastructure, and then we can apply it to staging.

skalk, 2024-12-02:
@chelmuth: I've tested the commit, fixed comment-typo, and re-based to staging. You can take commit ea83cab over in my eyes

chelmuth, 2024-12-02:
Merged to staging.

chelmuth, 2024-12-11:
Merged to master.

closed 2024-12-11 by chelmuth