genodelabs/genode #5356: Sculpt 24.10

nfeske, 2024-10-04:
Today I have bumped the Sculpt version at staging to 24.10. So the inter-component interfaces are meant to be stable for the release. Let's enter the testing phase and discuss practical issues here.

nfeske, 2024-10-04:
On the Gen12 Framework, I encountered the following issues:

* The touchpad driver produces a page fault at 0.
* When typing at a certain rate, the most recent character does not appear on the display. I presume that the effect is related to #5347, the driver is sleeping when it shouldn't. Interestingly, I'm unable to observe the issue on the x250. 
* VirtualBox failed to start. The error message features an unholy display size. So the problem might be related to the very recent change in #5353.

nfeske, 2024-10-04:
> When typing at a certain rate, the most recent character does not appear on the display.

Solved by commit https://github.com/genodelabs/genode/commit/b092b3248972466d5fdfa1f3ae2cd99815a33ca6

nfeske, 2024-10-04:
> VirtualBox failed to start. The error message features an unholy display size.

There was indeed a problem with the initial mode setting, which is fixed by 648e31e504c397a8d7d140a88f3a798c5d17c33f. However, vbox still fails to start on the Gen12 Framework due to a virtualization-related issue. It works on the x250. @alex-ab way this be related to #5314?

nfeske, 2024-10-04:
VBox6 on Gen12 Framework:
```
[core] [init -> runtime -> vbox6 -> vbox]   0x40000000 .. 0x4fffffff: stack area[0m[0m
[core] [init -> runtime -> vbox6 -> vbox]   0x30000 .. 0x14afff: ld.lib.so[0m[0m
[core] [init -> runtime -> vbox6 -> vbox]   0x2b19000 .. 0x2b42fff: egl.lib.so[0m[0m
[core] [init -> runtime -> vbox6 -> vbox]   0x10e19000 .. 0x10ffffff: libc.lib.so[0m[0m
[core] [init -> runtime -> vbox6 -> vbox]   0x10d76000 .. 0x10e18fff: vfs.lib.so[0m[0m
[core] [init -> runtime -> vbox6 -> vbox]   0x2b43000 .. 0x2c29fff: libiconv.lib.so[0m[0m
[core] [init -> runtime -> vbox6 -> vbox]   0x10d34000 .. 0x10d75fff: libm.lib.so[0m[0m
[core] [init -> runtime -> vbox6 -> vbox]   0x2c2a000 .. 0x2cc3fff: libyuv.lib.so[0m[0m
[core] [init -> runtime -> vbox6 -> vbox]   0x2cc4000 .. 0x2d03fff: jpeg.lib.so[0m[0m
[core] [init -> runtime -> vbox6 -> vbox]   0x2d04000 .. 0x2f81fff: stdcxx.lib.so[0m[0m
[core] [init -> runtime -> vbox6 -> vbox]   0x2f82000 .. 0x4a6afff: mesa.lib.so[0m[0m
[core] [init -> runtime -> vbox6 -> vbox]   0x4a6b000 .. 0x4a9afff: expat.lib.so[0m[0m
[core] [init -> runtime -> vbox6 -> vbox]   0x4a9b000 .. 0x4b26fff: glapi.lib.so[0m[0m
[core] [init -> runtime -> vbox6 -> vbox]   0x4b27000 .. 0x4b5cfff: libdrm.lib.so[0m[0m
[core] [init -> runtime -> vbox6 -> vbox]   0x10d15000 .. 0x10d2bfff: vfs_gpu.lib.so[0m[0m
[core] [init -> runtime -> vbox6 -> vbox]   0x4b5d000 .. 0x4b73fff: zlib.lib.so[0m[0m
[core] [init -> runtime -> vbox6 -> vbox]   0x4b74000 .. 0x4bf2fff: qemu-usb.lib.so[0m[0m
[core] [init -> runtime -> vbox6 -> vbox]   0x4bf3000 .. 0x4c49fff: virtualbox6-shaderlib.lib.so[0m[0m
[core] [init -> runtime -> vbox6 -> vbox]   0x10ce2000 .. 0x10d14fff: vfs_oss.lib.so[0m[0m
[core] [init -> runtime -> vbox6 -> vbox] OSS: verbose: 0 play_enabled: 1 min_ofrag_size: 8192 max_ofrag_size: 8192 record_enabled: 1 min_ifrag_size: 8192 max_ifrag_size: 8192 min_sample_rate: 8000 max_sample_rate: 48000 [0m[0m
[core] [init -> runtime -> vbox6 -> vbox]   0x10ccd000 .. 0x10ce1fff: vfs_pipe.lib.so[0m[0m
[core] [init -> runtime -> vbox6 -> vbox] main     Log created: 2024-10-04T14:52:44.000000000Z[0m
[core] [init -> runtime -> vbox6 -> vbox] main     Process ID:  0 (0x0)[0m
[core] [init -> runtime -> vbox6 -> vbox] main     Parent PID:  -1 (0xffffffff)[0m
[core] [init -> runtime -> vbox6 -> vbox] main     Executable:  /virtualbox6[0m
[core] [init -> runtime -> vbox6 -> vbox] [34mWarning: statfs provides bogus values for '//machine.vdi' (probably a shared folder)[0m[0m
[core] [init -> runtime -> vbox6 -> vbox]   0x4c4a000 .. 0x4c76fff: VBoxSharedClipboard.so[0m[0m
[core]  -> vbox] EMT-0     52 R3     0001  4002-4002        ACPI PM1a Enable[0m
[core] [init -> runtime -> vbox6 -> vbox] EMT-0     53 R3     0001  4004-4004        ACPI PM1a Control[0m
[core] [init -> runtime -> vbox6 -> vbox] EMT-0     54 R3     0001  4008-4008        ACPI PM Timer[0m
[core] [init -> runtime -> vbox6 -> vbox] EMT-0     55 R3     0001  4020-4020        ACPI GPE0 Status[0m
[core] [init -> runtime -> vbox6 -> vbox] EMT-0     56 R3     0001  4021-4021        ACPI GPE0 Enable[0m
[core] [init -> runtime -> vbox6 -> vbox] EMT-0     57 R3     0010  4100-410f        SMBus[0m
[core] [init -> runtime -> vbox6 -> vbox] EMT-0     58 R3     0001  442e-442e        ACPI SMI[0m
[core] [init -> runtime -> vbox6 -> vbox] EMT-0     59 R3     0001  3000-3000        ACPI Debug hex[0m
[core] [init -> runtime -> vbox6 -> vbox] EMT-0     60 R3     0001  3001-3001        ACPI Debug char[0m
[core] [init -> runtime -> vbox6 -> vbox] EMT-0     61 R3     0001  4040-4040        ACPI Battery status index[0m
[core] [init -> runtime -> vbox6 -> vbox] EMT-0     62 R3     0001  4044-4044        ACPI Battery status data[0m
[core] [init -> runtime -> vbox6 -> vbox] EMT-0     63 R3     0001  4048-4048        ACPI system info index[0m
[core] [init -> runtime -> vbox6 -> vbox] EMT-0     64 R3     0001  404c-404c        ACPI system info data[0m
[core] [init -> runtime -> vbox6 -> vbox] EMT-0     65 R3     0001  4050-4050        ACPI Reset[0m
[core] [init -> runtime -> vbox6 -> vbox] EMT-0    !![0m
[core] [init -> runtime -> vbox6 -> vbox] EMT-0    !! {irqroute}[0m
[core] [init -> runtime -> vbox6 -> vbox] EMT-0    !![0m
[core] [init -> runtime -> vbox6 -> vbox] EMT-0    PCI interrupt router at: 00:01:0[0m
[core] [init -> runtime -> vbox6 -> vbox] EMT-0    PIRQA disabled[0m
[core] [init -> runtime -> vbox6 -> vbox] EMT-0    PIRQB -> IRQ0[0m
[core] [init -> runtime -> vbox6 -> vbox] EMT-0    PIRQC -> IRQ0[0m
[core] [init -> runtime -> vbox6 -> vbox] EMT-0    PIRQD -> IRQ0[0m
[core] [init -> runtime -> vbox6 -> vbox] EMT-0    !![0m
[core] [init -> runtime -> vbox6 -> vbox] EMT-0    !! {lbr}[0m
[core] [init -> runtime -> vbox6 -> vbox] EMT-0    !![0m
[core] [init -> runtime -> vbox6 -> vbox] EMT-0    HM is not enabled for this VM![0m
[core] [init -> runtime -> vbox6 -> vbox] EMT-0[0m
[core] [init -> runtime -> vbox6 -> vbox] EMT-0    !!Assertion Failed!![0m
[core] [init -> runtime -> vbox6 -> vbox] EMT-0    Expression: HMIsSvmActive(pVM)[0m
[core] [init -> runtime -> vbox6 -> vbox] EMT-0    Location  : /plain/no/genode.git/contrib/virtualbox6-a58000c856a6a838d0da22df8c144a60211b4572/src/virtualbox6/src/VBox/VMM/VMMR3/HM.cpp(3392) void hmR3InfoLbr(PVM, PCDBGFINFOHLP, const ch
[core] ar*)[0m
[core] [init -> runtime -> vbox6 -> vbox] EMT-0    Stack     :[0m
[... cut ...]
[core] [34mWarning: unresolvable exception 3, pd 'init -> runtime -> vbox6 -> vbox', thread 'EMT-0', cpu 4, ip=0x1e6c00e sp=0x40ffc300 bp=0x40ffc350 no signal handler[0m
[core] [init -> runtime -> vbox6 -> vbox] [31mError: machine could not enter running state[0m[0m
[kernel] [ 4] Killed EC:0xffffffff81e8d1c0 SC:0xffffffff81d4b040 V:0x21 CR0:0x0 CR3:0x0 CR4:0x1 (IPC Abort)
[core] [31mError: illegal READ at address 0x0 by pager_object: pd='init -> runtime -> vbox6 -> vbox' thread='ep' ip=0x10251f4[0m

```

nfeske, 2024-10-04:
> I noticed that the text cursor in the menu view (wifi passphrase, text editor in the files section) is always displayed at the left position. 

Fixed by https://github.com/genodelabs/genode/commit/72dc8592272410df5c4569c459f4a99d6c36b984

nfeske, 2024-10-04:
> However, vbox still fails to start on the Gen12 Framework due to a virtualization-related issue. It works on the x250. @alex-ab way this be related to https://github.com/genodelabs/genode/issues/5314?

False alarm: While switching between 24.04 and my test builds, I missed to copy the NOVA version matching 24.10. With the up-to-date kernel, VBox works just fine on my Gen12 Framework.

nfeske, 2024-10-07:
When starting the system shell, the following warning appears:
```
[init -> runtime -> system_shell -> /bin/bash -> 1] Warning: sigaltstack using self chosen stack is not supported - stack ptr is ignored !!!
```
As this warning almost always appears in the Sculpt log, we should either find a way to avoid it (supporting the non-supported case) or inform the user more politely, at the very least removing the abundance of exclamation marks.

cnuke, 2024-10-07:
Commits [76f8f87](https://github.com/cnuke/genode/commit/76f8f87b8366d6e262c1e78de6fb28e59dba5573) and [e3c0870](https://github.com/cnuke/genode/commit/e3c0870a6988071733685ca1b0496e35d0d0f514) address the inconvenience due to the changed defaults regarding WLAN connectivity.

ssumpf, 2024-10-07:
During testing I experienced 1x1 window sizes for the `system_shell` and `terminal` tools. An investigation showed that a mode-change signal can overwrite the initial-window configuration in case `Gui::Connection::window` returns `Undefined`.  [de46c07](https://github.com/ssumpf/genode/commit/de46c07daed5564d100cc5208db491164f4e738c) is a possible solution for this problem while [b62f260](https://github.com/ssumpf/genode/commit/b62f26075be57b881130d6a851ddfcef12df0e8f) adjusts the cap quota for the terminal tool.

cnuke, 2024-10-08:
Commit [12d97d3](https://github.com/cnuke/genode-allwinner/commit/12d97d3186781e439aa2ad90e4d4d535354f6f14) fixes the `a64_wifi` driver.

@ssumpf please make sure to use it when testing the *PinePhone*.

cnuke, 2024-10-08:
* [0b9d3c3](https://github.com/cnuke/genode/commit/0b9d3c3dccd91e1643d9d3f68476740d96b42707) increases the initial CAP quota of the `wifi` driver triggered by Alder Lake systems
* [31f9d41](https://github.com/cnuke/genode/commit/31f9d414a386678abb8a202f23b7a5992de8b82b) remove the generation of an invalid '<network>'
* [40f2b58](https://github.com/cnuke/genode/commit/40f2b58098bf546ce665211f13574c597fb2a7d4) consolidates the `wifi_config` generator snippets
* [e4d2a4d](https://github.com/cnuke/genode/commit/e4d2a4df42f0f9484cf9afac3566f3b1d1ded20b) splits the `explicit_scan` intent
* [e0874ba](https://github.com/cnuke/genode/commit/e0874baa8fbddcdecd1920f3f494ae2ee53b931d) removes the `auto_connect` - this functionality is implicitly provided by adding/removing '<network>' nodes
* [57318b5](https://github.com/cnuke/genode-imx/commit/57318b52a33e1f8fc9544d58fe281f797f8092fc) **imx** .inc
* [57318b5](https://github.com/cnuke/genode-imx/commit/57318b52a33e1f8fc9544d58fe281f797f8092fc) **imx** auto_connect
* [b057aab](https://github.com/cnuke/genode-allwinner/commit/b057aab015d2414e0c163976a1eecda606eb6b73) **allwinner** .inc
* [0a991ed](https://github.com/cnuke/genode-allwinner/commit/0a991ed73dfd200c5b5de704f6ee105ba77ef53a) **allwinner** auto_connect

jschlatow, 2024-10-10:
Commit 9f9e539 updates the goa_testbed preset. The corresponding archives are now available on my depot for x86_64 and arm_v8a.

cnuke, 2024-10-10:
Commit [af1ab18](https://github.com/cnuke/genode-world/commit/af1ab187039bbbf4ef84f0280bd0836edc5ac769) restores the intended behavior of considering the initial mode specified via the configuration in case the window size is undefined in the SDL2 library.

Note: I have not tested all SDL2 based components but just a few of my own and SDL1 probably requires the same treatment.

cnuke, 2024-10-10:
The following commits realise mouse-grabbing with SDL2 components.

* [cc97428](https://github.com/cnuke/genode/commit/cc9742883ce2188e6b407d02f986b01e487b1129) adapts the `window_layouter` configuration
* [b0e77e0](https://github.com/cnuke/genode-world/commit/b0e77e0216e71f840c007613df7409387d43c45c) adds enabling the *exclusive* input mode to SDL2 by configuring `mouse_grab` in the component's configuration

So far the feature was only tested with some SDL2 games that already handle absolute vs. relative motion events transparently, so YMMV.

jschlatow, 2024-10-10:
@nfeske I'm trying to adapt the screenshot component to the capture-session changes. I noticed that `Capture::Connection::screen_size()` returns an invalid area if the client has not created a capture buffer yet because, I believe, the method always returns the bounding box of the current capture session. The former behaviour was to return the bounding box of all capture sessions. I am particularly wondering about the use of `Rect::intersect` in `screen_size()`: The view stack bounding box is computed as the bounding box of all capture sessions, hence the below code is, imho, the same as `return bounding_box().area`. Am I missing something?

https://github.com/genodelabs/genode/blob/61c970635313d25a834512a4b239694332632b2f/repos/os/src/server/nitpicker/capture_session.h#L197-L200

cnuke, 2024-10-10:
Commit [313a105](https://github.com/cnuke/genode/commit/313a10513dbd742882b27beeea74d9913378a791) addresses the problem with the small terminal size that @ssumpf already encountered (see https://github.com/genodelabs/genode/issues/5356#issuecomment-2397592040) but employs the pattern I used with SDL2.

cnuke, 2024-10-10:
I noticed what appear to be *graphical glitches* while resizing the terminal window. Parts of the terminal content are shown at the right side and at the bottom, clearly visible when resizing quickly.

jschlatow, 2024-10-11:
> Commit [9f9e539](https://github.com/genodelabs/genode/commit/9f9e539d9cc3de4ab5b08e30ddd9fff227694a30) updates the goa_testbed preset. The corresponding archives are now available on my depot for x86_64 and arm_v8a.

I forgot removing the _drv suffix from the mesa_gpu route and therefore updated the goa_testbed archive version. The new commit is c28d3a1. This also replaces the Audio_in/out routes with Record/Play, which we've apparently missed so far.

jschlatow, 2024-10-11:
On my x260, mesa_gears depleted its RAM quota. Commit e45ab67 adjusts this.

jschlatow, 2024-10-11:
> Commit [5980225](https://github.com/cnuke/genode-world/commit/5980225f99330391afef89511c482ad140656005) restores handling of key press events, which regressed with genodelabs/genode-world#386 and thus should be treated as _WIP_.
> 
> @jschlatow could you please check if `sdl_vnc` still exhibits the problems that prompted your adaptions as I to some degree reverted them. Sorry for noticing the regression with such a delay (TBH I have not excerised SDL2 components lately).

The commit restores the unmapped keysym warning, the key input still works, though. Thus, I'll simply remove the warning from the VNC server.

By the way, I had to increase the cap quota in sdl_vnc (see commit jschlatow/genode-world@dc8163c).

cnuke, 2024-10-14:
> The commit restores the unmapped keysym warning, the key input still works, though. Thus, I'll simply remove the warning from the VNC server.

Thank you for looking into it - for now patching it sounds reasonable. When we overhaul the SDL2 port finding a different solution would be sensible.

chelmuth, 2024-10-14:
I tried to gather all commits referenced in this issue since last Tuesday. Please check if I missed something.

chelmuth, 2024-10-14:
> @chelmuth Have you considered [e45ab67](https://github.com/genodelabs/genode/commit/e45ab67f664e04fda79b9f36bc6e8581f4f18c9e)?

Missed that one, merged just now.

cnuke, 2024-10-14:
> I tried to gather all commits referenced in this issue since last Tuesday. Please check if I missed something.

You caught all my commits but apparently used the ones from my working branch rather than the specific topic branches (the content is the same but the commit contains a commit message and reference to this issue in the later case).

jschlatow, 2024-10-14:
@cproc I noticed that the jemalloc versions of Falkon and Morph browser get quickly stuck in a RAM request. This typically occurs after opening the second or third website after having played a youtube video. This does not seem to occur with the non-jemalloc versions.

Moreover, I am unable to sign into github. The "sign in" button has no effect in either version of Falkon/Morph.

jschlatow, 2024-10-14:
> * The usb_webcam, test-capture test does not produce any picture on the x250. Even though I added a device policy to /config/usb, the log shows no lifesign of the webcam driver.

Same with the x230, x260 and t490s as well as my external webcam.

cnuke, 2024-10-15:
Commit [449694a](https://github.com/cnuke/genode/commit/449694a8b56529aa17fd05e21b5442533c46d576) mitigates a performance regression with Realtek network devices.

As @alex-ab noticed the `pc_nic` driver behaves unexpectedly (the throughput is low) on his machine using a Realtek device. I could reproduce the problem running *netperf_lwip* where staging produces around 0.1(!) MBit/s while the commit of the driver before the update to 6.6.47 gets around 500 MBit/s. I diffed the old and new driver version and one change that caught my eye was the activation of IRQ coalescing. When removing the function call the driver performs as before the update. I have not investigated why enabling the feature apparently leads to performance issues though.

chelmuth, 2024-10-15:
> You caught all my commits but apparently used the ones from my working branch rather than the specific topic branches (the content is the same but the commit contains a commit message and reference to this issue in the later case).

I did not notice that you're maintaining two flavors of commits. For me, it's okay to lose the reference to this issue. Do you agree?

chelmuth, 2024-10-15:
> As @alex-ab noticed the `pc_nic` driver behaves unexpectedly (the throughput is low) on his machine using a Realtek device. I could reproduce the problem running _netperf_lwip_ where staging [...]

Could this be related to the behavior discovered in #3939? May you open a dedicated issue for this instance?

cnuke, 2024-10-15:
> > You caught all my commits but apparently used the ones from my working branch rather than the specific topic branches (the content is the same but the commit contains a commit message and reference to this issue in the later case).
> 
> I did not notice that you're maintaining two flavors of commits.

I normally create a specific topic branch (*issue#xxx…*) that contains a proper commit message and the reference to the github issue and is the commit that I cite in the issue's comment after the fact. Sometimes I cherry-pick that commit to a WIP branch and deliberately remove the reference to no pollute the github issue's history.  

> For me, it's okay to lose the reference to this issue. Do you agree?

It is fine for me.

cnuke, 2024-10-15:
> Could this be related to the behavior discovered in #3939?

It could but since I do not know the iPXE driver operates the device I can only speculate. 

> May you open a dedicated issue for this instance?

Of course.

jschlatow, 2024-10-16:
> @cproc I noticed that the jemalloc versions of Falkon and Morph browser get quickly stuck in a RAM request. This typically occurs after opening the second or third website after having played a youtube video. This does not seem to occur with the non-jemalloc versions.

I think jschlatow/genode-world@ca0760b is a good solution for letting users adapt to higher RAM requirements of the browser. Note, this does not change the default RAM quota specified by the runtime but let's the user effectively increase the quota in a launcher file.

jschlatow, 2024-10-16:
> The x250 seems to mostly work fine [...]. Wakeup from standby does not work

Wakeup does not work on my x230, x260 or t490s either.

jschlatow, 2024-10-16:
> > The x250 seems to mostly work fine [...]. Wakeup from standby does not work
> 
> Wakeup does not work on my x230, x260 or t490s either.

It does work on base-hw, though (tested on x230 and x260).

skalk, 2024-10-16:
Commit 9c32d7d addresses the record_play_mixer instead of the oldfashioned audio mixer within the sculpt default index file. Now, it happens that the old mixer is shown in the index but not available, because the new one is part of the distribution packages.

alex-ab, 2024-10-16:
> > When typing at a certain rate, the most recent character does not appear on the display.
> 
> Solved by commit [b092b32](https://github.com/genodelabs/genode/commit/b092b3248972466d5fdfa1f3ae2cd99815a33ca6)

The issue still happens to @chelmuth and me. Commit 5fd5c06bfb49a2567754321d11f5e82c69ae33f5 fixes the wakeup/sleep handling for me. @chelmuth: please consider for staging

cproc, 2024-10-16:
> I think [jschlatow/genode-world@ca0760b](https://github.com/jschlatow/genode-world/commit/ca0760b) is a good solution for letting users adapt to higher RAM requirements of the browser. Note, this [...]

Thanks, I published new browser packages with this commit applied.

jschlatow, 2024-10-17:
I gave Sculpt 24.10 with base-hw a spin on the x230 and x260 and made the following observations:

- The UI feels much faster on the x230 (i5) than the x260 (i7). The latter has a very sluggish feel, when depot download is in progress whereas the x230 still feels very responsive. Yet, this is not a regression as I can reproduce the same effect with 24.04. I suspected the display resolution (1366x768 on the x230 vs 1920x1080 on the x260) but this has no significant effect on 24.04. **On 24.10, I'm not able to change the display resolution on the x260 by editing _/config/fb_.**
- After a few minutes running 24.10 on the x230, the mouse pointer freezes. The pointer coordinates still seem to be updated, though, as I'm still seeing hover effects when moving the mouse.

cnuke, 2024-10-17:
* [665d3ff](https://github.com/cnuke/genode/commit/665d3ffcae5ac9c654b862ffc2556069488a9da6) straightens out the invalid `<network/>` entry check (as was trigger by the *PinePhone* see next commit)
* [754331e](https://github.com/cnuke/genode/commit/754331e5a691b09d74621aeb94e05190c87c8fb5) changes the default wifi config to provide the exemplary `<network/>` node via a comment.

In case someone is testing the *PinePhone*, sometimes the wifi GUI gets stuck displaying the *scanning* label although fresh scan results are available because it seems to thing the AP selection list is already hovered. Leaving and entering the *Network* section works around this problem.

alex-ab, 2024-10-18:
> > > When typing at a certain rate, the most recent character does not appear on the display.
> > 
> > 
> > Solved by commit [b092b32](https://github.com/genodelabs/genode/commit/b092b3248972466d5fdfa1f3ae2cd99815a33ca6)
> 
> The issue still happens to @chelmuth and me. Commit [5fd5c06](https://github.com/genodelabs/genode/commit/5fd5c06bfb49a2567754321d11f5e82c69ae33f5) fixes the wakeup/sleep handling for me. @chelmuth: please consider for staging

After @chelmuth reported that the issue still happens to him, I instrumented the GUI stack down to nitpicker. After some investigation, it turns out that the leitzentrale -> nitpicker sometimes does not report content changes immediately if the changes are too close together (smaller 40ms). If no further change happens, or not close in a timely manner, the output seems too "hang" and appear quite late (up to 2s seen in inspect window in Leitzentrale). Commit 13d72ba31f16224dd7b7f77399aa3dfc2ec6b115 addresses the issue and make the symptom go away for me. @chelmuth: please give it a try

alex-ab, 2024-10-18:
I published before the weekend a new sculpt test image based on staging + 37e03d1131c16010dafc03ebca63a65f456bf713 and + 13d72ba31f16224dd7b7f77399aa3dfc2ec6b115 (acked by @chelmuth). @cproc: I assume 10-16 is the current falkon version, and not the 21.10 (date 10-13).

cproc, 2024-10-18:
> @cproc: I assume 10-16 is the current falkon version, and not the 21.10 (date 10-13).

yes, 10-16 is the current version

trimpim, 2024-10-18:
On my Lenovo T14 I got the following error:

```
[runtime -> wifi] Error: Function unregister_handler_proc not implemented yet
```

This happened when I started the WiFi driver, realized that I forgot to copy the WiFi-configuration, stopped the driver, copied the config to `/config/managed` (from the 24.04 release config) and started the driver again.

This doesn't happen, if I start/stop the driver multiple times in a row when a configuration is present.

trimpim, 2024-10-18:
@cproc sure

atopia, 2024-10-21:
> On my Lenovo T14 I got the following error:
> 
> ```
> [runtime -> wifi] Error: Function unregister_handler_proc not implemented yet
> ```

I just hit the same path on 24.04 on the X1 Nano (with wifi configuration). I'm not 100% sure but I believe I had accidentally booted into native Linux on the previous boot instead of selecting the 20.04 boot entry. Before that I had sucessfully run the driver testing 24.10 (also with a wifi configuration).
 @cnuke I'll email you the corresponding log to look at after your vacation.

chelmuth, 2024-10-21:
>     * The usb_webcam, test-capture test does not produce any picture on the x250. Even though I added a device policy to /config/usb, the log shows no lifesign of the webcam driver.

I adapted usb_webcam/test-capture to the changes in capture/GUI sessions with 6cbf55916a0229be2855616171abbb5fd2a5f86a. @nfeske @jschlatow please give it a spin.

nfeske, 2024-10-21:
> If no further change happens, or not close in a timely manner, the output seems too "hang" and appear quite late (up to 2s seen in inspect window in Leitzentrale)

@alex-ab commit https://github.com/genodelabs/genode/commit/cc12816d871d92b878a0b48eeaf71321543bba63 should fix this issue. BTW, the 2s come from the log view that is updated at that interval, even when located off-screen.

nfeske, 2024-10-21:
@skalk and @chelmuth, commit https://github.com/genodelabs/genode/commit/09ed07b89ed2e71ba81228e2b7cdf6829c9b2b5a should address the left-over pixels when closing the popup view, as you reported.

cproc, 2024-10-22:
I get the following error on the PinePhone with the `gpu test` preset (and also with my local WIP morph browser update):

```
[init -> runtime -> glmark2] Warning: ignoring unsupported RTLD_GLOBAL in dlopen()
[init -> runtime -> glmark2] libEGL warning: MESA-LOADER: failed to retrieve device information
[init -> runtime -> glmark2] 
[init -> runtime -> glmark2] Error: no plugin found for fstat(10044)
[init -> runtime -> glmark2] Error: Uncaught exception of type 'Genode::Attached_dataspace::Region_conflict'
[init -> runtime -> glmark2] Warning: abort called - thread: ep
```

ssumpf, 2024-10-22:
@cproc: Could you try out [b3a416c](https://github.com/ssumpf/genode/commit/b3a416c86f99a6934578a5211b8a3172675a7417) from my staging branch and see if this fixes  3D for you too?

cproc, 2024-10-22:
> @cproc: Could you try out [b3a416c](https://github.com/ssumpf/genode/commit/b3a416c86f99a6934578a5211b8a3172675a7417) from my staging branch and see if this fixes 3D for you too?

Yes, that fixed it.

Now there's another problem with the Morph browser on the PinePhone: when I load `genodians.org` the following error occurs:

```
[init -> runtime -> morph_browser -> morph-browser] Error: LD: jump slot relocation failed for symbol: 'util_format_unpack_description_neon'
[init -> runtime -> morph_browser -> morph-browser] Error: Uncaught exception of type 'Linker::Not_found'
[init -> runtime -> morph_browser -> morph-browser] Warning: abort called - thread: QSGRenderThread.24
```

The following change in mesa made the error go away, but it's probably not the correct fix:

```
diff --git a/src/util/format/u_format.c b/src/util/format/u_format.c
index 30dd65e..0ef6972 100644
--- a/src/util/format/u_format.c
+++ b/src/util/format/u_format.c
@@ -1221,7 +1221,7 @@ static void
 util_format_unpack_table_init(void)
 {
    for (enum pipe_format format = PIPE_FORMAT_NONE; format < PIPE_FORMAT_COUNT; format++) {
-#if (DETECT_ARCH_AARCH64 || DETECT_ARCH_ARM) && !defined(NO_FORMAT_ASM) && !defined(__SOFTFP__)
+#if (DETECT_ARCH_AARCH64 || DETECT_ARCH_ARM) && !defined(NO_FORMAT_ASM) && !defined(__SOFTFP__) && 0
       const struct util_format_unpack_description *unpack = util_format_unpack_description_neon(format);
       if (unpack) {
          util_format_unpack_table[format] = unpack;
```

I can try to get a backtrace to find a simpler test case to trigger the error if that helps.

atopia, 2024-10-22:
> > On my Lenovo T14 I got the following error:
> > ```
> > [runtime -> wifi] Error: Function unregister_handler_proc not implemented yet
> > ```

It seems I can reproduce this reliably by rebooting from Linux to Genode 24.04 but when rebooting from Linux to 24.10 the error didn't occur (at least in the one test I did).

atopia, 2024-10-22:
@chelmuth please consider f1a8d67aecbb1d1c606f534caddd21987858e290, 56ffad1bc8d3e81b9b18df96693c406f2375f13f, and 90f26b1591a3f4cc9a45b036bb6719030f6196f1 for staging.

chelmuth, 2024-10-22:
> @cproc: Could you try out [b3a416c](https://github.com/ssumpf/genode/commit/b3a416c86f99a6934578a5211b8a3172675a7417) from my staging branch and see if this fixes 3D for you too?

@ssumpf I revealed the mystery behind this obviously dubious fix: libdrm contains ioctl implementations for both lima and etnaviv which both implement `Gpu::Vram::mmap()` in different `Gpu::Vram` classes... I do not know why the linker does not whine about the duplicates but these must be resolved. I suggest do move the whole ioctl implementations into `namespace Etnaviv` resp. `namespace Lima`.

chelmuth, 2024-10-22:
> I do not know why the linker does not whine about the duplicates but these must be resolved. 

It's the *one-definition rule* of C++ that bites us: If a member function is defined in the class body, the compiler must enable the linker to decide which instance of the function ends up in the final object. Therefore, these member-function symbols are marked *weak* and no linker error is emitted.

ssumpf, 2024-10-22:
> ```
> [init -> runtime -> morph_browser -> morph-browser] Error: LD: jump slot relocation failed for symbol: 'util_format_unpack_description_neon'
> [init -> runtime -> morph_browser -> morph-browser] [...]
@cproc: looks like `src/util/format/u_format_unpack_neon.c` needs to be added to _libports/src/lib/mesa/files.list_ and _libports/lib/mk/mesa.inc_ (SRC_C)

atopia, 2024-10-22:
@nfeske Please have a look at 42c61218c7 as a suggested fix for the runtime_view issue. The fix solves the issue for me (4k screen, mirrored).

cproc, 2024-10-22:
> @cproc: looks like `src/util/format/u_format_unpack_neon.c` needs to be added to _libports/src/lib/mesa/files.list_ and _libports/lib/mk/mesa.inc_ (SRC_C)

Yes, that solved the problem. Fixed by 05c692f.

cproc, 2024-10-22:
Another problem with the Morph Browser on the PinePhone is that the Qt window area doesn't end at the top of the on-screen keyboard anymore. I see a window area of 720x1440 read from the GUI info.

nfeske, 2024-10-22:
> @cnuke: I noticed what appear to be graphical glitches while resizing the terminal window. Parts of the terminal content are shown at the right side and at the bottom, clearly visible when resizing quickly.

Thanks for reporting! Fixed by 342155320c3fa6d374d704ab3cc20a227b7da4ce

[40 more replies]

closed 2024-10-30 by nfeske