genodelabs/genode #5364: nic_router: assertion in dhcp_server.cc

jschlatow, 2024-10-16:
While test-running Sculpt 24.10 (#5356), I stumbled upon the following assertion that requires a restart of the Nic router:

```
[runtime -> nic_router] Error: /data/depot/genodelabs/src/nic_router/2024-10-07/src/server/nic_router/dhcp_server.cc:204:  assertion "_ip_alloc.free(ip.to_uint32_little_endian() - _ip_first_raw)" failed
```

I was able to reproduce this by starting my Linux VM with wifi enabled, switching to wired network once the VM has started and shutting the VM down afterwards. The assertion triggers during the shutdown procedure.

I have not investigated the issue any further, yet.

chelmuth, 2024-10-16:
Sheesh, this regression may be hidden in the extensive changes of ac42ade48c9176d002017f73b127a08747b19fa8 :face_with_peeking_eye:

jschlatow, 2024-10-16:
I ruled out a double free, i.e. `Dhcp_server::free_ip()` is only called once. Apparently, the DHCP server does not know about the IP, which indicates that the router tries to free the DHCP allocation on the wrong DHCP server. By instrumentation, I found out that there is one call to `with_dhcp_server()` that silently lands in the the `no_dhcp_server_fn`. The call is made from `Interface::_update_dhcp_allocations()`.

jschlatow, 2024-10-16:
Fixed by 3a56b9c.

@chelmuth I'm also thinking about removing the variant of `with_dhcp_server()` that does not take a `no_dhcp_server_fn` or, at least, renaming it to `with_optional_dhcp_server()`.

jschlatow, 2024-10-16:
> @chelmuth I'm also thinking about removing the variant of `with_dhcp_server()` that does not take a `no_dhcp_server_fn` or, at least, renaming it to `with_optional_dhcp_server()`.

Implemented by ef34f39.

chelmuth, 2024-10-16:
> > @chelmuth I'm also thinking about removing the variant of `with_dhcp_server()` that does not take a `no_dhcp_server_fn` or, at least, renaming it to `with_optional_dhcp_server()`.
> 
> Implemented by [ef34f39](https://github.com/genodelabs/genode/commit/ef34f39ed1113e69872103af5134c3580b3c0a61).

That's quite an improvement to understand what happens, thanks. Merged and squashed your commits.

commit 6b5adfb4 (2024-10-16): nic_router: fix DHCP deallocation on domain update

Commit ac42ade introduced a regression that triggered an assertion in
`Dhcp_server::free_ip()` because the DHCP allocation was not properly
removed during a domain update. The underlying issue was that
`with_dhcp_server()` silently landed in the `no_dhcp_server_fn`.

Fixes #5364
commit d87a235a (2024-10-30): nic_router: fix DHCP deallocation on domain update

Commit ac42ade introduced a regression that triggered an assertion in
`Dhcp_server::free_ip()` because the DHCP allocation was not properly
removed during a domain update. The underlying issue was that
`with_dhcp_server()` silently landed in the `no_dhcp_server_fn`.

Fixes #5364
closed 2024-10-30 by chelmuth