genodelabs/genode #5367: sculpt: sporadic infinite resource requests

nfeske, 2024-10-21:
@ssumpf reported a situation where the "runtime -> wm" subsystem would repeatedly issue resource requests upon the startup of a slightly modified "window_manager" preset (adding a few more system-shell instances) when deployed from the ram_fs. The log would display an infinite amount of messages like this
```
[init -> runtime] child "wm" requests resources: ram_quota=16588800
```

The problem occurs as follows:
* The "wm" subsystem is an init instance that acts as a GUI server, forwarding session requests to the embedded "wm" component.
* Multiple siblings of the "wm" are started simultaneously (e.g., three instances of the system shell along with the motif_decorator), each connecting the GUI service provided by the "wm" subsystem. Open startup, each GUI client upgrades its respective GUI session with the amount of memory needed for a pixel buffer. In the case of motif_decorator, this session upgrade amounts to 16 MiB for the double-buffered output in full-HD.
* When a GUI client upgrades its session, the "wm" init observes an update of its "session_requests" ROM as provided by the "runtime" init. It goes through the entries and distributes quota upgrades to the respective children, i.e., "runtime -> wm -> wm".
* The problem occurs when (many) GUI clients issue session upgrades at a higher rate than the "wm" init examining the session-requests ROM: Since each upgrade triggers a ROM-update signal, one session-requests state may be examined multiple times. In this case, init would wrongly distribute the amount as found in the session-requests ROM to a child twice, paying out of its own pocket. If this happens often enough, the resources of the "wm" init deplete, ultimately triggering the observed resource request.

nfeske, 2024-10-21:
Commit https://github.com/genodelabs/genode/commit/57acd634b9dac44a3e35f67d2ce3c06c5351b376 should solve this problem. @ssumpf may you give it a try?

ssumpf, 2024-10-21:
> Commit [57acd63](https://github.com/genodelabs/genode/commit/57acd634b9dac44a3e35f67d2ce3c06c5351b376) should solve this problem. @ssumpf may you give it a try?

@ssumpf: Will do & almost looks like my test fix :+1:

ssumpf, 2024-10-22:
@nfeske: Tested and works as expected!

commit 57acd634 (2024-10-21): init: prevent double session-upgrade forwarding

Fixes #5367
closed 2024-10-30 by chelmuth