genodelabs/genode #5369: Sculpt OS 24.10 - PinePhone

ssumpf, 2024-10-23:
Status of the PinePhone Sculpt 24.10 update

Presets/Launcher updates:

- [x] goa_test bed (jschlatow/2024-10-11)
- [ ] gpu_test (requires fixes in drm - pending)
- [x] nano 3d
- [x] system_shell
- [x] osci
- [x] camera test
- [ ] morph browser (new version/Mesa/drm fixes)
- [ ] jitsi (new version Mesa/drm fixes)
- [ ] screenshot (nfeske/2023-03-29)
- [ ] linphone (new preset)

related to genodelabs/genode#5356

chelmuth, 2024-10-23:
> gpu_test (requires fixes in drm - pending)

Merged 8fe6f939ba2b81b61109ad33aff5fb7fcde7c9b9 to staging.

nfeske, 2024-10-23:
> screenshot (nfeske/2023-03-29)

Updated by https://github.com/genodelabs/genode-allwinner/commit/f1b8eb965e06522420f108ad4af65c8fce31a111

cproc, 2024-10-23:
Commit d345dc4 updates the morph browser and jitsi presets.

cnuke, 2024-10-28:
@ssumpf input via the `touch_keyboard` appears to be broken in the morph-browser preset also (note I had to power-cycle the PinePhone after installing the preset's packages to the SD-card as it failed to deploy the preset because there was not enough RAM left and as the screen was blanked at this point it also failed to start the framebuffer driver again). Input works in the system-shell preset though.

@cproc do the presets work for you?

cproc, 2024-10-28:
I instrumented `QGenodePlatformWindow::_input()` and it receives only the `HOVER_LEAVE` event from the input session when I tap on the touch keyboard.

nfeske, 2024-10-28:
Does the touch keyboard work for the system-shell preset?

cnuke, 2024-10-28:
@nfeske yes it does for me

cproc, 2024-10-28:
> @nfeske yes it does for me

Works here too.

nfeske, 2024-10-28:
So the touch keyboard seems to work in principle.

May Qt create multiple GUI sessions? I guess so. If yes, could the order of their session creation have changed since the last release? It might be useful to instrument nitpicker's `Gui_root::create_session` to see if there are multiple sessions labeled "runtime -> focus" arriving at nitpicker. In this case, the routing of keyboard input is a matter of luck (session-creation time). The keyboard-focus session must be unambiguously labeled. But when looking at the morph present, all GUI session opened by the Qt app are labeled as "focus". @cproc would it be possible to single out the actual keyboard-handling session from the others by using two routing rules?

ssumpf, 2024-10-28:
@cproc: I also just had a look and found something suspicious. `QGenodePlatformWindow` seems to create it's own `Input::Session_client`. Shouldn't it use `Gui::Connection::input`? I can try this out. In what library does `QGenodePlatformWindow` end up?

cproc, 2024-10-28:
> @cproc: I also just had a look and found something suspicious. `QGenodePlatformWindow` seems to create it's own `Input::Session_client`. Shouldn't it use `Gui::Connection::input`? I can try this out. In what library does `QGenodePlatformWindow` end up?

Yes, looks like `QGenodeplatformWindow::_input_session` is obsolete now and can be replaced by `_gui_connection.input`. Commit https://github.com/cproc/qt5/commit/904096cb58b6b545704c58c31a9b8b833b8fab43 added a permanent Gui connection for the detection of screen resolution changes. Maybe that one is receiving the keyboard input now. `QGenodePlatformWindow` ends up in`libqgenode.lib.so`, which is part of the `qt5_base` depot package.

ssumpf, 2024-10-28:
> Maybe that one is receiving the keyboard input now.

@cproc: I can confirm that. I labeled the `Gui::Connection` in `QGenodeScreen` as "screen" and routed it in my preset to "parent". The other sessions (it is just) one, I routed to "parent label="focus". The input works as expected in this case.

ssumpf, 2024-10-28:
With the the Qt change above I was able to successfully test the Linphone preset.

ssumpf, 2024-10-29:
[4f3c2ff](https://github.com/ssumpf/genode-allwinner/commit/4f3c2fffc69fdfad41fd7d18836ac282d7170ae2) adds the Linphone preset.

chelmuth, 2024-10-29:
I updated genode-allwinner/staging.

9eb4b835 preset/linphone: provide VoIP functionality
1433aab1 fixup "sculpt/deploy: update Morph pkg" (keyboard focus)
1ec578b0 sculpt/pinephone: version 2024-10-11 of goa_testbed

cnuke, 2024-10-29:
Commit [4d8b14e](https://github.com/cnuke/genode/commit/4d8b14ea46d0a27c5c95d1fe366b6fcbed3e3fee) could be used to mitigate the access point list hovering issue while I am still looking for a proper solution.

nfeske, 2024-10-29:
> Commit [4d8b14e](https://github.com/cnuke/genode/commit/4d8b14ea46d0a27c5c95d1fe366b6fcbed3e3fee) could be used to mitigate the access point list hovering issue while I am still looking for a proper solution.

Thank you. I think that's fine.

ssumpf, 2024-10-30:
Completed with Sculpt 24.10 release.

closed 2024-10-30 by ssumpf