genodelabs/genode #5373: tool/create_builddir: prepend board-specific repos

chelmuth, 2024-11-01:
Board-specific repositories must appear before generic repositories (i.e., gems) in 'REPOSITORIES' to select board-specific sculpt/ configuration snippets over the defaults provided by gems. When listing the board-specific repositories last, Sculpt images for the PinePhone or MNT Reform may wrongly include launchers and presets of the PC version.

Commit cb5eb740c04209d67ae09e07e7c8067833037ca0 implements this.

chelmuth, 2024-11-01:
Unfortunately, riscv no longer build on staging.

```
[2024-10-31 10:40:43] /data/genode/repos/base/src/lib/base/raw_output.cc:20:10: fatal error: base/internal/raw_write_string.h: No such file or directory
[2024-10-31 10:40:43]    20 | #include <base/internal/raw_write_string.h>
[2024-10-31 10:40:43]       |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[2024-10-31 10:40:43] compilation terminated.
[2024-10-31 10:40:43] make[3]: *** [/data/genode/repos/base/mk/generic.mk:73: lib/base/raw_output.o] Error 1
[2024-10-31 10:40:43]     COMPILE  lib/base/sleep.o
[2024-10-31 10:40:43] /data/genode/repos/base-hw/src/lib/hw/capability.cc: In member function ‘bool Genode::Native_capability::valid() const’:
[2024-10-31 10:40:43] /data/genode/repos/base-hw/src/lib/hw/capability.cc:30:49: error: ‘cap_id_invalid’ is not a member of ‘Kernel’
[2024-10-31 10:40:43]    30 |         return (Genode::addr_t)_data != Kernel::cap_id_invalid(); }
[2024-10-31 10:40:43]       |                                                 ^~~~~~~~~~~~~~
[2024-10-31 10:40:43] make[3]: *** [/data/genode/repos/base/mk/generic.mk:72: hw/capability.o] Error 1
[2024-10-31 10:40:44] make[3]: Target 'all' not remade because of errors.
[2024-10-31 10:40:44] make[2]: *** [var/libdeps:1000: bootstrap-sram-migv.prg] Error 2
```

The reason is the following difference between the builds on master and staging.

```diff
--- ./bootstrap/board/migv.master/hw/capability.d	2024-11-01 07:56:58.118761883 +0100
+++ ./bootstrap/board/migv.staging/hw/capability.d	2024-11-01 07:56:21.578629650 +0100
@@ -1,34 +1,106 @@
 hw/capability.o hw/capability.d: \
  .../repos/base-hw/src/lib/hw/capability.cc \
- .../repos/base-hw/src/include/base/internal/capability_space.h \
- .../repos/base/include/base/capability.h \
- .../repos/base/include/util/string.h \
+ .../repos/base/src/include/base/internal/capability_space.h \
+ .../repos/base/include/base/thread.h \
+ .../repos/base/include/base/exception.h \
+ .../repos/base/include/base/blockade.h \
+ .../repos/base/include/base/lock.h \
+ .../repos/base/include/util/noncopyable.h \
+ .../repos/base/include/base/trace/logger.h \
+ .../repos/base/include/base/trace/buffer.h \
  .../repos/base/include/base/stdint.h \
  .../repos/base/include/spec/64bit/base/fixed_stdint.h \
+ .../repos/base/include/cpu_session/cpu_session.h \
+ .../repos/base/include/util/attempt.h \
+ .../repos/base/include/cpu_session/capability.h \
+ .../repos/base/include/base/capability.h \
+ .../repos/base/include/util/string.h \
  .../repos/base/include/base/output.h \
  .../repos/base/include/util/interface.h \
  .../repos/base/include/util/misc_math.h \
- .../repos/base/include/util/noncopyable.h \
  .../repos/base/include/cpu/string.h \
  .../repos/base/include/base/rpc.h \
  .../repos/base/include/util/meta.h \
  .../repos/base/include/base/native_capability.h \
- .../repos/base/include/base/exception.h \
- .../repos/base-hw/include/kernel/interface.h \
- .../repos/base-hw/include/kernel/types.h
-.../repos/base-hw/src/include/base/internal/capability_space.h:
-.../repos/base/include/base/capability.h:
-.../repos/base/include/util/string.h:
+ .../repos/base/include/cpu_thread/cpu_thread.h \
+ .../repos/base/include/base/thread_state.h \
+ .../repos/base/include/spec/riscv/cpu/cpu_state.h \
+ .../repos/base/include/base/signal.h \
+ .../repos/base/include/util/list.h \
+ .../repos/base/include/base/semaphore.h \
+ .../repos/base/include/base/mutex.h \
+ .../repos/base/include/util/fifo.h \
+ .../repos/base/include/base/affinity.h \
+ .../repos/base/include/util/xml_node.h \
+ .../repos/base/include/base/log.h \
+ .../repos/base/include/base/buffered_output.h \
[... 51 lines cut ...]
+.../repos/base/include/util/fifo.h:
+.../repos/base/include/base/affinity.h:
+.../repos/base/include/util/xml_node.h:
+.../repos/base/include/base/log.h:
+.../repos/base/include/base/buffered_output.h:
+.../repos/base/include/spec/riscv/trace/timestamp.h:
+.../repos/base/include/util/token.h:
+.../repos/base/include/dataspace/capability.h:
+.../repos/base/include/dataspace/dataspace.h:
+.../repos/base/include/base/rpc_args.h:
+.../repos/base/include/session/session.h:
+.../repos/base/include/base/quota_guard.h:
+.../repos/base/include/base/session_label.h:
+.../repos/base/include/util/arg_string.h:
+.../repos/base/include/pd_session/pd_session.h:
+.../repos/base/include/region_map/region_map.h:
+.../repos/base/include/base/ram_allocator.h:
+.../repos/base/include/base/cache.h:
+.../repos/base/include/util/register.h:
+.../repos/base/include/spec/riscv/cpu/consts.h:
+.../repos/base/src/include/base/internal/capability_data.h:
+.../repos/base/include/util/avl_tree.h:
+.../repos/base/src/include/base/internal/rpc_obj_key.h:

```

ssumpf, 2024-11-04:
@chelmuth: [38b61b0](https://github.com/ssumpf/genode-riscv/commit/38b61b08473337e5c7a7968df39c55f7e579d4fc) is a suggestions of how to fix this issue.

nfeske, 2024-11-04:
Thanks @ssumpf for pinpointing the issue so quickly.

It might be safer to determine to kernel directory based on its content rather than relying on its name and sibling position (in principle the riscv.git can reside outside genode/repos/). How do you think about the approach taken at https://github.com/genodelabs/genode/blob/master/repos/base-hw/lib/mk/core-hw.inc#L70?

ssumpf, 2024-11-04:
> Thanks @ssumpf for pinpointing the issue so quickly.
>
> It might be safer to determine to kernel directory based on its content rather than relying on its name and sibling position (in principle the [...]

Sure, as long as we only support RISC-V on base-hw this will do. Commit follows.

closed 2024-11-05 by chelmuth