genodelabs/genode #5375: sculpt: focus-follows pointer does not always work

nfeske, 2024-11-04:
As reported in the [user forum](https://genode.discourse.group/t/sculpt-24-10-impressions-compatibility/100/7), the focus-follows-pointer policy of the window manager sometimes produces unexpected results, in particular when no backdrop is present (that is when the pointer leaves the window towards an area unoccupied by any application).

nfeske, 2024-11-04:
Fixed by https://github.com/genodelabs/genode/commit/48738e46f3133884f6c0d02c465787c29094cc10

cnuke, 2024-11-05:
@nfeske thanks, the commit address the mentioned problem.

tomga, 2024-12-17:
Hi @nfeske, I'm not sure it is a right place to raise it, but I can reproduce this issue when using 24.10.3 release, where I assume this should not happen anymore. I'll try to describe conditions to reproduce this and hopefully you'll understand what is happening under the hood.

When I mentioned this focus problem on a forum, I also mentioned that  `wm_backdrop` component has issues on my setup. This was identified by you as a quota problem, that should already be solved. Currently it is still reproducible for me on each reboot, but I mention it only because it is a prerequisite for reproducing problem with focus follow mouse behavior.

Until I restart `wm_backdrop` I can fool focus follow mouse mechanism, by quickly moving between windows over an area with broken `wm_backdrop`. Focus stays on previously selected window even though mouse is already over another. After I restart `wm_backdrop`, the issue with focus is not reproducible, even though I'm not sure if it starts successfully (initially I have a Genode graphical background on one screen and on others background contains whatever was last painted on it, e.g. mouse cursor trace; after a restart of backdrop all displays are black, but everything seems to work properly).

Is it possible that this fix did not get included in 24.10.3 release of Sculpt?

nfeske, 2024-12-18:
Thanks @tomga for sharing your findings. With your detailed description, I'm able to reproduce this issue.

nfeske, 2024-12-18:
Commit 38d99c6dd1d55875c7aa2e8f785e00fbece25494 should fix this problem.

commit 48738e46 (2024-11-04): nitpicker: drop superfluous abs motion events

This patch fixes focus-follows-mouse problem reported for Sculpt 24.10
when using no (operational) backdrop.

In the absence of a background, nitpicker wrongly delivered absolute
motion events to the most recently hovered client even after leaving the
client's area to a area occupied by no client. These motion events, in
turn, confused the internal consistency of the window managers
enter/leave tracking.

Fixes #5375
commit 8d50d320 (2024-11-05): nitpicker: drop superfluous abs motion events

This patch fixes focus-follows-mouse problem reported for Sculpt 24.10
when using no (operational) backdrop.

In the absence of a background, nitpicker wrongly delivered absolute
motion events to the most recently hovered client even after leaving the
client's area to a area occupied by no client. These motion events, in
turn, confused the internal consistency of the window managers
enter/leave tracking.

Fixes #5375
commit 38d99c6d (2024-12-18): nitpicker: no absolute motion without hover

This patch enforces the invariant that absolute motion events are
delivered to the hovered client only. If no client is hovered, the event
is discarded.

Otherwise, in a situation where no client is hovered (i.e., due to a
background that does not cover the entire screen) but a focus is
defined, absolute motion events would be delivered to the focused
session. From a client's perspective, when moving the pointer from the
client to emptiness, the client would observe a leave event followed by
absolute motion. This, in turn, confuses the window manage [... cut]
commit 3600705a (2025-01-30): nitpicker: no absolute motion without hover

This patch enforces the invariant that absolute motion events are
delivered to the hovered client only. If no client is hovered, the event
is discarded.

Otherwise, in a situation where no client is hovered (i.e., due to a
background that does not cover the entire screen) but a focus is
defined, absolute motion events would be delivered to the focused
session. From a client's perspective, when moving the pointer from the
client to emptiness, the client would observe a leave event followed by
absolute motion. This, in turn, confuses the window manage [... cut]
closed 2025-01-30 by chelmuth