genodelabs/genode #5378: Decrease possibility of blank display when starting Sculpt on the MNT Pocket Reform

cnuke, 2024-11-08:
Starting the `imx8mq_fb` driver component on the MNT Pocket Reform in the usual way on Sculpt can lead to a non-working display. This is normally embodied by a `xfer timed out` message (increasing the timeout in the contrib source does not help). Restarting the driver in this situation via the heartbeat mechanism is not reliable either as the panel can still stay blank although the driver appears to be working fine (no noticeable messages in the LOG pointing to the contrary).

(As a side note, manually restarting the driver from a working state, i.e. Sculpt booted up fine, leads to a non-working display as well. The panel shows an after-image that gets increasingly dark. Again, judging by the driver LOG messages everything appears to be fine.)

However, moving the driver to another CPU leads to a working display most of the time - I have rebooted the Pocket about 20 times in a row without any display related problems before I encountered a black screen again. So it does not fix the issue completely but at least makes it somewhat more manageable (judging by the included Linux patches where certain `mdelay` timeouts were already increased looking into the timing behavior we experience with DDE Linux might be worthwhile).

cnuke, 2024-11-08:
The following commits allow for starting drivers in Sculpt with a different affinity.

* [09ec5c5](https://github.com/cnuke/genode/commit/09ec5c5e2ab5c17d484592583bee3aface9887c2) set the affinity space before everything else (otherwise drivers complain being started outside of the affinity space boundaries, which is { 1, 1 } initially)
* [98c77ca](https://github.com/cnuke/genode/commit/98c77ca0b5dd1f9a3862570fb63d4d2481664948) allows for setting the affinity in the `Child_state`
* [437f23e](https://github.com/cnuke/genode/commit/437f23eb7deebce649f1471ed60dc7a36bf97d78) stores the board name in `Board_info::Soc` and should be treated as PoC
* [c98ca86](https://github.com/cnuke/genode/commit/c98ca86f2478f347554ef1e581b177300850a9b0) checks the board to set the appropriate affinity.

nfeske, 2024-11-11:
I concur with the changes https://github.com/cnuke/genode/commit/09ec5c5e2ab5c17d484592583bee3aface9887c2 and https://github.com/cnuke/genode/commit/98c77ca0b5dd1f9a3862570fb63d4d2481664948.

However, the addition of `Board_info::Soc::board` is not appropriate. First, an SoC has no board but a board as a SoC. So the member is counter-intuitive. (BTW, the change missed to consider the `operator !=`) Second, the name of the board is of no interest but merely is used as input for the heuristics in `fb.h`. Exposing the name does not bring any benefit but widens the interface (creating uncertainty).

Since the affinity hack is board specific, it should better be expressed as that: Adding a `bool fb_on_dedicated_cpu` member to `Board_info`. Since `fb.h` has access to the `Board_info` anyway, the code stays mechnical `if (fb_on_dedicated_cpu) ...`, and the heuristics remains at the place outside, where `fb_on_dedicated_cpu` is controlled.

cnuke, 2024-11-15:
Commit [6885e71](https://github.com/cnuke/genode/commit/6885e71cea0b2dff195315e56d97092c733d1b5d) and [c0950b3](https://github.com/cnuke/genode/commit/c0950b3607828ec013649ec8ec8c160fdbd7d259) implement the functionality in the proposed way.

nfeske, 2024-11-15:
Thank you @cnuke. I merged your commit series to staging now.

closed 2024-11-20 by nfeske