genodelabs/genode #5380: hw: core leaks virtual memory

skalk, 2024-11-11:
Currently, core's region map detach function is not implemented. Moreover, a thread's UTCBs that need to be maped eagerly for the kernel do not get unmaped when a thread gets destroyed. I a system with enough dynamic workload this triggers an empty page-table allocator within core, which leads to the attempt to flush the page-table's of core/kernel. This effectively renders the system unuseable.
The solution should be to fix the virtual memory leakage, and to prevent the flushing of the page-tables if core's allocator gets empty. Thereby, at least the components already running with no further resource demand are kept alive.
Thanks to @trimpim for detecting & reporting the system hang!

nfeske, 2024-11-13:
Merged to staging.

skalk, 2024-11-15:
@nfeske: please consider to apply the fixup commit 2fa9ab8 to staging, it fixes the ordering of UTCB kernel detach and thread kernel object destruction

closed 2024-11-20 by chelmuth