genodelabs/genode #5386: libc: synchronize implicit pthread mutex initialization

cproc, 2024-11-25:
Related to #3262 and #3741. Also add a check to cond and rwlock initialization to detect if a different thread has already initialized the internal object in the meantime.

cproc, 2024-11-25:
Fixed by 85be218.

nfeske, 2024-11-25:
I may miss to see the complete picture but I think that the `try { } catch` can be omitted because the `Libc::Allocator` should never raise an exception. As far as I know, our implementation of `malloc` never returns NULL. Whenever the available memory is exhausted, a resource request is issued.

cproc, 2024-11-26:
> I may miss to see the complete picture but I think that the `try { } catch` can be omitted because the `Libc::Allocator` should never raise an exception. As far as I know, our implementation of `malloc` never returns NULL. Whenever the available memory is exhausted, a resource request is issued.

Commit 3784275 has the `try { } catch` removed from `mutex_init()` and commit 35a679d contains the corresponding cond/rwlock changes.

nfeske, 2024-11-26:
Thank you @cproc. I merged your commits to staging now.

commit 37842757 (2024-11-26): libc: synchronize implicit pthread mutex initialization

Fixes #5386
closed 2024-12-11 by chelmuth