genodelabs/genode #5390: Multi-monitor window management

nfeske, 2024-11-29:
With version 24.10, Sculpt OS became usable for mirrored and panoramic multi-monitor setups. In a panoramic setup, each display shows a portion of a larger panorama maintained by the nitpicker GUI server. So the displays combined behave like a very large display. The administrative user interface is already multi-monitor-aware. It is presented only on the display that hosts the mouse pointer, thereby ensuring that the entirety of the user interface is reachable at all times, and to limit the maximum size of the user interface to the resolution of the largest display (not the total panorama size, which can be much larger).

However, as the window manager/layouter/decorator are not yet multi-monitor-aware, window operations effect the entire panorama as a whole, which has several usability drawbacks:
* When placing a window at far right at the panorama, and then disconnecting the display showing this part of the panorama, the window position becomes out of reach for the user. The user can fix this situation only be manually adjusting the layouter's rules file, which is not user friendly.
* When maximizing a window, the windows covers the whole panorama, not only one display. In situations where the user designates displays to distinct purposes (e.g., using one display for browsing documentation while using another display for the development environment), the maximize feature interferes.
* In a similar way, the current notion of virtual desktops maps poorly to the use of the entire panorama. To accommodate distinct roles of different displays well, it would be much more useful to have an independent set of virtual desktops on each display.

To overcome these drawbacks, the window layouter must be extended with the new notion of *display*, with the configurable mapping of screens to the present displays, and with an interactive way for arranging windows across displays and virtual desktops.

nfeske, 2024-11-29:
The commit series above implements this idea for the most part.

nfeske, 2024-12-03:
Commit https://github.com/nfeske/genode/commit/9b2908c411ae262b6c1b734f09cc9c8af5fc5ad3 complements this line of work with improved focus switching via the keyboard.

The switching within the focus history (Super-Tab) is now restricted to windows that are located at visible screens. This avoids the need to tab through invisible windows, which is a nuisance when using windows scattered across several virtual desktops.

Furthermore, the focus is now automatically changed to the most recently focused visible window whenever the currently focused window becomes invisible. This happens when switching between virtual desktops. This behavior hopefully also fits well with focus handling of temporary windows. When a temporary window appears, it needs to be focused manually (to prevent focus stealing). But once the temporary window disappears, the focus returns to the previously focused window.

tomga, 2024-12-18:
> In a similar way, the current notion of virtual desktops maps poorly to the use of the entire panorama. To accommodate distinct roles of different displays well, it would be much more useful to have an independent set of virtual desktops on each display.

I'd like to challenge your opinion that current behavior "maps poorly" to entire panorama. I know this is a matter of habit, but would you consider to make this change configurable?

I know it may be only my preference, but for me switching desktops is more like switching whole set of apps that I'm using. One for mail with web, one for development (all tools and testing windows), one for remote desktop, which for me typically spans two monitors. For special cases when I wanted to have some app on multiple desktops I could set sticky bit on Linux (I used sawfish wm which allows me to set such options).

After this change is implemented, would it be possible to have one app spanning to monitors? Do you plan to support that? And what would happen after switching one of the displays?

chelmuth, 2024-12-19:
> After this change is implemented, would it be possible to have one app spanning to monitors? Do you plan to support that? And what would happen after switching one of the displays?

The display-aware design also supports `<screen/>` (and it's assigned sessions) spanning the whole panorama (see https://github.com/nfeske/genode/blob/c533765facadb39ceb6a62938fb7ec11b70378ea/repos/gems/recipes/raw/motif_wm/layouter.config#L18 where screen_0 represents the spanning background). Admittedly, the assignment must be done manually at the current stage and, thus, your dynamic concerns incl. disconnecting displays is not addressed.

tomga, 2024-12-19:
@chelmuth, thank you for this information.

When I wrote my comment yesterday I didn't notice that feature was already marked as fixed and I thought I'll be able to influence the decision during implementation. Due to that I did not attempt yet to understand what capabilities are available.

Based on information from you, I should be able to achieve similar behavior like the current one if dynamic connecting/disconnecting displays is not a requirement, and it isn't for me. My comment was a (maybe too quick) response after reading, that the feature that I used on Linux for years and for last two days on Sculpt does not work well and will be gone :-)

nfeske, 2025-02-28:
Featured in Genode 25.02