genodelabs/genode #5395: bomb on Nova Error: dangling session in parent-side ID space: service=ROM cid=5 args3

dufresnep, 2024-12-02:
The following is on Genode 24.11, but was first discovered on 24.05 (reading Foundation book)
Suspicious part:
```
[init -> bomb-master -> bomb_g5.2 -> bomb_g4.1 -> bomb_g3.2] Error: dangling session in parent-side ID space: service=ROM cid=5 args3
[init -> bomb-master -> bomb_g5.2 -> bomb_g4.1 -> bomb_g3.2] Error: Uncaught exception of type 'Genode::Ipc_error'
[init -> bomb-master -> bomb_g5.2 -> bomb_g4.1 -> bomb_g3.2] Warning: abort called - thread: ep
```

make -C build/x86_64 run/bomb KERNEL=nova BOARD=pc
```
Booting `Genode on NOVA'
Bender Version 0.9-beta7-51-g28ba2ad-dirty
Patching BDA with I/O port 0x3f8.
Bender: Hello World.
hwp config: eeo=na irq=na hwp=na epp=na epb=na

NOVA Microhypervisor v10-f74d1d0 (x86_64): [gcc 12.3.0] [MBI2]

[ 0] TSC:3624356 kHz BUS:1006821 kHz (measured)
[ 0] CORE:00:00:0 6:1a:3:0 [1] Intel Core i7 9xx (Nehalem Core i7, IBRS update) HLT+E+I

Hypervisor NOVA (API v10)
Hypervisor reports 1x1 CPU 
Warning: CPU has no invariant TSC.
mapping: affinity space -> kernel cpu id - package:core:thread
 remap (0x0) -> 0 - 0:0:0 boot cpu
ROM modules:
 ROM: [0000000031e48000,0000000031e6ab20) bomb
 ROM: [0000000031e47000,0000000031e4722e) catalog.xml
 ROM: [0000000031e46000,0000000031e4635f) config
 ROM: [0000000000009000,000000000000d000) core_log
 ROM: [0000000031e6b000,0000000031ed9160) init
 ROM: [0000000031eda000,0000000031fc3268) ld.lib.so
 ROM: [0000000000005000,0000000000008000) platform_info
 ROM: [0000000031fc4000,0000000031fdde48) timer

36892K kernel memory

Genode 24.11
759 MiB RAM and 63252 caps assigned to init
[init] Warning: bomb-master: configured RAM exceeds available RAM, proceed with 792412976
[init] Warning: bomb-master: configured caps exceed available caps, proceed with 62969
[init -> bomb-master] --- bomb started ---
[init -> bomb-master] rounds=20 generations=6 children=2 sleep=1500 demand=1024K
[init -> bomb-master] [0] It's time to start all my children...
[init -> bomb-master -> bomb_g5.1] --- bomb started ---
[init -> bomb-master -> bomb_g5.1] [0] It's time to start all my children...
[init -> bomb-master -> bomb_g5.2] --- bomb started ---
[init -> bomb-master -> bomb_g5.2] [0] It's time to start all my children...
[init -> bomb-master -> bomb_g5.1 -> bomb_g4.1] --- bomb started ---
[init -> bomb-master -> bomb_g5.1 -> bomb_g4.2] --- bomb started ---
[init -> bomb-master -> bomb_g5.1 -> bomb_g4.2] [0] It's time to start all my children...
[init -> bomb-master -> bomb_g5.1 -> bomb_g4.1] [0] It's time to start all my children...
[init -> bomb-master -> bomb_g5.2 -> bomb_g4.1] --- bomb started ---
[init -> bomb-master -> bomb_g5.2 -> bomb_g4.1] [0] It's time to start all my children...
[init -> bomb-master -> bomb_g5.2 -> bomb_g4.2] --- bomb started ---
[init -> bomb-master -> bomb_g5.2 -> bomb_g4.2] [0] It's time to start all my children...
[init -> bomb-master -> bomb_g5.1 -> bomb_g4.1 -> bomb_g3.1] --- bomb started ---
[init -> bomb-master -> bomb_g5.1 -> bomb_g4.1 -> bomb_g3.1] [0] It's time to start all my children...
[init -> bomb-master -> bomb_g5.1 -> bomb_g4.1 -> bomb_g3.2] --- bomb started ---
[init -> bomb-master -> bomb_g5.1 -> bomb_g4.1 -> bomb_g3.2] [0] It's time to start all my children...
[init -> bomb-master -> bomb_g5.2 -> bomb_g4.2 -> bomb_g3.1] --- bomb started ---
[init -> bomb-master -> bomb_g5.2 -> bomb_g4.2 -> bomb_g3.1] [0] It's time to start all my children...
[init -> bomb-master -> bomb_g5.2 -> bomb_g4.2 -> bomb_g3.2] --- bomb started ---
[init -> bomb-master -> bomb_g5.1 -> bomb_g4.2 -> bomb_g3.1] --- bomb started ---
[init -> bomb-master -> bomb_g5.1 -> bomb_g4.2 -> bomb_g3.1] [0] It's time to start all my children...
[init -> bomb-master -> bomb_g5.1 -> bomb_g4.2 -> bomb_g3.2] --- bomb started ---
[init -> bomb-master -> bomb_g5.1 -> bomb_g4.2 -> bomb_g3.2] [0] It's time to start all my children...
[init -> bomb-master -> bomb_g5.2 -> bomb_g4.1 -> bomb_g3.1] --- bomb started ---
[init -> bomb-master -> bomb_g5.2 -> bomb_g4.1 -> bomb_g3.1] [0] It's time to start all my children...
[init -> bomb-master -> bomb_g5.2 -> bomb_g4.2 -> bomb_g3.2] [0] It's time to start all my children...
[init -> bomb-master -> bomb_g5.2 -> bomb_g4.1 -> bomb_g3.2] --- bomb started ---
[init -> bomb-master -> bomb_g5.2 -> bomb_g4.1 -> bomb_g3.2] [0] It's time to start all my children...
[init -> bomb-master] [0] It's time to kill all my children...
[init -> bomb-master] virtual Bomb_child::~Bomb_child()
[init -> bomb-master -> bomb_g5.2 -> bomb_g4.1 -> bomb_g3.2] Error: dangling session in parent-side ID space: service=ROM cid=5 args3
[init -> bomb-master -> bomb_g5.2 -> bomb_g4.1 -> bomb_g3.2] Error: Uncaught exception of type 'Genode::Ipc_error'
[init -> bomb-master -> bomb_g5.2 -> bomb_g4.1 -> bomb_g3.2] Warning: abort called - thread: ep
[init -> bomb-master] virtual Bomb_child::~Bomb_child()
[init -> bomb-master -> bomb_g5.1 -> bomb_g4.1 -> bomb_g3.1 -> bomb_g2.1] Error: ROM-session creation failed (label="config", ram_qu)
[init -> bomb-master -> bomb_g5.1 -> bomb_g4.1 -> bomb_g3.1 -> bomb_g2.1] Error: could not open ROM session for "config"
[init -> bomb-master -> bomb_g5.1 -> bomb_g4.1 -> bomb_g3.1 -> bomb_g2.1] Error: Uncaught exception of type 'Genode::Rom_connection:'
[init -> bomb-master -> bomb_g5.1 -> bomb_g4.1 -> bomb_g3.1 -> bomb_g2.1] Warning: abort called - thread: ep
[init -> bomb-master -> bomb_g5.1 -> bomb_g4.1 -> bomb_g3.2 -> bomb_g2.2] Error: ROM-session creation failed (label="config", ram_qu)
[init -> bomb-master -> bomb_g5.1 -> bomb_g4.1 -> bomb_g3.2 -> bomb_g2.2] Error: could not open ROM session for "config"
[init -> bomb-master -> bomb_g5.1 -> bomb_g4.1 -> bomb_g3.2 -> bomb_g2.2] Error: Uncaught exception of type 'Genode::Rom_connection:'
[init -> bomb-master -> bomb_g5.1 -> bomb_g4.1 -> bomb_g3.2 -> bomb_g2.2] Warning: abort called - thread: ep
[init -> bomb-master] [0] Done.
[init -> bomb-master] [1] It's time to start all my children...
```

chelmuth, 2025-02-28:
I see no issue here. Other opinions?

alex-ab, 2025-02-28:
I agree, no issue as long as bomb-master continues to run. The other components/childs are destructed without any kind of agreement which may lead to arbitrary (bad) states and messages of them.

closed 2025-03-03 by chelmuth