genodelabs/genode #5399: nova/init_smp: fails occasionally on CI runs

alex-ab, 2024-12-05:
```
...
[2024-12-05 06:16:56] [init -> init_1 -> bomb-master] [8] Done.
[2024-12-05 06:16:56] [init -> init_1 -> bomb-master] [9] It's time to start all my children...
[2024-12-05 06:16:56] [init -> init_2 -> bomb-master -> bomb_g2.2 -> bomb_g1.2] --- bomb started ---
[2024-12-05 06:16:56] [init -> init_2 -> bomb-master -> bomb_g2.2 -> bomb_g1.2] I ran out of capabilities.
[2024-12-05 06:16:56] [init -> init_4 -> bomb-master] [4] It's time to kill all my children...
[2024-12-05 06:16:56] [init -> init_4 -> bomb-master] virtual Bomb_child::~Bomb_child()
[2024-12-05 06:16:56] [init -> init_2 -> bomb-master -> bomb_g2.2 -> bomb_g1.1] --- bomb started ---
[2024-12-05 06:16:56] [init -> init_2 -> bomb-master -> bomb_g2.2 -> bomb_g1.1] I ran out of capabilities.
[2024-12-05 06:16:56] [ 0] Pd::root Killed EC:0xffffffff817acb00 SC:0xffffffff8b1283c0 V:0x6 CS:0x2b IP:0x167b33(0x1) CR2:0xffffffffe0042d40 ERR:0x0 CONT:0xffffffff81000028 (PT not found)
[2024-12-05 06:16:56] Warning: unresolvable exception 0, pd 'init -> init_5 -> bomb-master -> bomb_g2.1 -> bomb_g1.2', thread 'bomb_g1.2', cpu 4, ip=0x14000 sp=0x401fe4c8 bp=0x0 no signal handler
[2024-12-05 06:16:56] Warning: unresolvable exception 0, pd 'init -> init_7 -> bomb-master', thread 'bomb-master', cpu 6, ip=0x2800 sp=0x401fd748 bp=0x0 no signal handler
[2024-12-05 06:16:57] Error: illegal WRITE at address 0x401eeff8 by pager_object: pd='init' thread='ep' ip=0x79047
[2024-12-05 06:16:57] Warning: unresolvable exception 0, pd 'init -> init_2 -> bomb-master -> bomb_g2.3 -> bomb_g1.2', thread 'bomb_g1.2', cpu 2, ip=0x6bd80 sp=0x401fe5e8 bp=0x0 no signal handler
[2024-12-05 06:16:57] Warning: unresolvable exception 0, pd 'init -> init_5', thread 'init_5', cpu 4, ip=0x14cd sp=0x401fe4c8 bp=0x0 no signal handler
...
```

Concerning here is that the kernel says that a Core thread is killed/stopped with an invalid opcode (V:0x6).
objdump -dpl build/x86_64/var/run/init_smp.core for IP 0x167b33 is:

```
_ZN4Core12Pager_objectD2Ev():
/home/alex/genode/genode.staging/repos/base-nova/src/core/pager.cc:751
  167af0:       49 bb c0 84 0e 00 00    movabs $0xe84c0,%r11
  167af7:       00 00 00 
  167afa:       41 57                   push   %r15
  167afc:       48 b8 98 f8 ff ff ff    movabs $0xfffffffffffff898,%rax
  167b03:       ff ff ff 
  167b06:       41 56                   push   %r14
  167b08:       41 55                   push   %r13
  167b0a:       41 54                   push   %r12
  167b0c:       55                      push   %rbp
  167b0d:       48 89 fd                mov    %rdi,%rbp
  167b10:       53                      push   %rbx
  167b11:       48 8d 1d d8 ff ff ff    lea    -0x28(%rip),%rbx        # 167af0 <_ZN4Core12Pager_objectD1Ev>
  167b18:       4c 01 db                add    %r11,%rbx
  167b1b:       48 83 ec 28             sub    $0x28,%rsp
  167b1f:       48 8b 04 03             mov    (%rbx,%rax,1),%rax
  167b23:       48 83 c0 10             add    $0x10,%rax
  167b27:       48 89 07                mov    %rax,(%rdi)
/home/alex/genode/genode.staging/repos/base-nova/src/core/pager.cc:754
  167b2a:       f6 87 98 01 00 00 10    testb  $0x10,0x198(%rdi)
  167b31:       75 02                   jne    167b35 <_ZN4Core12Pager_objectD1Ev+0x45>
_Z8nova_diev():
/home/alex/genode/genode.staging/repos/base-nova/include/nova/util.h:23
  167b33:       0f 0b                   ud2
_ZN4Nova10Descriptor7_assignILm4503599627370495ELm12EEEvm():
/home/alex/genode/genode.staging/repos/base-nova/include/nova/syscall-generic.h:366
  167b35:       48 8b b5 98 00 00 00    mov    0x98(%rbp),%rsi
_ZN4Nova9syscall_5ENS_7SyscallEhmRmS1_m():
/home/alex/genode/genode.staging/repos/base-nova/include/spec/64bit/nova/syscalls.h:129
```

```
Pager_object::~Pager_object()                                                   
 752 {                                                                               
 753 .   /* sanity check that object got dissolved already - otherwise bug */        
 754 .   if (!_state.dissolved())                                                    
 755 .   .   nova_die();                                                             
 756                                                                                 
 757 .   /* revoke portal used for the cleanup call and sm cap for blocking state */ 
 758 .   revoke(Obj_crd(_selectors, 2));     
```

Several points:
- why it is in the wrong state
- do we/I (author of the line) should stop the thread in this harsh way or should we turn it into a error for liveliness reasons

alex-ab, 2024-12-05:
It turns out that the code assumes that a Pager_object is _ever_ in a managed state, and thereby must be dissolved before destruction. However, Pager_object construction and manage are two separate steps, whereby the second step may fail due to Out_of_caps and/or Out_of_rams. In this situation the Pager_object got never managed and thereby also don't need to be dissolved. I would vote for removing this "debug" code to detect a potentially wrong situation (Variant B).

nfeske, 2024-12-06:
@alex-ab thanks for the very welcome fix. I agree with you and merged https://github.com/genodelabs/genode/commit/e530ae2b9a367dc2b4fe23b93ffa6a5b752bbed6 to staging.

commit e530ae2b (2024-12-06): nova: avoid false pager assertion

Fixes #5399
commit d0522706 (2024-12-10): nova: avoid false pager assertion

Fixes #5399
closed 2024-12-11 by chelmuth