genodelabs/genode #5400: vfs.h: Directory::for_each_entry does not work for recursive rmdir

sidhussmann, 2024-12-06:
When trying to recursively remove directories on file-systems that don't support recursive unlink, such as rump/ext2, the following code would not remove all contents and throw an exception because `unlink` returns with `UNLINK_ERR_NOT_EMPTY`:

```c++
		void _remove_dir(Directory::Path const &path)
		{
			Directory dir { _vfs_env, path };
			dir.for_each_entry([&] (Directory::Entry const &e) {
				auto entry_path { Directory::join(path, e.name()) };
				if (e.dir())
					_remove_dir(entry_path);

				auto unlink_res = _vfs_env.root_dir().unlink(entry_path.string());
				if (unlink_res != Vfs::Directory_service::Unlink_result::UNLINK_OK) {
					error("unlink ", entry_path," failed unlink_res=", (unsigned) unlink_res);
					throw Exception();
				}
			});
		}
```

sidhussmann, 2024-12-06:
I've pushed 28ed750 that fixes the issue for my use-case. However, since this change is in `os/vfs.h`, which is used by many crucial components, I'm not sure how I should proceed. Here are the options that I see:

- Create a throw-away run script that demonstrates the behavior
- Adapt `empty_dir()` function in `repos/os/src/test/vfs_stress/main.cc` to use the `Directory::for_each_entry`
- Add a further `rmdir` test in `repos/os/src/test/vfs_stress/main.cc` to use the `Directory::for_each_entry` that tests 
- Use my own implementation for my `rmdir` use-case and close the issue

Please let me know how to proceed.

chelmuth, 2024-12-06:
Doesn't your change lead to prematurely abort of the loop at https://github.com/sidhussmann/genode/blob/28ed7509d0b437aa43376d91a45edbe7cadcbb69/repos/os/include/os/vfs.h#L222-L223?

sidhussmann, 2024-12-07:
@chelmuth Maybe I'm missing something, but I'm pretty sure this is not the case. The iteration starts at the last non-END entry and traverses towards entry zero. Consider the following example with six entries:

```
[entry 0         ]
[entry 1         ]
[entry 2         ]
[entry 3         ] ^ direction
[entry 4         ] |
[entry 5         ] <- start
[Dirent_type::END] <- num_dirent = 6
```

However, your comment points out that the mentioned condition never matches - except if maybe another client of the backing File_system server removes an entry simultaneously during traversal.

chelmuth, 2024-12-09:
Thanks for the clarification.

chelmuth, 2024-12-09:
I had a closer look at your fix and possible impact on other components. What still puzzles me is the backwards iteration that may be unexpected. Also, I like to see `for_each_entry()` as a const function (which it's not currently). Other components use it as const for gathering potential paths to modify but postpone the modification after the loop (e.g., https://github.com/genodelabs/genode/blob/master/repos/gems/src/app/depot_remove/main.cc by @a-dmg). May this be a practical solution for your use case too?

nfeske, 2024-12-10:
I would appreciate keeping the concern about recursive deletion of directories outside of os/vfs.h. The header exists merely to accommodate the use cases by Genode itself and does not try to be exhaustive or suited as application API.

Cementing subtle behavior like the scanning of a directory in reverse order would prevent us to change this aspect in the future. Worse, since this assumption is not covered by a test case, we could break it any time. In contrast, right now, the caller can make no assumption about the order, giving us the freedom to change the implementation later without risking regressions.

My recommendation would be hosting a `remove_recursive` function in a separate header, applying the approach mentioned by @chelmuth. Let me reinforce his remark that the scanning should best be decoupled from modification.

sidhussmann, 2024-12-10:
Thank you @chelmuth and @nfeske, for your feedback. 
I understand your hesitation in merging the commit as is. I too, find it may be confusing that this function iterates backwards through the entries. I agree that the caller should not have to make any assumptions about the order of the entries.
While implementing my recursive rmdir feature, I also stumbled upon the `depot_remove` example you mentioned. However, this approach only works with directories that either contain directory-only or file-only elements:

```
dir  
├── dirA  
│   └── dir  
│       └── file  
└── dirB  
   ├── file1  
   └── file2
```

It does not work for directories that contain mixed elements, such as this example:

```
dir  
├── dirA  
│   └── dir  
│       └── file  
└── dirB  
   ├── dir  
   │   └── file  
   ├── file1  
   └── file2
```

Therefore, I discarded that approach for the time being. Today, I tried to adapt `depot_remove.run` to showcase the behavior using `genodelabs/image/sculpt-pc-2024-11-21`. However, this turned out to be less straightforward than expected. 

Looking into the `depot_remove/main.cc` again, I see how the approach for scanning directories can work for mixed-entry directories. My use case works with the changes in e89b2bd4d58ab8651acc5b0319f2f147c3f3ecb2.

@nfeske, I like the idea to add the `remove_recursive` function to a separate header. Where would be a good location for this file?

> Worse, since this assumption is not covered by a test case, we could break it any time.

As mentioned in https://github.com/genodelabs/genode/issues/5400#issuecomment-2522755707, I care about adding tests. I'm also okay with closing this issue.

nfeske, 2024-12-10:
> I'm also okay with closing this issue.

Until there is a use case of `remove_recursive` within Genode, I would prefer to that.

Regarding the implementation, I'd go for an iterative idea as sketched below:

```
void remove_files_and_directories(...)
{
  /* returns true if any progress was made */
  auto remove_any_leaf_or_empty_dir = [&]
  {
     bool progress = false;
     ... iteratively find leaf or empty dir
         if found, remove it and set progress = true ... 
     return progress;
  };
  while (remove_any_leaf_or_empty_dir());
}
```

As this approach avoids recursion, the stack usage does not depend on the file-system structure, which is desirable in my opinion. It is of course less efficient because the same hierarchies are traversed multiple times. To mitigate that effect, the function `remove_any_leaf_or_empty_dir` could remove multiple files and empty directories in one call. Since the function works actually not recursively, I picked another name.

But as I said, this is merely a rough sketch I would pursue if I tried implementing this feature.

sidhussmann, 2024-12-11:
Thank you @nfeske, for sharing your thoughts and suggestions. Your approach seems like the right way to go forward.

closed 2024-12-11 by sidhussmann