genodelabs/genode #5404: intel/display: switching connectors off is not reflected (regression)

chelmuth, 2024-12-12:
If connectors are switched off in Sculpt 24.10.3, the capture session is terminated but the *connectors* report does reflect the `enabled="false"` state (while `brightness` becomes `0` on my notebook-internal connector though).

alex-ab, 2024-12-12:
@chelmuth: please try the commit fee0c4a. Somehow the reporting of the connectors enable state seems to be not reported at all atm.

chelmuth, 2024-12-12:
IIRC we refrained from adding this attribute some weeks ago as a missing `<mode ... used="true">` reflects the same information. I had a look into *sculpt_manager* that implements its *Off* state this way and, thus, could not profit from the patch.

alex-ab, 2024-12-12:
Okay, I got than the bug report wrong, which talked about connector and enable. Though, commit 7baeea7 should help if is related to the used state.

chelmuth, 2024-12-12:
Great, it works again with the patch applied.

commit a9df9fcc (2025-01-30): intel/display: report no inuse mode on disabled connector

Fixes #5404
closed 2025-01-30 by chelmuth