genodelabs/genode #5412: intel_fb: memory allocation failed

cproc, 2025-01-08:
I get the following error (with added debug messages) on one of my test machines when I try to run Sculpt:

```
[init -> runtime -> intel_fb] Allocator_avl_base::alloc_aligned(): size: 8388608, align: 23, avail: 52093
Allocator_avl_base::alloc_aligned(): size: 8392704, align: 23, avail: 67542831104
Allocator_avl_base::alloc_aligned(): size: 8392704, align: 22, avail: 67542831104
Allocator_avl_base::alloc_aligned(): size: 8392704, align: 21, avail: 67542831104
Allocator_avl_base::alloc_aligned(): size: 8392704, align: 23, avail: 140736137416704
[init -> drivers -> platform] Allocator_avl_base::alloc_aligned(): size: 8396800, align: 23, avail: 4217597952
Allocator_avl_base::alloc_aligned(): size: 8392704, align: 21, avail: 140735321972736
Allocator_avl_base::alloc_aligned(): size: 8392704, align: 21, avail: 140735623946240
[init -> runtime -> intel_fb] Lx_kit::Mem_allocator::alloc() first attempt failed, retrying with 8392703 bytes added
[init -> runtime -> intel_fb] Allocator_avl_base::alloc_aligned(): size: 8388608, align: 23, avail: 8444796
[init -> runtime -> intel_fb] Error: memory allocation failed for 8388608 align 8388608
Error: illegal WRITE at address 0x34 by pager_object: pd='init -> runtime -> intel_fb' thread='ep' ip=0x11d0a04
```

It works when I add 'align' to the size of the new buffer, but that's another ~8M in this case and depending on how big other allocations can be, might be too much as a general solution.

cproc, 2025-01-08:
edit: the first line of the log output was missing. It's

```
[init -> runtime -> intel_fb] Lx_kit::Mem_allocator::alloc(): size: 8388608, align: 8388608
```

alex-ab, 2025-01-14:
Can you please provide the debugging patch? I don't find any code with "first attempt" or "retrying", so one has to guess what you did.

cproc, 2025-01-17:
@alex-ab: commit f9961d6 is the debugging patch

alex-ab, 2025-01-20:
Thanks @cproc. Can you please check whether  buffer.virt_addr() (line 95) is aligned according 'align'. If not, then the second attempt can not succeed, since a region is added which does not fulfill the align requirement.

cproc, 2025-01-20:
> Thanks [@cproc](https://github.com/cproc). Can you please check whether buffer.virt_addr() (line 95) is aligned according to 'align'. If not, then the second attempt can not succeed, since a region is added which does not fulfill the align requirement.

`buffer.virt_addr()` is 0x400000 and `align` is 0x800000.

alex-ab, 2025-01-23:
We had a coffee round discussion regarding the topic, I try to summarize. There exist several options:

- Extend the Genode API such, that the rm_region.attach also consider an explicit align attribute, which currently is not supported. There was mainly the common view, that this should be done eagerly, as it breaks the ABI. It may fit later better, e.g. when the  region management of a component moves from core to the components themself (in form of a utility).
- The other option is to manage the region in lx_kit explicitly, to avoid the auto merging for attached regions as done by the avl implementation currently.
- One may try to add some add_discrete(parameters + align option) in a non ABI breaking way to the avl implementation, which avoids the merging of region.
- If this does not work out, a copy of the avl implementation with this feature in lx_kit is the other option.

At moment no one is working on trying this out. @cproc feel free if you can/want to investigate.

alex-ab, 2025-01-23:
Sorry, "should be done eagerly" -> "should _not_ be done eagerly"

alex-ab, 2025-01-23:
I updated the text again with a bit more information.

chelmuth, 2025-06-04:
@alex-ab how do you interpret https://github.com/genodelabs/genode/blob/fdb74f44591315f55602ff8a45d1e7ed56e88ead/repos/base/src/core/region_map_component.cc#L218-L228 from the background of this issue? How does it intervene with the lx_kit code?

alex-ab, 2025-06-04:
I extended a bit the shown lines, to see the code regarding weaking the alignment (map_src_addr())

> [@alex-ab](https://github.com/alex-ab) how do you interpret
> 
> [genode/repos/base/src/core/region_map_component.cc](https://github.com/genodelabs/genode/blob/fdb74f44591315f55602ff8a45d1e7ed56e88ead/repos/base/src/core/region_map_component.cc#L218-L235)
> 

The loop/code does try to align the _physical_ address of the dataspace to the virtual address (presumably to reduce the mapping entries in the page table). So, this contradicts kind of the "expectation" (there is no guarantee on the Genode API level) of the caller (lx_kit here), that the chosen virtual address by core matches from the alignment the dataspace size.