genodelabs/genode #5414: lib/core-hw: depends on environment variable 'BOARD`

trimpim, 2025-01-10:
When building `<depot-usr>/bin/arm_v8a/base-hw-imx8mp_iot_gate` with `depot/create` in a shell where
teh environment variable `BOARD` isn't set, in `repos/base-hw/lib/mk/core-hw.inc` an empty string is set as
`BOARD_NAME`.

I do not have currently an idea, how to proper fix this. In the meanwhile I have a patch that aborts building the archive
if the variable is not set.

skalk, 2025-01-10:
@trimpim thank you for pointing us to this issue. It was not intended nor wanted behaviour, when adding the board name to the platform_info ROM inside of hw's core. I've prepared a commit 8d35a28, which should solve this issue.

trimpim, 2025-01-13:
@skalk thanks for the quick fix. I integrated it in to our branch.