genodelabs/genode #5423: sel4: support MSI for x86

alex-ab, 2025-01-16:
More drivers on Sculpt@seL4 on modern notebooks will work instantly with MSI support.

chelmuth, 2025-01-22:
*base: support specifying PCI bdf on irq creation*

As I get it *device_config_phys* and *bdf* carry the same information just in different flavors. Could we avoid this redundancy?

alex-ab, 2025-01-23:
I don't think you, please explain it why you think so.

chelmuth, 2025-01-23:
- device_config_phys is an MMIO address inside ECAM memory region
- the value of device_config_phys is calculated with ECAM start address + offset for PCI device
- the offset is based on the BDF value and hardware page size

As this calculation could be reversed I think both parameters carry the same information.

alex-ab, 2025-01-23:
As discussed offline with @chelmuth the description is correct for him, but unfortunately core does not know the ECAM address. The address is currently looked up by the acpi driver. Since this is the case, the calculation can not be done atm in core. As soon as the pre_boot work becomes available, which may provide the ECAM address to core in a convenient way, the double usage of the bdf and device_config_phys can be removed.

chelmuth, 2025-01-23:
Thanks for recording the outcome of our discussion. Both commits entered staging just now.

commit d12b491a (2025-01-23): sel4: add MSI support for x86

Fixes #5423
closed 2025-01-30 by chelmuth