genodelabs/genode #5428: SIMD-based blitting using ARM Neon and Intel AVX

nfeske, 2025-01-21:
At the moment, we use SIMD instructions for 2D-pixel copying only on the x86 platform. I'd like to foster the use of SIMD on ARMv8 as well, and also leverage more modern SIMD variants (AVX) on x86_64.

nfeske, 2025-01-21:
Implemented for ARM Neon and SSE on my `simd` branch https://github.com/nfeske/genode/commits/simd/.

nfeske, 2025-01-23:
My updated simd branch contains quite a few changes made after benchmarking on the PinePhone. The non-SIMD (slow) variant reads sequentially now, instead of writing sequentially. This is useful because the source is usually cached memory whereas the destination is uncached in most display drivers.

The Neon variant has been improved for cache locality and - most importantly - cache prefetching in the rotation case. On the PinePhone, the rotation adds about 40% overhead compared the non-rotated back2front copy. A complete screen update (1440x720) takes about 5.4 ms w/o rotation, and 7.7 ms with rotation.

nfeske, 2025-01-30:
The commit series for ARM Neon and SSE4 for x86_64 went to staging today. I'll defer the AVX topic to a later point.

nfeske, 2025-01-31:
Yesterday, I missed to push the adaptations in the rpi, allwinner, and imx repository. I updated those just now.

chelmuth, 2025-01-31:
Besides *driver/framebuffer/virtio* and *server/vmm* also *test/fb_bench* is broken on *arm_v8a*.

nfeske, 2025-02-28:
Featured in Genode 25.02

closed 2025-02-28 by nfeske

jschlatow, 2025-03-28:
@nfeske I noticed that, with 25.02, I cannot boot Sculpt or run/intel_fb.run on my x230. Bisecting led to a5ae1e1 as the first bad commit. Do you have any explanation? Is this reproducible on any other x230?

cnuke, 2025-03-28:
I tested the *run/intel_fb.run* run-script with current *-stating* on our *test3-x201* (i5 520M) w/o external display where it works.

chelmuth, 2025-03-28:
> @nfeske I noticed that, with 25.02, I cannot boot Sculpt or run/intel_fb.run on my x230. Bisecting led to [a5ae1e1](https://github.com/genodelabs/genode/commit/a5ae1e12bdb7d06fc5cc2e0e6eaefe8b66d7da70) as the first bad commit. Do you have any explanation? Is this reproducible on any other x230?

May you try to use the boot_fb instead of intel_fb. The line length of 1366 pixels is not a multiple of 16 bytes...

nfeske, 2025-03-28:
I'm positively shocked by the non-by-8-divisible screen resolution of a 1366px with screen. This will ultimately require us to make the blit library more flexible in this respect. I'm on it. :-/

jschlatow, 2025-03-31:
> > [@nfeske](https://github.com/nfeske) I noticed that, with 25.02, I cannot boot Sculpt or run/intel_fb.run on my x230. Bisecting led to [a5ae1e1](https://github.com/genodelabs/genode/commit/a5ae1e12bdb7d06fc5cc2e0e6eaefe8b66d7da70) as the first bad commit. Do you have any explanation? Is this reproducible on any other x230?
> 
> May you try to use the boot_fb instead of intel_fb. The line length of 1366 pixels is not a multiple of ~16~ 32 bytes...

Same with boot_fb.

chelmuth, 2025-03-31:
> Same with boot_fb.

Sorry for the red herring - boot_fb uses the same code that fails currently.

alex-ab, 2025-04-06:
> Same with boot_fb.

@jschlatow: may you please provide the boot_fb output of the resolution with also contains a pitch value ( the number of bytes that are in each row).. I was discussing with @nfeske whether there some assumptions can be made potentially

jschlatow, 2025-04-07:
> Mmh, that is unfortunate. The start address seems to be bogus for X230. Several questions would be good to be answered for X230:
> 
> * During boot, is the graphical mode enabled by GRUB (so you see the GENODE logo) ?
>   
>   * if not, boot_fb can not work - in that case we would have to convince/look into GRUB

That explains it on the x230 as there is no GENODE logo.

On the x260 and Starlite, however, the boot_fb seems to got broken between 25.02 and staging.

> * If yes, did it work on 24.10 or 25.02 with boot_fb ?

Apparently, it never did on my x230.

jschlatow, 2025-04-07:
> [@jschlatow](https://github.com/jschlatow): you may try to review your firmware settings regarding uefi vs bios legacy options, if they exists

Thanks for the hint. Switching to legacy boot does indeed restore the boot logo. However, for some reason, the Sculpt manager decides to use vesa_fb instead of boot_fb. The _platform_info_ ROM, however, reports a framebuffer with 1366x768 and pitch 5504.

jschlatow, 2025-04-07:
@alex-ab With this patch:

```
--- a/repos/base-nova/src/core/platform.cc
+++ b/repos/base-nova/src/core/platform.cc
@@ -294,7 +294,7 @@ static bool cpuid_invariant_tsc()
 struct Resolution : Register<64>
 {
        struct Bpp    : Bitfield<0, 8> { };
-       struct Type   : Bitfield<8, 8> { enum { VGA_TEXT = 2 }; };
+       struct Type   : Bitfield<8, 8> { enum { RGB = 1, VGA_TEXT = 2 }; };
        struct Height : Bitfield<16, 24> { };
        struct Width  : Bitfield<40, 24> { };
 };
@@ -688,7 +688,7 @@ Core::Platform::Platform()
                                        if (!boot_fb)
                                                return;
 
-                                       if (!efi_boot && (Resolution::Type::get(boot_fb->size) != Resolution::Type::VGA_TEXT))
+                                       if (!efi_boot && (Resolution::Type::get(boot_fb->size) != Resolution::Type::RGB))
                                                return;
 
                                        xml.node("framebuffer", [&] {
```

I get the following output:

```
[init -> runtime -> boot_fb] using boot framebuffer: 1366x768x32 @ 0xe0000000 type=1 pitch=5504
[init -> runtime -> boot_fb] Warning: surface 1376x768 mismatches texture 1366x768

```

alex-ab, 2025-04-08:
> Thanks for the hint. Switching to legacy boot does indeed restore the boot logo. However, for some reason, the Sculpt manager decides to use vesa_fb instead of boot_fb. The _platform_info_ ROM, [...]

Yes, currently vesa_fb on master/staging is preferred, because with vesa_fb you may change the resolution during runtime.

alex-ab, 2025-04-08:
@jschlatow: may you please try fa5c4fe8ebdc5d9c63dc199f36829fd7f7131392, which should give you a working boot_fb

jschlatow, 2025-04-08:
> Yes, currently vesa_fb on master/staging is preferred, because with vesa_fb you may change the resolution during runtime.

According to _repos/gems/src/app/sculpt_manager/driver/fb.h_, it looks to me as if boot_fb is preferred over vesa_fb. This is also what I observed when booting the same image on another device. Only on the x230, vesa_fb is used.

alex-ab, 2025-04-08:
> According to _repos/gems/src/app/sculpt_manager/driver/fb.h_, it looks to me as if boot_fb is preferred over vesa_fb. This is also what I observed when booting the same image on another device. Only [...]

As you noticed by your patch, in repos/base-nova/src/core/platform.cc the framebuffer is not announced on BIOS legacy booted systems so that only vesa_fb is "visible"/available. On UEFI booted systems vesa_fb most the time did not work well or VESA is not available at all.

jschlatow, 2025-04-08:
Okay. What I still don't get, however, is why base-nova still announces the boot framebuffer on BIOS legacy booted systems if the framebuffer is in text mode (in which, I suppose, width/height refer to characters not pixels).

jschlatow, 2025-04-09:
> [@jschlatow](https://github.com/jschlatow): may you please try [fa5c4fe](https://github.com/genodelabs/genode/commit/fa5c4fe8ebdc5d9c63dc199f36829fd7f7131392), which should give you a working boot_fb

Works perfectly.