genodelabs/genode #5431: arm_64, riscv: resolve ambiguity between libc's uint64_t and genode_uint64_t

nfeske, 2025-01-22:
Compilation units that include both libc headers as well as Genode's `base/fixed_stdint.h` can run into the following problem on riscv and arm_64:
```
error: reference to ‘uint64_t’ is ambiguous
In file included from .../include/libc/stdint.h:38,
                 from /usr/local/genode/tool/23.05/bin/../lib/gcc/aarch64-none-elf/12.3.0/include/arm_neon.h:35,
                 from .../repos/os/include/blit/internal/neon.h:25,
                 from .../repos/os/include/spec/arm_64/blit/blit.h:18,
                 from .../repos/os/include/nitpicker_gfx/texture_painter.h:17,
                 from .../repos/gems/src/app/menu_view/types.h:21:
.../include/libc/sys/_stdint.h:71:33: note: candidates are: ‘typedef __uint64_t uint64_t’
   71 | typedef __uint64_t              uint64_t;
      |                                 ^~~~~~~~
In file included from .../repos/base/include/base/stdint.h:18,
                 from .../repos/base/include/base/output.h:17,
                 from .../repos/base/include/base/log.h:17,
                 from .../repos/base/include/util/xml_node.h:17,
                 from .../repos/os/include/os/reporter.h:18,
                 from .../repos/gems/src/app/menu_view/main.cc:16:
.../repos/base/include/spec/64bit/base/fixed_stdint.h:48:33: note:                 ‘typedef genode_uint64_t Genode::uint64_t’
   48 |         typedef genode_uint64_t uint64_t;
```

The problem is already addressed for x86 and arm by `libports/src/lib/libc/patches/types.h`. The patch must merely be complemented for riscv and arm_64.

chelmuth, 2025-01-23:
A new issue became apparent last night on arm_v8a.

```
[2025-01-23 08:56:46] /data/genode/contrib/libc-bc721727d40eed2e4034055bd0400374fcaa4d61/include/libc/inttypes.h:53:17: error: declaration of ‘intmax_t imaxabs(intmax_t)’ conflicts with built-in declaration ‘long int imaxabs(long int)’ [-Werror=builtin-declaration-mismatch]
[2025-01-23 08:56:46]    53 | intmax_t        imaxabs(intmax_t) __pure2;
[2025-01-23 08:56:46]       |                 ^~~~~~~
```

I checked the situation like follows.

```
> echo -n "arm_v6:  "; echo | genode-arm-gcc     -march=armv6k       -c - -o - -dD -E | grep __INTMAX_TYPE__ && \
  echo -n "arm_v7a: "; echo | genode-arm-gcc     -march=armv7-a      -c - -o - -dD -E | grep __INTMAX_TYPE__ && \
  echo -n "arm_v8a: "; echo | genode-aarch64-gcc -march=armv8-a      -c - -o - -dD -E | grep __INTMAX_TYPE__ && \
  echo -n "x86_32:  "; echo | genode-x86-gcc     -m32 -march=i686    -c - -o - -dD -E | grep __INTMAX_TYPE__ && \
  echo -n "x86_64:  "; echo | genode-x86-gcc     -m64 -mcmodel=large -c - -o - -dD -E | grep __INTMAX_TYPE__ && \
  echo -n "riscv:   "; echo | genode-riscv-gcc                       -c - -o - -dD -E | grep __INTMAX_TYPE__

arm_v6:  #define __INTMAX_TYPE__ long long int
arm_v7a: #define __INTMAX_TYPE__ long long int
arm_v8a: #define __INTMAX_TYPE__ long int
x86_32:  #define __INTMAX_TYPE__ long long int
x86_64:  #define __INTMAX_TYPE__ long int
riscv:   #define __INTMAX_TYPE__ long int
```

I don't know why it breaks on arm_v8a while it works on x86_64.

ssumpf, 2025-01-23:
> A new issue became apparent last night on arm_v8a.
>
> ```
> [2025-01-23 08:56:46] /data/genode/contrib/libc-bc721727d40eed2e4034055bd0400374fcaa4d61/include/libc/inttypes.h:53:17: error: declaration of [...]

@chelmuth: The problem for riscv/arm_v8a is that in 

contrib/libc-<hash>/include/spec/arm_64/libc/machine/_types.h
contrib/libc-<hash>/include/spec/riscv/libc/machine/_types.h

`__int64_t` as well as `__unit64_t` are
```
typedef long long __int64_t;
typedef unsigned long long __uint64_t;
```

and `__intmax_t` is

`typedef __int64_t __intmax_t;`

Should we patch this?

chelmuth, 2025-01-24:
> I don't know why it breaks on arm_v8a while it works on x86_64.

We do not patch `typedef long long __int64_t;` for x86_64. See *include/libc/x86/_types.h*.

```c
typedef long __int64_t;
```

That fits `x86_64:  #define __INTMAX_TYPE__ long int` and prevent the error with `imaxabs`.

chelmuth, 2025-01-24:
Rolled back the int64_t part of the libc patch with 74283a1771024db2f2116e40a49610aff4c5adf9.

nfeske, 2025-01-24:
The `__int64_t` modification on arm64 was accidental. This caused the build failure for app/extract.  Fixup https://github.com/genodelabs/genode/commit/fa4c2af372bcd00045ef1b7bda579897f41a2314 rolls back this part of the patch.

@cproc It would be nice to take the upcoming tool-chain update as an opportunity for systematically harmonizing (or at least revisiting) all the compiler's builtin types along with the libc's `_types.h` with `genode_*_t` types. E.g., right now, `int64_t` is defined differently between 32bit and 64bit architectures. So if this type happened to occur as argument in a C++ API  exposed as ABI, the symbol would differ between 32bit and 64bit. Today we are lucky that this does not happen. It would be good to remove the remaining uncertainty for tomorrow.

chelmuth, 2025-01-27:
I guess the following error is also caused by changes in this issue (qt5_webengine/arm_v8a).

```
../../../../../../../../../../../src/qt5_webengine/2025-01-25/src/lib/qt5/qtwebengine/src/3rdparty/chromium/third_party/crashpad/crashpad/util/posix/scoped_mmap.cc: In member function ‘bool c
rashpad::ScopedMmap::ResetAddrLen(void*, size_t)’:
../../../../../../../../../../../src/qt5_webengine/2025-01-25/src/lib/qt5/qtwebengine/src/3rdparty/chromium/third_party/crashpad/crashpad/util/posix/scoped_mmap.cc:76:42: error: no matching function for call to ‘min(const size_t&, long long unsigned int)’
   76 |       result &= Munmap(old_addr, std::min(old_len_round, new_addr - old_addr));
```

closed 2025-01-30 by chelmuth

cproc, 2025-02-07:
> I guess the following error is also caused by changes in this issue (qt5_webengine/arm_v8a).
>
> ``` [...]

The mismatching types in this case are `size_t` and `uintptr_t`. Commit 729418b uses the `unsigned long` tool chain type macros for `uintptr_t` and other types which were previously derived from `__uint64_t` to fix this error and potential similar errors on arm_64 and riscv. The tool chain has type macros for many more of the types in `_types.h` which we could use as well once they are unified in the tool chain.

chelmuth, 2025-02-07:
> once they are unified in the tool chain

I love these words and am looking forward to test and merge the upcoming tool-chain update ;-)

commit fa4c2af3 (2025-01-24): fixup "libc: unify base types for arm_64 and riscv" (revert unintended __int64_t modification)

Fixes #5431
commit 287d68bb (2025-01-24): fixup "libc: unify base types for arm_64 and riscv" (revert unintended __int64_t modification)

Fixes #5431