genodelabs/genode #5432: tool/autopilot: mark incompletely executed tests

rite, 2025-01-22:


nfeske, 2025-01-30:
Thank you @rite for the excellent problem statement and the sensible suggestion for improving the tooling. Even though I haven't yet had a closer look at your commits, I very much appreciate the direction. It will certainly improve the quality of data that we at Genode Labs gather from our CI.

I'm looking forward to review your commits in detail once I have wrapped up my current line of work (#5428).

nfeske, 2025-02-04:
Now - after having looked at your commits - I like it even more. Let's go for it!

> naming the procedure assert feels "a bit too global" [?]

The naming is just right. It is intuitive and void of ceremony.

What do you think of how we should go about updating the existing run scripts?

rite, 2025-02-05:
Thanks for the affirmative feedback.

I also like to rename the procedure `get_cmd_switch` to `cmd_switch` along with adding the procedure `have_cmd_switch`. I think this blends in nicely with the now omnipresent pattern with `have_[spec|board|installed|include]`. Further, the pretty common `assert ![have_cmd_switch --autopilot]` "feels better" than `assert ![get_cmd_switch --autopilot]`. Do you approve?

I suggest that I:

1. (optionally) Provide a commit that introduces `cmd_switch` (a copy of `get_cmd_switch`) and  `have_cmd_switch`.
2. Provide commits that each updates the run scripts on a "per repo"-basis, including replacing calls of `assert_spec` with `assert [have_spec]`.
3. When these changes made it to master, external repos like genode-world, genode-imx, gapfruit-community, etc. may have to be updated accordingly.
4. Provide a commit that removes the procedures `assert_spec` and `get_cmd_switch` from `tool/run/run`.

I expect that almost everything is trivial, but I miss the infrastructure to test all changes myself. And my focus is on testing with base-linux, base-nova, base-hw.

nfeske, 2025-02-05:
> Do you approve?

I don't merely approve, but very much appreciate it!

Thanks for the generous plan, which is much more than I hoped for. Hats off for your willingness to adjust the run scripts. It goes without saying that we will do our share of testing and (potentially) fixing collateral effects that we'll observe in our CI.

rite, 2025-02-11:
I rebased on current master and pushed a refined implementation of the proposed changes in "tool" to my feature branch, together with a commit that updates the run scripts in the repo "os".

nfeske, 2025-02-13:
Splendid!

rite, 2025-02-13:
Thank you! I pushed three fixup commits regarding your feedback. Some `get_cmd_switch`- and `assert_spec`-calls that slipped through initially are now also fixed.

chelmuth, 2025-02-18:
The results of last night's CI run look promising. Though, we got 5031 lines matching `grep 'Warning: .* is deprecated, use .* instead'`...

rite, 2025-02-19:
Here is an overview of the work in progress regarding the update of the existing run script:

- [x] repos/base
- [ ] repos/base-foc
- [x] repos/base-hw (_no changes_)
- [x] repos/base-linux
- [x] repos/base-nova
- [ ] repos/base-okl4
- [ ] repos/dde_bsd
- [ ] repos/dde_linux
- [ ] repos/dde_rump
- [ ] repos/gems
- [ ] repos/hello_tutorial
- [x] repos/libports
- [x] repos/os
- [x] repos/pc
- [x] repos/ports
---
- [ ] repos/allwinner
- [ ] repos/imx
- [ ] repos/riscv
- [ ] repos/rpi
- [ ] repos/world
- [ ] repos/zynq

rite, 2025-02-19:
> Though, we got 5031 lines matching `grep 'Warning: .* is deprecated, use .* instead'`...

I introduced this warnings intentionally, but I underestimated the sheer amount of them during the transition... Commit 2382d44 and 7d46115 remove the warnings so that they don't mess up the logs. When all existing run script have been updated, there should be no warnings anymore. Maybe we could add the warning again (skip these fixups) when all repositories in genode have been updated?

rite, 2025-02-23:
I just pushed commits ^ that address the review findings:
- restore the condition in `run/cpu_balancer` (see https://github.com/genodelabs/genode/commit/e87b82d04b42c946199bbe5ebde3a88986b177e7#r152686248)
- drop `assert {![have_cmd_arg --autopilot]}` (see https://github.com/rite/genode/commit/3c8b133d62e9535a82ea53067a5169731aeacfe5#commitcomment-152847935 and https://github.com/rite/genode/commit/880b9b4ec6e38576c650eaaad64c2003fce501ee#commitcomment-152848102)

rite, 2025-02-27:
And last but not least, gems:

chelmuth, 2025-02-28:
Merged to staging.

nfeske, 2025-04-10:
All the remaining adaptation entered the master branch today. Thanks a lot @rite for the thorough work!