genodelabs/genode #5436: wrong indication of disabled screen in multiscreen settings ui

tomga, 2025-01-22:
I have spotted a minor bug in multi monitor configuration ui.

After turning screen off in: 

[image]

only boxes indicating screen brightness are changing:

[image]

Pressing on already selected option is required to enable screen.

alex-ab, 2025-01-23:
@tomga: it may be a duplicate of #5404. Can you please try with the commit of #5404. Currently it is not part of any official releases. I can also provide a image, but it may take a moment.

tomga, 2025-01-25:
Thanks @alex-ab, it looks like can be right and it can be a duplicate.

I will not test it now, as I don't want to restart because of this. You can close the issue now if you prefer to not keep it open.

closed 2025-01-30 by nfeske