genodelabs/genode #5439: doc: video demo of leitcentrale

kumavis, 2025-01-24:
I enjoyed watching @atopia's presentation on SculptOS https://www.youtube.com/watch?v=cQi0Aq2I48k&t=858s
I'm especially interested in the interaction design of leitcentrale

while the talk is good, the live demo in the video linked above was not very successful (as happens and is understandable). Is there a recorded video demo of it working? If not, I think it would be very helpful in communicating about your project and worth the time investment.

ssumpf, 2025-01-25:
@kumavis: I would suggest to create this issue as a topic (like "Video demo of Leitzentrale") in our user forum [1] because it is not related to ongoing Genode development. You will get some relevant responses there. I also suggest to close this issue.

[1] https://genode.discourse.group/

closed 2025-01-26 by kumavis

ssumpf, 2025-01-27:
@kumavis: FYI: I have started a topic on this: https://genode.discourse.group/t/collection-of-genode-video-urls/144

nfeske, 2025-01-30:
@kumavis you are not the first one suggesting this. But your posting actually prompted me to action. ;-) You can watch the result at https://genodians.org/nfeske/2025-01-30-sculpt-os-walkthrough